realsenseai/librealsense #11297: intel real sense viewer. No frames recieved.

hemanthvema, 2023-01-06:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|233522070956|
|**Firmware Version**|05.14.00.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|224123120491|
|**Firmware Update Id**|224123120491|

Please provide a description of the problem

After connecting the D435 camera to my system, I am getting the message on the viewer like No Frame recieved. I don't know  what should be exact reason, Is it because of the graphics in my system? My system specifications are Intel(R) Core(TM) i5-8265U CPU @ 1.60GHz   1.80 GHz, 8 GB RAM and Intel(R) UHD Graphics.  My goal is to connect the camera through python, I was just testing with the software before writing the code. Please help me where shoulod be the problem and what should I do to resolve the problem.

MartyG-RealSense, 2023-01-06:
Hi @hemanthvema  Your computer specification is good enough.  

When using the RealSense Viewer, are you enabling three stream types simultaneously (Depth, RGB and Motion Module).  If you are then the Viewer has a known issue where enabling these three streams can cause No Frames Received on one of the streams.  There is no fix for this issue in the Viewer except to not enable the Motion Module if you are enabling Depth and RGB.

hemanthvema, 2023-01-06:
Hi @MartyG-RealSense Thanks for your reply.

I am not enabling three streams simultaneoulsy. I am only enabling one stream at a time but unfortunately it's not working and I cannot even see where is the motion module. Allow me to send you the screenshots of my screen.

MartyG-RealSense, 2023-01-06:
Thank you.  You can insert images into the comment writing box.

hemanthvema, 2023-01-06:
Hera are the images 
[image]
[image]
[image]

MartyG-RealSense, 2023-01-06:
Thank you very much for the images.

Has the camera been stored unused for a period of weeks / months before you used the camera and found that the streams had the No Frames Received problem?

hemanthvema, 2023-01-06:
I have tested the camera at the Lab computer, there it is working but here in my personal system it is not working. So I am asking whether there is any system specification problem or viewer problem.

MartyG-RealSense, 2023-01-06:
Thank you.  Next, please check the Windows camera privacy settings to confirm that permission to access the camera and permission for applications to access the camera are both set to **On**.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

hemanthvema, 2023-01-06:
I have just checked the camera camera privacy settings, its already on. But the intel camera is not working.

hemanthvema, 2023-01-06:
I have enabled the firmware logs for the stereostream then it is working but not working in the RGB module even after the firmware logs enabled.
[image]
[image]

hemanthvema, 2023-01-06:
Hi, yes, you are correct. I am not using the official one, Now I am using the official USB 3.2 and it is working now in both stero and RGB streams. But why it worked when I used the USB 2.1 at the computer in my Lab.

Also one more doubt, you are recommending to use only the official USB 3.2 cable but for my application in the lab, I need atleast 2.5 to 3 metre USB cable. What should I do to get the official USB cable as you recommended?

hemanthvema, 2023-01-09:
Hi @MartyG-RealSense, I installed the pyrealsense2 library in my environment. Now I want to use its methods and functions for my application but there is no clear documentation for it and I've seen some examples in Python SDK files. I'm wondering if you have some sort of tutorial video on how to use it.

MartyG-RealSense, 2023-01-09:
Hi @hemanthvema  The SDK's pyrealsense2 wrapper has a searchable documentation site at the link below.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html

A good way to find examples of pyrealsense2 scripting with web searches is to start each search with the keywords **librealsense python** so that the search results are mainly content from this librealsense support forum that is about Python related issues.  For example: **librealsense python exposure**.

Videos about pyrealsense2 are usually related to the installation of it rather than programming with it, unfortunately.

closed 2023-01-20 by MartyG-RealSense