realsenseai/librealsense #11301: .ply to .bag file

alexmehta, 2023-01-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Mint 21 | 
| Kernel Version (Linux Only)    | 5.15.0-57-generic                                      | 
| Platform                                 | PC  |
| SDK Version                            |         2.53.1                 | 
| Language                            |  unity                           | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to import ply files from datasets to realsense in unity. One solution would be to convert ply files to bag files. The only issue is it seems rs-convert is a one way conversion. Any way I can use these ply datasets in my unity project?

MartyG-RealSense, 2023-01-09:
Hi @alexmehta  You could use the free open-source tool **MeshLab** to convert the ply files to **.obj** files to import into Unity as 3D models.

https://www.meshlab.net/

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

If you would prefer to keep the data in its original format then there is a ply importer script for Unity at the link below, though it is several years old now.

https://github.com/3DBear/PlyImporter

MartyG-RealSense, 2023-01-14:
Hi @alexmehta  Do you require further assistance with this case, please?  Thanks!

alexmehta, 2023-01-15:
All is well. It worked great. I'm closing the issue. 

Thanks for the help!

closed 2023-01-15 by alexmehta

MartyG-RealSense, 2023-01-15:
That's great to hear, @alexmehta - thanks very much for the update!