realsenseai/librealsense #11302: rs-server destructor

IndiGleb, 2023-01-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Sometimes rs-server crashes and call destructor. I can't understend why this is happening. If i restart rs-server it crashes again. How i can solve this problem ?

IndiGleb, 2023-01-09:
I want to know what is the reason for calling the destructor

IndiGleb, 2023-01-09:
[image]

MartyG-RealSense, 2023-01-09:
Hi @IndiGleb  Are you using librealsense SDK version **2.34.0** or newer, as 2.34.0 is the minimum for the rs-server networking system.

Have you tried installing and running the complete pre-built SD card image of the ethernet system that Intel provides at the link below to see whether the rs-server crash problem also occurs on that pre-built image?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#23-preparing-the-sd-card

IndiGleb, 2023-01-09:
i use 2.53.1. Yes i install pre-build image

MartyG-RealSense, 2023-01-09:
Have you tried installing the pre-build image without installing 2.53.1 too so that the librealsense version used is the 2.34.0 one that is included in the image?

IndiGleb, 2023-01-09:
I install pre build image, but later i install 2.53.1 to a new sd card.

IndiGleb, 2023-01-09:
rs-server works stably, but sometimes for unknown reasons call the destructor.

IndiGleb, 2023-01-09:
this rarely happens

IndiGleb, 2023-01-09:
when disconnecting the device, rs-server crashes, but without calling the destructor

MartyG-RealSense, 2023-01-09:
You could track memory usage over time with an Ubuntu system monitoring tool such as **htop** to see whether the Pi's available memory capacity is reducing over time (a 'memory leak').  A memory leak could lead to a program becoming unstable and possibly crashing.

IndiGleb, 2023-01-10:
i wasn't see memory leak. All time 235-240 mb

MartyG-RealSense, 2023-01-10:
When resetting rs-server after a crash, does it still crash again if you unplug the camera from the Pi's USB port and re-insert it before issuing the rs-server reset command?

IndiGleb, 2023-01-11:
This problem occurs very rarely, so I can't try what you suggested

MartyG-RealSense, 2023-01-11:
If using SDK 2.53.1 then the firmware version installed in the camera should be **5.14.0.0**.  If you are not using 5.14.0.0 then I highly recommend updating the firmware to that version.

MartyG-RealSense, 2023-01-17:
Hi @IndiGleb  Do you have an update about this case that you can provide please?  Thanks!

closed 2023-01-29 by MartyG-RealSense