realsenseai/librealsense #11306: "cfg.enable_device_from_file" for a ROS2 bag file

josyulakrishna, 2023-01-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Linux (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |  5.11                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.53.1                           | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description

Hi, I've recorded the data and saved it as a ros2 bag file, and I'm trying to process the data on the PC with the following code: 

```
import cv2                                
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
import pyrealsense2 as rs   
import rclpy
from rclpy.node import Node
print("Environment Ready")

def main(args=None):
    rclpy.init(args=args)
    node = Node('my_node_name')
        
    # Setup:
    pipe = rs.pipeline()
    cfg = rs.config()
    cfg.enable_device_from_file("/home/josyula/Documents/DataAndModels/7in_1.bag") 
    profile = pipe.start(cfg)

    # Skip 5 first frames to give the Auto-Exposure time to adjust
    for x in range(5):
      pipe.wait_for_frames()
      
    # Store next frameset for later processing:
    frameset = pipe.wait_for_frames()
    color_frame = frameset.get_color_frame()
    depth_frame = frameset.get_depth_frame()

    # Cleanup:
    pipe.stop()
    print("Frames Captured")
    # Setup:
    pipe = rs.pipeline()
    cfg = rs.config()
    cfg.enable_device_from_file("/home/josyula/Documents/DataAndModels/7in_1.bag") 
    profile = pipe.start(cfg)

    # Skip 5 first frames to give the Auto-Exposure time to adjust
    for x in range(5):
      pipe.wait_for_frames()
      
    # Store next frameset for later processing:
    frameset = pipe.wait_for_frames()
    color_frame = frameset.get_color_frame()
    depth_frame = frameset.get_depth_frame()

    # Cleanup:
    pipe.stop()
    print("Frames Captured")
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

When i run the code i get the following error message

```

    profile = pipe.start(cfg)
RuntimeError: Failed to resolve request. Request to enable_device_from_file("/home/josyula/Documents/DataAndModels/7in_1.bag") was invalid, Reason: Failed to create ros reader: Error reading version line

```

I believe this might be due to the ros2 bag compatibility issues. Is there any way around this, I cannot have the hardware with me, so recording a bag file is unavoidable. Any help is appreciated thank yoiu.

MartyG-RealSense, 2023-01-10:
Hi @josyulakrishna  Rosbag files recorded in ROS can be incompatible with the RealSense SDK's version of the bag format, as described at https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

Was the data recorded in the RealSense ROS2 wrapper **ros-development** please?

https://github.com/IntelRealSense/realsense-ros/releases/tag/4.51.1

You can confirm whether it is an incompatibility issue by testing one of the SDK's sample bag files from the link below with your script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

If use of ROS2 is not vital to your project then you can record SDK-bags in the **RealSense Viewer** tool using its 'Record' button.  These bags should be more compatible with an SDK script than a bag recorded with **rosbag record**.

MartyG-RealSense, 2023-01-10:
You are very welcome, @josyulakrishna - thanks very much for the update!

josyulakrishna, 2023-01-12:
@MartyG-RealSense  I am having a little trouble converting the points in the depth image to the world coordinates using the camera intrinsics 

The algorithm and code are here: 

1.  Read the data from bag file 
2. Align the depth and color images 
3. Get the intrinsics matrix from the depth_frame 
4. Project the points to world coordinates using `rs2_deproject_pixel_to_point` 

Problem: 

- Most of the pixels in the depth frame are being projected to the world origin (0,0,0) and this is giving me a distance of 0 between any two such points but this is definitely not the case 

code: 

```
    # Setup:
    pipe = rs.pipeline()
    cfg = rs.config()
    cfg.enable_device_from_file("/home/josyula/Documents/DataAndModels/treedepth/new_trees.bag") 
    profile = pipe.start(cfg)

    # Skip 5 first frames to give the Auto-Exposure time to adjust
    for x in range(2000):
      pipe.wait_for_frames()
      
    # Store next frameset for later processing:
    frameset = pipe.wait_for_frames()
    color_frame = frameset.get_color_frame()
    depth_frame = frameset.get_depth_frame()
    color = np.asanyarray(color_frame.get_data())
    plt.rcParams["axes.grid"] = False
    plt.rcParams['figure.figsize'] = [12, 6]
    plt.imshow(color)
    colorizer = rs.colorizer()
    colorized_depth = np.asanyarray(colorizer.colorize(depth_frame).get_data())
    plt.imshow(colorized_depth)
    
    # Create alignment primitive with color as its target stream:
    align = rs.align(rs.stream.color)
    frameset = align.process(frameset)

    # Update color and depth frames:
    aligned_depth_frame = frameset.get_depth_frame()
    colorized_depth = np.asanyarray(colorizer.colorize(aligned_depth_frame).get_data())

    # Show the two frames together:
    images = np.hstack((color, colorized_depth))
    plt.imshow(images)

    # Cleanup:
    pipe.stop()
    print("Frames Captured")

    dist = np.zeros((480, 848))
    for i in range(480):
      for j in range(848):
          dist[i, j] = depth_frame.get_distance(j, i)

    inds = np.logical_and(dist>0 ,dist < 0.3)

    depth_data = depth_frame.as_frame().get_data()
    mask_image = np.zeros((480,848,3))

    np.putmask(mask_image[:, :, 0], inds, 1)
    np.putmask(mask_image[:, :, 1], inds, 1)
    np.putmask(mask_image[:, :, 2], inds, 1)
    
    #see the mask imaage 
    plt.imshow(mask_image)

    dilation = cv2.dilate(mask_image, np.ones((10, 10), np.uint8))

    dilation_img = np.array(dilation * 255).astype("uint8")

    dilation_bw = cv2.cvtColor(dilation_img, cv2.COLOR_BGR2GRAY)
    im, contours, hierarchy = cv2.findContours(dilation_bw, cv2.RETR_EXTERNAL, cv2.CHAIN_APPROX_NONE)

    cv2.drawContours(dilation, contours, -1, (0, 255, 0), 3)
    cv2.imshow("1", dilation)
    cv2.waitKey(0)
    cv2.destroyWindow("1")
    
    intrinsics = depth_frame.profile.as_video_stream_profile().intrinsics

    boundaries = []
    boundaries_min = []
    points1 = []
    points2 = []
    for cnt in contours:
      cnt = cnt.reshape(-1,2)
      for y in list(set(cnt[:,1])):
        vals = cnt[cnt[:, 1] == y][:, 0]
        pix1 = [min(vals), y]
        pix2 = [max(vals), y]
        point1 = rs.rs2_deproject_pixel_to_point(intrinsics, pix1, depth_frame.get_distance(*pix1))
        point2 = rs.rs2_deproject_pixel_to_point(intrinsics, pix2, depth_frame.get_distance(*pix2))
        points1.append(point1)
        points2.append(point2)
        dist = np.linalg.norm(np.array(point2)-np.array(point1))
        boundaries += [pix1, pix2]
        boundaries_min += [pix1+ [dist]]

    boundaries_min = np.array(boundaries_min)
    boundaries = np.array(boundaries) 

    plt.plot(boundaries[:, 0], boundaries[:, 1], "r+")
    plt.imshow(dilation_bw)

```

output 

[image]

The image contains the distances between a line connecting two red points opposite each other, most values are zero while there is a visible gap between them. 

Here's the bag file used in the experiment 

https://drive.google.com/file/d/1V49lWR_pZpvqaCD9XrbamEj6UCJfihHZ/view?usp=share_link

Projected points(output detracted): 
```
points1
Out[40]: 
[[0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
...
 [-0.0, 0.0, 0.0],
 [-0.0, 0.0, 0.0]]

In [41]: points2
Out[41]: 
[[0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
...
 [0.28857186436653137, -0.08321467787027359, 0.31299999356269836],
 [0.28820353746414185, -0.08258392661809921, 0.313400000333786],
 [0.0, -0.0, 0.0],
...
[0.2838301360607147, -0.09395676851272583, 0.28939998149871826],
 [0.2831436097621918, -0.09305049479007721, 0.2886999845504761],
 [0.28284937143325806, -0.09227550774812698, 0.28839999437332153],
 [0.2821628451347351, -0.09137487411499023, 0.28769999742507935],
 [0.28118208050727844, -0.09038295596837997, 0.28669998049736023],
 [0.28020134568214417, -0.08939576148986816, 0.2856999933719635],
 [0.27980902791023254, -0.08859957754611969, 0.28529998660087585],
 [0.2795148193836212, -0.08783610910177231, 0.2849999964237213],
 [0.2792206108570099, -0.08707405626773834, 0.2847000062465668],
...
 [0.0, -0.0, 0.0],
 [0.0, -0.0, 0.0],
 [0.44234007596969604, -0.08740385621786118, 0.8071999549865723],
 [0.44234007596969604, -0.08550535142421722, 0.8071999549865723],
 [0.439022958278656, -0.08333756029605865, 0.8046000003814697],
```

Why is this happening? Is there any apparent mistake I'm making? Help is very much appreciated. Thank you!

MartyG-RealSense, 2023-01-12:
Hi @josyulakrishna  The line below draws my attention.

`colorized_depth = np.asanyarray(colorizer.colorize(aligned_depth_frame).get_data())`

Colorized depth is not RGB color data.  It is the color-shading of depth pixels in colors relating to the distance of that depth pixel from the camera. 

In the RealSense SDK's **align_depth2color.py** depth-color alignment example program, the following line is used instead:

color_frame = aligned_frames.get_color_frame()

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py#L74-L75

[image]

The **align_to** instruction aligns between **two streams**, such as depth to color or color to depth.  As colorized depth is the depth stream, aligning between depth and colorized depth would be aligning **depth to depth** and so align_to would not be able to perform alignment.

josyulakrishna, 2023-01-12:
Thank you, @MartyG-RealSense

> it does not appear that the RGB color frames are being rendered in the aligned image that your script outputs.

 Indeed I am working only with colorized depth data, I retrieved the RGB frame for later processing if its required. 

>If your goal is to align depth to color and then convert to 3D points with rs2_deproject_pixel_to_point 

 can I use `rs2_deproject_pixel_to_point` to convert a **depth pixel** to a 3D world coordinate?

MartyG-RealSense, 2023-01-13:
1.  **rs2_deproject_pixel_to_point** is typically used in a situation where depth is first aligned to color and then 2D pixels are projected to 3D points with rs2_deproject_pixel_to_point afterwards.  However, the instruction should also work with depth only without performing depth to color alignment.

An alternative way to obtain a 2D to 3D conversion is with RGB data, using **rs2_project_color_pixel_to_depth_pixel** to convert a single specific 2D color pixel into a 3D depth pixel instead of aligning the entire image.  https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008 has an example of Python code for using this instruction.

2.  You are generating your distance value like this:

```
dist = np.zeros((480, 848))
for i in range(480):
for j in range(848):
dist[i, j] = depth_frame.get_distance(j, i)
```

It may not need to be as complex, as shown in the similar Python code at https://github.com/IntelRealSense/librealsense/issues/9939 where the distance measurement is simply stored in a **dist** variable instead of dist[i,j]

```
frames = pipeline.wait_for_frames()
depth = frames.get_depth_frame()
width = depth.get_width()
height = depth.get_height()
dist = depth.get_distance(width/2, height/2)
print(dist)
```

josyulakrishna, 2023-01-13:
Thank you @MartyG-RealSense  I'll try these answers and get back to you.

MartyG-RealSense, 2023-01-24:
Hi @josyulakrishna  Do you have an update about this case that you can provide, please?  Thanks!

josyulakrishna, 2023-01-24:
@MartyG-RealSense, thanks for catching up. I still face the same issue, I made sure my depth and color frames are aligned and queried the world coordinates in the color frame, yet I get `(0,0,0)` as the output for  `rs2_deproject_pixel_to_point`

MartyG-RealSense, 2023-01-25:
I performed a further analysis of the **new_trees.bag** file in the RealSense Viewer.  The red areas on your image seem to correspond to the black silhoutte around the branch on the bag image.  These are plain black areas that are empty of depth detail and so have zero depth.  So it is understandable that these black areas surrounding the branch would return a depth value of 0.

I note in the settings of the bag that the **Depth Units** are set to '0.000100', more commonly referred to as '0.0001'.  Whilst using this value as the depth scale can result in image improvement on most RealSense 400 Series cameras, the D405 model uses different scaling (a default of '0.01' instead of the '0.001' default on the other models) and so using 0.0001 as the scale could have a very negative impact on the quality of a D405 depth image.

[image]

josyulakrishna, 2023-01-26:
Thanks again @MartyG-RealSense ! I overlooked that they were in the dark region. I will recollect the data with the depth scale you've suggested and try this again. I'll have to acquire the hardware, and once I get it, I'll update here again.

MartyG-RealSense, 2023-01-26:
You are very welcome, @josyulakrishna - I look forward to your next update.  Good luck!

josyulakrishna, 2023-01-30:
You're absolutely right @MartyG-RealSense  the points were corresponding to the black areas which is why I was getting a reading of zeros, one last question, If I get two random pixel locations at locations `(x1,y1,z1)` and `(x2,y2,z2)` and compute the distance between them using `sqrt((x2-x1)^2+(y2-y1)^2 +(z2-z1)^2)` how do I convert the distance between two points to meters? what does depth scale value indicate here? thanks again for your help!

MartyG-RealSense, 2023-01-30:
Looking at the Soarwing52 point to point measurement example program for Python that I linked to earlier in this discussion, it looks as though the approach taken in that script was to use the**get_distance** instruction (which returns distance in meters) to convert the xy of the two pixel locations to meters before inserting the meter-converted results into rs2_deproject_pixel_to_point 

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py#L94-L99

josyulakrishna, 2023-01-30:
I see. Thank you for clarification, @MartyG-RealSense,  If you don't mind, can you also tell me the significance of the "depth scale" (0.001)  while recording the bag, how is it used? And how are the depth values in the depth frame calculated?  It'd be very helpful for my project documentation write up.

MartyG-RealSense, 2023-01-30:
The section of Intel's camera tuning guide linked to below describes the purpose of the depth scale.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#change-the-depth-step-size

Section 2.3 on pages 16-17 of the data sheet document provides an explanation for how depth is calculated.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

To quote that section:

********

The left and right imagers capture the scene and send imager data to the depth imaging (vision) processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via the shift between a point on the Left image and the Right image.  The depth pixel values are processed to generate a depth frame.  Subsequent depth frames create a depth video stream. 

The depth pixel value is a measurement from the parallel plane of the imagers and not the absolute range as illustrated below.

[image]

josyulakrishna, 2023-01-31:
Thank you very much @MartyG-RealSense that's really helpful!! Do you have any recommended method to compute `pixels-per-metric` for the D405? Is this readily available? since most of the time the masks I extract seem to be in the dark areas of the depth data, I'm not getting valid depth, If I can establish a relationship between pixels per mm,  I could use these measurements for an accurate width estimate in my data. Looking forward to your response!

MartyG-RealSense, 2023-01-31:
If you mean **pixels per meter**, I do not know of a RealSense-specific method of calculating that but the link below makes some general suggestions about how to find pixels per meter for a camera.

https://stackoverflow.com/questions/50408347/how-to-find-pixel-per-meter/52093485

It may be better to instead use the RealSense SDK's **Depth Quality Tool** to analyze your depth stream and receive feedback about accuracy / error.  

https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality

Intel have a PDF guide at the link below called **Camera Depth Testing Methodology** that describes how to use the Depth Quality Tool to test depth quality.

https://dev.intelrealsense.com/docs/camera-depth-testing-methodology

If you built the SDK from packages and installed the package **librealsense2-utils**, or you built the SDK from source code with CMake and included the build flag **-DBUILD_GRAPHICAL_EXAMPLES=true** in the CMake build instruction, then the Depth Quality Tool should have been installed on your computer and be launchable in the Ubuntu terminal with the command **rs-depth-quality**

josyulakrishna, 2023-02-01:
I did build the SDK and I have the **rs-depth-quality** I will use this tool for analyzing the depth data, your responses are as always very helpful, thank you!

MartyG-RealSense, 2023-02-01:
You are very welcome, @josyulakrishna - I'm pleased that I could help.  Thanks very much for the update!

MartyG-RealSense, 2023-02-06:
Hi @josyulakrishna  Do you require further assistance with this case, please?  Thanks!

josyulakrishna, 2023-02-07:
Hi @MartyG-RealSense, not for this one. I think I can close this one, but for any questions in general, is there any way to reach out to you without having to open a new issue? thank you.

MartyG-RealSense, 2023-02-07:
Whilst it is preferable that a new issue is created on this support forum, if there is a question that you would like to raise privately because its subject cannot be disclosed publicly then you can email me.  martyx.grover@intel.com

josyulakrishna, 2023-02-07:
Perfect, thank you so much, I'll close this issue.

closed 2023-02-07 by josyulakrishna