realsenseai/librealsense #11309: ubuntu d435 Cannot access

SeoMinJong, 2023-01-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.53.1) | 
| Operating System & Version |  Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  5.15.0                                         | 
| Platform                                 |  intel mini PC |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {C }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Running "realsense-viewer" will output "Cannot access /sys/class/video4linux" and will not recognize the camera.
In fact, we confirmed that USB recognized the camera, and we tried to install it after seeing the message that "sudo apt-get-y install v4l-utils" module was installed and solved, but the result was the same.
Is there any other solution?
I'm sorry, but I'm using a translator.

SeoMinJong, 2023-01-11:
@MartyG-RealSense First of all, thank you.
I installed it following "distribution_linux.md" and then following "installation.md".

SeoMinJong, 2023-01-11:
[#2747 ](https://github.com/IntelRealSense/librealsense/issues/2747#issuecomment-453267250)
@jav_solo 's solution solved the problem.

MartyG-RealSense, 2023-01-11:
It's great to hear that you were successful, @SeoMinJong - thanks very much for the update!

closed 2023-01-17 by MartyG-RealSense