realsenseai/librealsense #11310: Build error v2.50.0 SDK without patching Kernel

dongdongbh, 2023-01-11:
### Issue Description
I want to use realsense-ros with ROS1, it recommands to use v2.50.0 SDK, the apt get seems only install last stable version, so I have to install from source following [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). The problem is my corporation IT do not allow disable secure-boot, so I can't patch the kernel. I build v2.50.0 with **DFORCE_RSUSB_BACKEND=true** flag, it shows following error:
```
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(easy.c.o): relocation R_X86_64_32S against symbol `malloc@@GLIBC_2.2.5' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(getinfo.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(mime.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(mprintf.c.o): relocation R_X86_64_32 against `.text' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(multi.c.o): relocation R_X86_64_32 against `.text' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(progress.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(rand.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(sendf.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(setopt.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(share.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(socketpair.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(socks.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(speedcheck.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(strerror.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(transfer.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(url.c.o): relocation R_X86_64_32S against symbol `fwrite@@GLIBC_2.2.5' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(urlapi.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(vtls.c.o): relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(altsvc.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(asyn-thread.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(base64.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(conncache.c.o): relocation R_X86_64_32 against hidden symbol `Curl_hash_str' can not be used when making a PIE object
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(connect.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(content_encoding.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(cookie.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(curl_ctype.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(curl_threads.c.o): relocation R_X86_64_32 against `.text' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(dotdot.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(escape.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(hash.c.o): relocation R_X86_64_32 against `.text' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(hostip.c.o): relocation R_X86_64_32 against `.text' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(http.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(http_chunks.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(http_proxy.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(inet_ntop.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(netrc.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(parsedate.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(openssl.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(doh.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(formdata.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(hostcheck.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(keylog.c.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a PIE object; recompile with -fPIE
collect2: error: ld returned 1 exit status
make[2]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/build.make:498: tools/realsense-viewer/realsense-viewer] Error 1
make[1]: *** [CMakeFiles/Makefile2:2076: tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 96%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/on-chip-calib.cpp.o
[... cut ...]
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | NA | 
| Operating System & Version |    Linux (Ubuntu 20.04)   | 
| Kernel Version (Linux Only)    | 5.4.0-136-generic                                    | 
| Platform                                 | PC  |
| SDK Version                            |  v2.50.0                          | 
| Language                            |  C                        | 
| Segment			|  Robot                        |

MartyG-RealSense, 2023-01-11:
Hi @dongdongbh  Can you confirm if you have you tried installing the ROS1 wrapper and librealsense together at the same time from packages with **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** using the wrapper's **Method 1** instructions?  This package installation uses the RSUSB backend.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#method-1-the-ros-distribution

dongdongbh, 2023-01-11:
Hi @MartyG-RealSense I installed ros-noetic-librealsense2 with source code. Because I modified part of librealsense-ros code, I do not want to use the orignal installation.

MartyG-RealSense, 2023-01-11:
The errors in your opening comment have not previously been reported in relation to installing librealsense.

If you would like to install a specific version of librealsense from packages, such as 2.50.0, there is guidance at https://github.com/IntelRealSense/librealsense/issues/10526 about installing a pa

dongdongbh, 2023-01-11:
librealsense2-dkms and librealsense2-utils
```
$ sudo apt-get install librealsense2-dkms=2.50.0-0~realsense0.6128
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '2.50.0-0~realsense0.6128' for 'librealsense2-dkms' was not found
$ sudo apt-get install librealsense2-utils=2.50.0-0~realsense0.6128
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-utils : Depends: librealsense2 (= 2.50.0-0~realsense0.6128) but 2.53.1-0~realsense0.8250 is to be installed
                       Depends: librealsense2-gl (= 2.50.0-0~realsense0.6128) but 2.53.1-0~realsense0.8250 is to be installed
                       Depends: librealsense2-net (= 2.50.0-0~realsense0.6128) but 2.53.1-0~realsense0.8250 is to be installed
E: Unable to correct problems, you have held broken packages.
```
directly install librealsense2
```
$ sudo apt install librealsense2=2.50.0-0~realsense0.6128
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2 librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 38 not upgraded.
Need to get 9,352 kB of archives.
After this operation, 22.7 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 librealsense2 amd64 2.50.0-0~realsense0.6128 [9,344 kB]
Get:2 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 librealsense2-udev-rules amd64 2.53.1-0~realsense0.8250 [7,780 B]
Fetched 9,352 kB in 4s (2,474 kB/s)
Selecting previously unselected package librealsense2:amd64.
(Reading database ... 269224 files and directories currently installed.)
Preparing to unpack .../librealsense2_2.50.0-0~realsense0.6128_amd64.deb ...
Unpacking librealsense2:amd64 (2.50.0-0~realsense0.6128) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.53.1-0~realsense0.8250_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.53.1-0~realsense0.8250) ...
Setting up librealsense2:amd64 (2.50.0-0~realsense0.6128) ...
Setting up librealsense2-udev-rules:amd64 (2.53.1-0~realsense0.8250) ...

Postinst script activated...
Presets deployment:
no src dir
tgt dir exists
Presets deployment is skipped
Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
```
then their is no `realsense-viewer` avalible

MartyG-RealSense, 2023-01-11:
Tools and examples such as realsense-viewer should be installed when the package **librealsense-utils** is installed.  I see that you installed the 2.50.0 version of librealsense2-utils but it expected the 2.53.0 packages for librealsense.

Does the same problem occur if you install both packages with a single apt install instruction, such as this:

`sudo apt-get install librealsense2-dkms=2.50.0-0~realsense0.6128 librealsense2-utils=2.50.0-0~realsense0.6128`

dongdongbh, 2023-01-11:
```
$ sudo apt-get install librealsense2-dkms=2.50.0-0~realsense0.6128 librealsense2-utils=2.50.0-0~realsense0.6128
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '2.50.0-0~realsense0.6128' for 'librealsense2-dkms' was not found
```

MartyG-RealSense, 2023-01-11:
And you connected to the server before inputting the apt-get install commands, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

dongdongbh, 2023-01-11:
Yes, I have do that
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Executing: /tmp/apt-key-gpghome.n67RssM0K7/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
$ sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u

Hit:1 https://dl.google.com/linux/chrome/deb stable InRelease
Hit:2 http://archive.ubuntu.com/ubuntu focal InRelease
Hit:3 http://ppa.launchpad.net/apandada1/brightness-controller/ubuntu focal InRelease
Hit:4 http://packages.ros.org/ros/ubuntu focal InRelease
Hit:5 http://archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:6 https://desktop-download.mendeley.com/download/apt stable InRelease
Hit:7 https://librealsense.intel.com/Debian/apt-repo focal InRelease
Hit:8 http://archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:9 http://archive.ubuntu.com/ubuntu focal-security InRelease
Reading package lists... Done
$ sudo apt-get install librealsense2-dkms=2.50.0-0~realsense0.6128 librealsense2-utils=2.50.0-0~realsense0.6128
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '2.50.0-0~realsense0.6128' for 'librealsense2-dkms' was not found
```

MartyG-RealSense, 2023-01-11:
if using the wrapper's Method 1 worked for you but you need to install your own custom version of the wrapper, a workaround may be:

1.  Install both librealsense and the wrapper with Method 1.

2.  Remove the ROS Noetic wrapper packages.

`sudo apt-get remove ros-noetic-realsense2-camera`

3.  Build your custom ROS1 wrapper from source code.

dongdongbh, 2023-01-11:
When I use Method 1.
```
The following NEW packages will be installed:
ros-noetic-librealsense2 ros-noetic-realsense2-camer
```
After that there still no `realsense-viewer` avalible.

MartyG-RealSense, 2023-01-11:
You are correct, the realsense-viewer tool is not included with a Method 1 installation.   My apologies.

dongdongbh, 2023-01-16:
Any update on this issue?

MartyG-RealSense, 2023-01-16:
The **relocation R_X86_64_32** errors that you reported in your opening comment have not been previously reported in relation to RealSense ROS, which suggests that it may be caused by an issue outside of the wrapper such as the particular computer that you are using.

The **Method 1** installation method seems to be the only one that works for you on your computer, though this method does not install realsense-viewer.  The viewer program is not necessary to use the RealSense ROS wrapper, though I can understand that it would be desirable to have access to it.

In this situation, the only course of action that I can think of would be to build realsense-viewer separately from its source code.

https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

MartyG-RealSense, 2023-01-22:
Hi @dongdongbh  Do you require further assistance with this case, please?  Thanks!

dongdongbh, 2023-01-24:
How to build realsense-viewer from code?

MartyG-RealSense, 2023-01-24:
Whilst RealSense users can create their own custom realsense-viewer and some have done so, there is not an established procedure for compiling your own realsense-viewer program.

If I were attempting it myself then I might use the following method:

1.  Download the zipped source code folder for the SDK from the 'Assets' file list on the **Releases** page and extract its contents.

https://github.com/IntelRealSense/librealsense/releases

[image]

2.  Once the zip is extracted, open the unzipped source code folder and navigate to the directory where the source of the realsense-viewer tool is located.  For example for SDK version 2.53.1:

**\librealsense-2.53.1 > tools > realsense-viewer**

[image]

3.  Use CMake to build the realsense-viewer tool with the CMakeLists.txt file in this directory.

MartyG-RealSense, 2023-01-30:
Hi @dongdongbh  Was the information in the comment above helpful, please?  Thanks!

dongdongbh, 2023-01-31:
@MartyG-RealSense I have done follows:
```
cd  realsense-viewer
mkdir build && cd build
cmake ../
make
```
But the `make` command does nothing, seems the `Cmakelist.txt` is a submodule for the whole project, it can not work standard alone.

MartyG-RealSense, 2023-01-31:
It may be a similar situation to the RealSense tutorial project **Getting Started**, which uses the rs-hello-realsense SDK example script but has a more complex CMakeLists.txt file than the one that rs-hello-realsense has in order to enable the Getting Started program to find the librealsense libraries and so run standalone.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

dongdongbh, 2023-02-06:
Finally, I manage to make it work by
```
sudo apt-get install librealsense2-dkms=2.50.0-0~realsense0.6128
sudo apt install librealsense2=2.50.0-0~realsense0.6128 librealsense2-gl=2.50.0-0~realsense0.6128 librealsense2-net=2.50.0-0~realsense0.6128
sudo apt install librealsense2-utils=2.50.0-0~realsense0.6128
```

closed 2023-02-06 by dongdongbh

MartyG-RealSense, 2023-02-06:
That's great news, @dongdongbh - thanks very much for the update and for sharing your solution!