realsenseai/librealsense #11315: D455  RGB data:No frames recieved on Viewer (v2.53.1/ubuntu20.04 /VMware Workstation 17 Pro)

qingliuzjj, 2023-01-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400-->D455 | 
| Firmware Version                   |5.14.0.0| 
| Operating System & Version |    Linux ,Ubuntu20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-57-generic                                 | 
| Platform                                 | PC, VMware® Workstation 17 Pro  |
| SDK Version                            |  v2.53.1                        | 
| Language                            |  intel realsense viewer                     | 
| Segment			|  RGB Camera display                       | 

### Issue Description
I can't get RGB data from D455 camera in ubuntu. When the stereo Module and Motion Module are enabled separately , the data can be received and displayed normally, but an error occurs when the RGB Camera is enabled:

```
"No Frames Recieved!",
librealsense-2.53.1/src/proc/synthetic-stream.cpp:48 - Exception was thrown during callback: 6rscuda9alloc_devI4int2EESt10shared_ptrIT_EiEUlPS1_E_
```

The screenshot of the error is as follows;
[image]

I searched for existing related issues and tried the following:
- Confirm the version: All the devices I use conform to the supported platforms and firmware versions indicated in v2.53.1.
- Check whether the camera is damaged: when the same device uses realsense_viewer.exe in the windows system, all data are received normally;
- Check the usb interface of the virtual machine: confirm that USB3.1 is selected, and USB3.2 is displayed when connecting to Viewer (there should be no problem here).

Please help me analyze where the problem may be (installation problem, VMware virtual machine problem, or something) and  how to solve it

closed 2023-01-12 by qingliuzjj

qingliuzjj, 2023-01-12:
I forgot the cuda factor, because I enabled cuda when I installed it, but my computer does not have a GPU unit, so  error occured. After disable the cuda option and reinstalling librealsense, all the data is normal.
RGB stream :No frames Received build with cuda (cuda 11.3) on ubuntu 20.04 (both l515 and d455) #9619

MartyG-RealSense, 2023-01-12:
It's excellent to hear that you achieved a solution, @qingliuzjj - thanks very much for the update!