realsenseai/librealsense #11320: global timestamp glitches every 4292 seconds (can go backwards)

denis-giri-inbolt, 2023-01-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | not available, as exact camera may have been moved since recording was done | 
| Operating System & Version |   Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.15.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0.3812                          | 
| Language                            |  python                          | 

### Issue Description

Every 2^32 milliseconds (4292 seconds), the realsense global time glitches for a second or so, before going back to where it's supposed to be.

The following recording was observed while retrieving 400Hz gyro samples from a D435i (see above for details), but frames going back in time were observed with different configurations (while streaming depth frames only, imu only or a combination of multiple channels), cameras (D455, D435, D435i), firmwares and SDKs (including up to at least 2.51.1.4348)

Attached, 4 images showing the periodicity of the issue as well as zooms on the issues of the recording shown.
[image]
[image]
[image]
[image]

The main suspect would be CLinearCoefficients::calc_linear_coefs not compensating properly for hardware_time register wrapping (max_device_time).
As there is no change in this file since version 2.38, I don't see any point in checking on the latest SDKs & firmwares.

MartyG-RealSense, 2023-01-13:
Hi @denis-giri-inbolt  There have been a couple of past cases at https://github.com/IntelRealSense/realsense-ros/issues/1469#issuecomment-727844189 and https://github.com/IntelRealSense/realsense-ros/issues/1756 of the IMU timestamp going back in time.  It has usually been reported in the RealSense ROS wrapper rather than librealsense.

Typically with IMU timestamp problems, disabling global time corrects the problem.  Python code for setting the on / off status of global time can be found at https://github.com/IntelRealSense/librealsense/issues/9172#issuecomment-855723153

denis-giri-inbolt, 2023-01-13:
@MartyG-RealSense , thanks for the links. I didn't think to check on the ros repo, as I use python.
Not quite sure the 2 ros tickets are relevant though, as they seem to mostly talk about IMU packet timings & interpolation, not timestamp generation. The closest thing to it might be https://github.com/IntelRealSense/realsense-ros/issues/1469#issuecomment-727844189, but doesn't match the periodicity I observed.

As for the use of hw_time (sensor.set_option(rs.option.global_time_enabled, 0)),
on the camera currently on my desk today (a D455), the hw_time seems to go at about 1.000076x real time, which is a bit too far from 1x real time for me. I kinda need the realsense frames to be timestamped on a monotonic clock that progresses at the same speed as my system clock (I'm hoping for ~10ms per day of drift)

So, my current options are:
- wait for the global_timestamp_reader lib to be fixed by someone who knows it well (how hopeful should I be?)
- try to fix it myself
- implement something similar (hopefully simpler) in python
- find a 3rd party library that does the job

MartyG-RealSense, 2023-01-13:
When global time is enabled, it attempts to reduce drift by comparing device time with system time, as described at https://github.com/IntelRealSense/librealsense/pull/3909

At https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803 and the comments below it, a RealSense team member provides detailed information about the different types of timestamp that are available and their timings.

MartyG-RealSense, 2023-01-20:
Hi @denis-giri-inbolt  Was the information in the above comment helpful to you, please?  Thanks!

denis-giri-inbolt, 2023-01-20:
not really. But this is no longer a blocking point for me:

I went with "implement something similar (hopefully simpler) in python", as the system clock used in global_timestamp_reader.cpp isn't monotonic, and as the formula for linear regression used in global_timestamp_reader.cpp has a singularity when the hardware clock is very regular... i.e. when it's a "good clock", the denominator = 0.
Basically, I used the more stable & more expensive cov(x,y)/var(x) with y = time.monotonic() (not time.time()) and x=dewrapped-non-global-time (2 separate steps, not trying to meld the 2 clock cleanup steps together)

MartyG-RealSense, 2023-01-20:
It's great to hear that you achieved a solution.  Thanks very much for the update and for sharing your method!

MartyG-RealSense, 2023-01-24:
Case closed due to solution achieved.

closed 2023-01-24 by MartyG-RealSense