realsenseai/librealsense #11325: FOV shrink

JohanBeone, 2023-01-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D405  | 
| Firmware Version                   | Latest avail as of today | 
| Operating System & Version | Ubuntu 2204  | 
| Kernel Version (Linux Only)    | Can't remember, it works |   
| Platform                                 | PC |
| SDK Version                            |  2|                          | 
| Language                               |  {C++ AND BUT NOT ALWAYS PCL                         | 
| Segment			         |  others                          | 

### Issue Description
I am trying to change the fov of the camera (X & Y) I know there is a depth filter (z), but I couldn't find anything on x and y. This is 
 to eliminate artefacts and false readings outside the area I work in (eg shrink fov top and bottom and left and right). If such a function or system is not built in, do I then have to edit the pointclpoud to have the data removed manually?

MartyG-RealSense, 2023-01-17:
Hi @JohanBeone  It is possible to crop XY but there is not a built-in function to do so and it is complicated to implement with self-created scripting.  https://github.com/IntelRealSense/librealsense/issues/11140#issuecomment-1327606972 has an example of a cropping solution that a RealSense user created, with some C++ code provided, whilst another approach is described at https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157

JohanBeone, 2023-01-17:
@MartyG-RealSense  Thank you very much for your rapid response. I will look into these.

MartyG-RealSense, 2023-01-24:
Hi @JohanBeone  Do you require further assistance with this case, please?  Thanks!

JohanBeone, 2023-01-25:
@MartyG-RealSense  Not at the moment no, I still need to implement the above but I am currently only at the POC phase of my project. If the project progresses, I'll look to implement these.

MartyG-RealSense, 2023-01-25:
Thanks very much @JohanBeone for the update.  I will close this case as you do not currently require further assistance.  Please feel free to re-open it at a future date or create a new issue if you need to do so.  Thanks again!

closed 2023-01-25 by MartyG-RealSense