realsenseai/librealsense #11332: L515 external trigger not working

ghoshsuman, 2023-01-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.04.01 | 
| Operating System & Version |  Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.8.0-63-generic     | 
| Platform                                 | PC |
| SDK Version                            |  2.51.1      | 
| Language                            |  C++ / Python                     | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to use the Realsense L515 LiDAR camera with other sensors (like cameras, IMU, Optitrack motion capture). In order to temporally align the data generated across all the sensors, I want to use an external HW sync trigger (using a microcontroller) to trigger simultaneous capture in all devices and thus later align all the timestamps in post-processing. So far, I am unable to make the HW external triggering on the L515 work.

I followed the whitepaper on multi-camera configurations and checked several issues here, but was unable to solve my problem. In order to test if HW triggering is working, I turned ON the `inter_cam_sync_mode` in the Realsense-viewer, and then started streaming depth. I connected the sync pins to a signal generator, through which I am sending pulse waves with 30 Hz and 3.3V peak-to-peak voltage and 10% duty cycle. 

Sync pins connection:
[image]

Signal generator: 
[image]

Oscilloscope:
[image]

It seems that no matter what I do (turn OFF the signal from the signal generator, change the frequency, duty cycle etc.), the depth keeps streaming in a free-running mode at the same exact frequency (~30.01 Hz). 

[image]

The L515 is basically unaffected by the external trigger signal. 

In `slave` mode, shouldn't the L515 not stream at all if no input signal is provided via the sync pins? Can you please help me solve this issue?

MartyG-RealSense, 2023-01-18:
Hi @ghoshsuman  The L515 multicam hardware sync system has a different purpose to the one for the 400 Series stereo depth cameras.  

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

Whilst the 400 Series system enables cameras to wait indefinitely for an external trigger signal before initiating capture, the L515 hardware sync system is intended only to stop multiple L515 cameras from interfering with one another by staggering their capture times.  

As you apparently only have one L515 camera, the L515 multicam hardware sync system would not be an appropriate choice for your project unfortunately.

**********

In regard to alternatives for syncing your L515 with non-RealSense sensors, using **camera metadata** to do so may be an option.  An example of doing so is described at https://github.com/IntelRealSense/librealsense/issues/2186

ghoshsuman, 2023-01-18:
Hi @MartyG-RealSense thanks for your response. However some things are still not clear to me. 

What is the expected response from L515 when set to `slave` mode but no input is provided? Is it supposed to free-run? In that case, if a low input does not stop the L515 from streaming, wouldn't it interfere with the other L515 cameras in the multi-camera setup?

With the D435 camera, I guess the waiting before trigger action happens only in GenLock mode, right? What kind of signal does the camera expect in such a mode to start recording again? I can try to emulate it with my analog signal generator.

Could you please elaborate how to use the metadata for temporal alignment? The issue you mentioned was quite vague about it. I can access the metadata for each frame using ROS.

MartyG-RealSense, 2023-01-18:
Although the L515 sync system is designed to time multi camera captures in a way that avoids interference between cameras, the paper confirms that it waits for a trigger before capture like the 400 Series genlock system does.  To quote the L515 paper:

******* 

When implementing hardware synchronization using the Intel RealSense SDK 2.0, it is can be done by enabling “Inter_Cam_Sync_Mode” which will place each L515 camera into a “slave” mode.  This ensures that cameras will only trigger their laser scanning and capture when it receives a voltage-high signal on the designated hardware trigger port.

************

In the 400 Series sync system, it waits indefinitely for a trigger in Inter Cam Sync Mode 4 and above (genlock).  In the original hardware sync slave mode 2 though, the slave camera listens for a trigger on each frame and then stops listening after a certain time period and initiates a capture independently (unsynced).

The genlock sync paper has now been withdrawn by Intel as it was an experimental proof of concept, and slave mode 2 is considered to be the validated and mature method of performing hardware sync.  The paper can be downloaded as an archived PDF document from the link below though if you wish to research the details of how its triggering via external pulse works.

[External Synchronization.pdf](https://github.com/IntelRealSense/realsense-ros/files/10359354/External.Synchronization.pdf)

There are no further details available regarding how to implement the camera metadata method outlined in https://github.com/IntelRealSense/librealsense/issues/2186 - it may be worth leaving a comment on that issue to ask for further information from the RealSense user who successfully implemented the system.

ghoshsuman, 2023-01-24:
I could not reproduce the expected behavior of waiting in the L515 camera when no input trigger is provided, as mentioned in the documentation. Is it possible to test this at your end?

Thanks for the documentation about genlock mode on the D435. I will give it a try this week.

MartyG-RealSense, 2023-01-24:
I do not have hardware sync cabling or trigger pulse generation equipment to test sync with my own L515, unfortunately.

I look forward to hearing your test results about genlock on D435.  Good luck!

MartyG-RealSense, 2023-02-05:
Hi @ghoshsuman  Do you have an update about your genlock mode tests that you can provide, please?  Thanks!

closed 2023-02-16 by MartyG-RealSense