realsenseai/librealsense #11333: Frames time out with multiple devices connected

CodyFagley, 2023-01-18:
### Required Info

| Specs | |
| ------------------------------ | ------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 0.5.13.00.50 | 
| Operating System & Version |   Rasbian Linux  | 
| Kernel Version (Linux Only)    |  5.15.74-v8   | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            | 2.51.1 | 
| Language                            |  Python   | 
| Segment			|  Robot   | 

### Issue Description
**D405 is giving an error for `[ERROR] "RuntimeError: Frame didn't arrive within 5000"` when adding secondary USB device reading from serial bus.**

The issue occurs when I connect a [USB-connected scale](https://www.adamequipment.com/cruiser-bench-checkweighing-scales).  Before the scale is connected, there is no issue producing and saving a depth image, but once the scale is connected I consistently get a timeout error.  When both devices are connected, the scale has no issue operating normally.  

The scale drivers are written to pull the data from serial stream `/dev/tty/USB0`.

MartyG-RealSense, 2023-01-19:
Hi @CodyFagley  The error message suggests to me that when the scale is connected it is disconnecting the D405 camera, causing the RealSense program to time-out after 5 seconds ('5000') has passed since the last time that new camera frames were received.

Whilst this particular problem is not one that I have previously seen reported on this librealsense support forum, a non-RealSense discussion about /dev/tty/USB0 conflicts suggested handling the devices by creating a new **udev rule**.

http://hintshop.ludvig.co.nz/show/persistent-names-usb-serial-devices/

MartyG-RealSense, 2023-01-20:
Thanks very much @CodyFagley for the update.  I'm pleased to hear that you achieved a solution by changing to a laptop, though I appreciate that the ideal solution would have been to continue using the Raspberry Pi.

MartyG-RealSense, 2023-01-26:
Case closed due to solution achieved.

closed 2023-01-26 by MartyG-RealSense