realsenseai/librealsense #11335: L515: Set max depth(code)

weisterki, 2023-01-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Platform                                 | PC|
| Language                            |  {python}                          | 

### Issue Description
I want to set the maximum depth of point cloud obtained by L515, I know it cloud be set on RealSense Viewer, but I want to operate it in coding.

I set the min depth using this:
dist = sensor_dep.set_option( rs.option.min_distance, 0)
So I try to set max depth similarly, like
dist = sensor_dep.set_option( rs.option.max_distance, 0)
but failed. I also tried the method used for D400 series, as
advnc_mode.set_depth_control(current_std_depth_control_group)
but I could not even run the code, it seems that L515 is not supported in find_device_that_supports_advanced_mode(). 

So if there is any code to set the max depth of L515? I would like to do that because I have to clipp points far away to keep the cloud efficient. Thank you!

MartyG-RealSense, 2023-01-20:
Hi @weisterki  The L515 camera model does not have Advanced Mode.  It also only supports the 'Temporal' depth post-processing filter, so the Threshold filter of the 400 Series cameras cannot be used to set a minimum and maximum depth.  Limits for the rendering of depth can instead be set with the depth visualization settings Min Distance and Max Distance, as you have done.

The Min Distance and Max Distance depth visualization settings are not sensor options but instead are software settings of the depth **colorizer**.  Here is some Python code for defining Min and Max Distance for the colorizer.

```
colorizer = rs.colorizer()
colorizer.set_option(rs.option.visual_preset, 1) # 0=Dynamic, 1=Fixed, 2=Near, 3=Far
colorizer.set_option(rs.option.min_distance, value_min)
colorizer.set_option(rs.option.max_distance, value_max)

depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
```

weisterki, 2023-01-20:
> 

Thanks Marty,  I used the code in this way:
[image]
but I can not see any difference. I tried to change several settings but they look same. By the way, could this method wipe off points farther than 1.5m?  I mean change the point cloud but not only the visualization.

MartyG-RealSense, 2023-01-20:
I noe that in your script the max_distance code is placed after the depth_image has been stored in the numpy array.  I recommend moving the colorizer code further up the script, such as just after the pipe start instruction.

weisterki, 2023-01-20:
> 

I move the code to this place, but still not work :<
[image]

MartyG-RealSense, 2023-01-20:
Please try setting the preset to '0' for Dynamic, as that is the one that the RealSense Viewer uses by default.

weisterki, 2023-01-26:
> Please try setting the preset to '0' for Dynamic, as that is the one that the RealSense Viewer uses by default in its colorization settings.

Sorry to say that it does still not work :(. I don't know which module results in this situation, so I post my code

# Show camera stream and export PLY data

from ctypes import sizeof
import math
import time
import pyrealsense2 as rs
import numpy as np
import cv2
import os

class AppState:
    def __init__(self, *args, **kwargs):
        self.WIN_NAME = 'My RealSense Controller'
        self.pitch, self.yaw = math.radians(-10), math.radians(-15)
        self.translation = np.array([0, 0, -1], dtype=np.float32)
        self.distance = 2
        self.prev_mouse = 0, 0
        self.mouse_btns = [False, False, False]
        self.paused = False
        self.decimate = 0
        self.scale = True
        self.color = True

    def reset(self):
        self.pitch, self.yaw, self.distance = 0, 0, 2
        self.translation[:] = 0, 0, -1

    @property
    def rotation(self):
        Rx, _ = cv2.Rodrigues((self.pitch, 0, 0))
        Ry, _ = cv2.Rodrigues((0, self.yaw, 0))
        return np.dot(Ry, Rx).astype(np.float32)

    @property
    def pivot(self):
        return self.translation + np.array((0, 0, self.distance), dtype=np.float32)

def mouse_cb(event, x, y, flags, param):
    
    if event == cv2.EVENT_LBUTTONDOWN:
        state.mouse_btns[0] = True

    if event == cv2.EVENT_LBUTTONUP:
        state.mouse_btns[0] = False

    if event == cv2.EVENT_RBUTTONDOWN:
        state.mouse_btns[1] = True

    if event == cv2.EVENT_RBUTTONUP:
        state.mouse_btns[1] = False

    if event == cv2.EVENT_MBUTTONDOWN:
        state.mouse_btns[2] = True

    if event == cv2.EVENT_MBUTTONUP:
        state.mouse_btns[2] = False

    if event == cv2.EVENT_MOUSEMOVE:

        h, w = out.shape[:2]
        dx, dy = x - state.prev_mouse[0], y - state.prev_mouse[1]

        if state.mouse_btns[0]:
            state.yaw += float(dx) / w * 2
            state.pitch -= float(dy) / h * 2

[... 251 lines cut ...]
    cv2.setWindowTitle(
        state.WIN_NAME, "My RealSense Controller: (%dx%d) %dFPS (%.2fms) %s " %
        (w, h, 1.0/dt, dt*1000, "PAUSED" if state.paused else ""))

    cv2.imshow(state.WIN_NAME, out)
    key = cv2.waitKey(1)

    if key == ord("r"):
        state.reset()

    if key == ord("p"):
        state.paused ^= True

    if key == ord("e"):
        file = "./output"
        if not os.path.exists(file):
            os.mkdir(file)
        points.export_to_ply("./output/out.ply", mapped_frame)
        print("Output Successfully!")

    if key in (27, ord("q")) or cv2.getWindowProperty(state.WIN_NAME, cv2.WND_PROP_AUTOSIZE) < 0:
        break

# Stop streaming
pipeline.stop()

MartyG-RealSense, 2023-01-26:
Whilst min_distance and max_distance will affect how the depth data is colored, it will not remove depth data that is outside of the defined range like the post-processing **threshold filter** on the 400 Series cameras, which the L515 camera model does not support.

An approach that has been used in the past to exclude certain depth values is to check the depth of each pixel and set its value to '0' (no depth) if the depth value is larger than a certain number.  A C++ example of doing so is at https://github.com/IntelRealSense/librealsense/issues/1027 (there is not a Python example of the technique available).

Alternatively, it may be worth trying the L515 instruction **rs2_option_enable_max_usable_range** which "turns on / off the maximum usable depth sensor range, given the amount of ambient light in the scene".  A Python example of turning off the maximum usable range with this instruction (setting it to '0') in combination with min_distance is at https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501

weisterki, 2023-01-27:
> Whilst min_distance and max_distance will affect how the depth data is colored, it will not remove depth data that is outside of the defined range like the post-processing **threshold filter** on the 400 Series cameras, which the L515 camera model does not support.
> 
> An approach that has been used in the past to exclude certain depth values is to check the depth of each pixel and set its value to '0' (no depth) if the depth value is larger than a certain number. A C++ example of doing so is at #1027 (there is not a Python example of the technique available).
> 
> Alternatively, it may be worth trying the L515 instruction **rs2_option_enable_max_usable_range** which "turns on / off the maximum usable depth sensor range, given the amount of ambient light in the scene". A Python example of turning off the maximum usable range with this instruction (setting it to '0') in combination with min_distance is at [#8244 (comment)](https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501)

Thanks, Marty! Your answer is useful. I tried the method of https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501, I set laser power to 100 and receiver gain at 18, but the result was acceptable only when I set the confidence at 3. Here is my code, hope to be useful for any guy has same problem.
`sensor_dep.set_option(rs.option.enable_max_usable_range, 0)`
`power = sensor_dep.set_option(rs.option.laser_power, 100)`
`gain = sensor_dep.set_option(rs.option.receiver_gain, 18)`
`confidence = sensor_dep.set_option(rs.option.confidence_threshold, 3)`

MartyG-RealSense, 2023-01-27:
It's great to hear that you achieved a solution!  Thanks so much for the update and for sharing your solution with the RealSense community  :)

weisterki, 2023-01-27:
> > Whilst min_distance and max_distance will affect how the depth data is colored, it will not remove depth data that is outside of the defined range like the post-processing **threshold filter** on the 400 Series cameras, which the L515 camera model does not support.
> > An approach that has been used in the past to exclude certain depth values is to check the depth of each pixel and set its value to '0' (no depth) if the depth value is larger than a certain number. A C++ example of doing so is at #1027 (there is not a Python example of the technique available).
> > Alternatively, it may be worth trying the L515 instruction **rs2_option_enable_max_usable_range** which "turns on / off the maximum usable depth sensor range, given the amount of ambient light in the scene". A Python example of turning off the maximum usable range with this instruction (setting it to '0') in combination with min_distance is at [#8244 (comment)](https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501)
> 
> Thanks, Marty! Your answer is useful. I tried the method of [#8244 (comment)](https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501), I set laser power to 100 and receiver gain at 18, but the result was acceptable only when I set the confidence at 3. Here is my code, hope to be useful for any guy has same problem. `sensor_dep.set_option(rs.option.enable_max_usable_range, 0)` `power = sensor_dep.set_option(rs.option.laser_power, 100)` `gain = sensor_dep.set_option(rs.option.receiver_gain, 18)` `confidence = sensor_dep.set_option(rs.option.confidence_threshold, 3)`

> Whilst min_distance and max_distance will affect how the depth data is colored, it will not remove depth data that is outside of the defined range like the post-processing **threshold filter** on [...]

> > Whilst min_distance and max_distance will affect how the depth data is colored, it will not remove depth data that is outside of the defined range like the post-processing **threshold filter** on the 400 Series cameras, which the L515 camera model does not support.
> > An approach that has been used in the past to exclude certain depth values is to check the depth of each pixel and set its value to '0' (no depth) if the depth value is larger than a certain number. A C++ example of doing so is at #1027 (there is not a Python example of the technique available).
> > Alternatively, it may be worth trying the L515 instruction **rs2_option_enable_max_usable_range** which "turns on / off the maximum usable depth sensor range, given the amount of ambient light in the scene". A Python example of turning off the maximum usable range with this instruction (setting it to '0') in combination with min_distance is at [#8244 (comment)](https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501)
> 
> Thanks, Marty! Your answer is useful. I tried the method of [#8244 (comment)](https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501), I set laser power to 100 and receiver gain at 18, but the result was acceptable only when I set the confidence at 3. Here is my code, hope to be useful for any guy has same problem. `sensor_dep.set_option(rs.option.enable_max_usable_range, 0)` `power = sensor_dep.set_option(rs.option.laser_power, 100)` `gain = sensor_dep.set_option(rs.option.receiver_gain, 18)` `confidence = sensor_dep.set_option(rs.option.confidence_threshold, 3)`

However, I found the depth filter used in this issue https://github.com/IntelRealSense/librealsense/issues/10344#issue-1184715482 could bring better performance. I tried the depth filter in this way:
`frames = pipeline.wait_for_frames()`
`depth_frame = frames.get_depth_frame()`,
` threshold_filter = rs.threshold_filter(min_dist=0.1, max_dist=1)`
`depth_frame = threshold_filter.process(depth_frame)`
`color_frame = frames.get_color_frame()`
`depth_frame = decimate.process(depth_frame)`
 and yes could bring point cloud in required range, but the edge of the cloud always flickering and drifting, as shown in the figure.
[image]
I thought that may caused by the colour frame, thus I change to this:
`frames = pipeline.wait_for_frames()`
`depth_frame = frames.get_depth_frame()`
`threshold_filter = rs.threshold_filter(min_dist=0.1, max_dist=1)`
`frame = threshold_filter.process(depth_frame)`
`color_frame = frames.get_color_frame()`
`depth_frame = decimate.process(depth_frame)`
I don't know why but it worked. And if I  change the 
`depth_frame = decimate.process(depth_frame)` to 
`depth_frame = decimate.process(frame)` it flickers again. I want to ask am I right to use the depth filter, and if yes, how to use it corrrectly?

MartyG-RealSense, 2023-01-27:
In https://github.com/IntelRealSense/librealsense/issues/10344 I highlight later in the discussion at https://github.com/IntelRealSense/librealsense/issues/10344#issuecomment-1081885052 that the L515 only supports the **Temporal** filter for depth and the **Decimation** filter for color.

 If you are experiencing depth flickering then changing the Temporal filter's **Smooth Alpha** setting to a value of '0.1' instead of its default of '0.4' stabilizes fluctuations and helps to keep the depth values consistent.

weisterki, 2023-01-30:
> **Temporal** filter

Thanx. Could you please explain the temporal filter in L515?

MartyG-RealSense, 2023-01-30:
Information about the temporal filter can be found in Intel's RealSense post-processing filter guide at the link below.

https://dev.intelrealsense.com/docs/post-processing-filters#temporal-filter

weisterki, 2023-01-31:
> Information about the temporal filter can be found in Intel's RealSense post-processing filter guide at the link below.
> 
> https://dev.intelrealsense.com/docs/post-processing-filters#temporal-filter

> Thanks, Marty! Your answer is useful. I tried the method of [#8244 (comment)](https://github.com/IntelRealSense/librealsense/issues/8244#issuecomment-779127501), I set laser power to 100 and [...]

> Whilst min_distance and max_distance will affect how the depth data is colored, it will not remove depth data that is outside of the defined range like the post-processing **threshold filter** on [...]

I tried to turn on the maximum range of L515 using the code:
`sensor_dep.set_option(rs.option.enable_max_usable_range, 1)`,
but I got this ERROR：‘Please set 'VGA' resolution and 'Max Range' preset before enabling Max Usable Range‘. The same ERROR occurred in the RealSense Viewer as well. How to deal with that?

MartyG-RealSense, 2023-01-31:
VGA resolution on the L515 is 640x480.  I confirmed that this error occurs unless the Visual Preset is Max Range **and** the resolution is 640x480.

The 'Max Range' preset is not the same as the **Colorizer Visual Preset**, which is in the 'Depth Visualization' section of the Viewer's side-panel.  You can set the main Visual Preset to 'Max Range' and the resolution to 640x480 but change the colorizer preset and the Enable Max Usable Range option will still work.

weisterki, 2023-01-31:
> n

Problem solved! Thank you：)!

MartyG-RealSense, 2023-01-31:
That's excellent news.  You are very welcome  :)