realsenseai/librealsense #11336: const char* + int

xandox, 2023-01-20:
https://github.com/IntelRealSense/librealsense/blob/v2.53.1/src/win/win-helpers.cpp#L864

Yesterday I got error "enNamedMutex returned error" and found that string formatting is wrong on line I have provided.

MartyG-RealSense, 2023-01-20:
Hi @xandox  This error message has not been previously reported on this librealsense support forum.

Are you using a Windows computer?  And if you are, have you installed the RealSense SDK using the **Intel.RealSense.SDK-WIN10** automated installer file in the 'Assets' file list of the **Releases** page, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

[image]

MartyG-RealSense, 2023-01-20:
Is the RealSense SDK able to work normally despite the message?

xandox, 2023-01-20:
Bug which issue this message, was in 2.49.0 version, but in 2.53.1 it's look like is fixed. Right now I am checking this. 
But, again, this is minor bug, but still bug. This line of code should looks something like:
`throw std::runtime_error(std::string("OpenNamedMutex returned error ") + std::to_string(stsOpenMutex));`

MartyG-RealSense, 2023-01-20:
Thanks very much for the report.  :)

closed 2023-02-01 by MartyG-RealSense