realsenseai/librealsense #11337: Reading Intrinsics of IR1/IR2

Siegel20, 2023-01-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.53.1 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |   | 
| Platform                                 | PC |
| SDK Version                            |  2.0                          | 
| Language                            |  python }                          | 
| Segment			|  Robot                          | 

### Issue Description
I would like to read the intrinsics of the IR1 and IR2 camera. If I understood correctly, the 1280 x 800 images are either distortion corrected (Y8) or the original sensor RAW image (Y16). 

Only in the case of Y8 I can get some intrsinsics (the distortion coefficients are all zero). But then the paramerts differ from what I read with Intel.Realsense.CustomRW.exe -r. 
=> I assume that the distortion corrected image changes the principal point and the focal length due to the alignment ?

In the other case (Y16) I cannot read any intrinsics at all. I would like to get them to keep them with such images so that I could do the distortion correction offline at some other point if required.
=> Is there a way to read them via the API rather than using Intel.Realsense.CustomRW.exe -r ?

Thanks!

MartyG-RealSense, 2023-01-20:
Hi @Siegel20   You are correct, the Y8 infrared format is rectified (it has a distortion model applied to it) and Y16 is unrectified.

You can list the intrinsics and extrinsics for all resolutions supported by your particular camera model using the program 'rs-enumerate-devices' by launching it in calibration information mode with the command **rs-enumerate-devices -c**

If you have installed the full RealSense SDK on Windows using the installer file **Intel.RealSense.SDK-WIN10** then instructions for launching a pre-built executable version of rs-enumerate-devices in the Windows command prompt interface can be found at https://github.com/IntelRealSense/librealsense/issues/8689#issuecomment-809118054

If you would like to retrieve intrinsics with Python then https://github.com/IntelRealSense/librealsense/issues/3986 provides an example of code for doing so with the API instruction **get_intrinsics**

In regard to all-zero coefficients, a RealSense team member explains the reason for this at https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916

Since that information was originally provided, there have been instances where coefficients have been non-zero, such as the color stream on the D455 camera model.

MartyG-RealSense, 2023-01-23:
You are very welcome, @Siegel20 - thanks very much for the update!

Siegel20, 2023-06-14:
Hello Marty,

Siegel20, 2023-06-14:
Hello again,

I read https://github.com/IntelRealSense/librealsense/issues/10323  and I found out, that I am now facing the exact same problem which means, I would also like to sign an NDA and receive infos about the homography that Intel uses to align the sensors based on the calibration. Could you please provide me with whatever is needed for these steps? Thank you very much!

MartyG-RealSense, 2023-06-14:
Hi @Siegel20  Please email your name and address to the email address below and I will pass the contact details on to my Intel RealSense colleagues.

martyx.grover@intel.com

Siegel20, 2023-06-15:
Thanks for establishing the contact.

closed 2023-06-15 by Siegel20