realsenseai/librealsense #11340: Hardware sync pyrealsense 2

dashizm, 2023-01-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Firmware Version                   | (5.14.0) | 
| Operating System & Version |   {Win 10) / Ubuntu 22.04   | 
| Kernel Version (Linux Only)    |  ( 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.53.1 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
I want to try an external trigger, I already set up the function generator and send the trigger signal. Whenever I run the code I got this error:
Cannot change Inter-camera HW synchronization mode while streaming
Here is my code:

import pyrealsense2 as rs
pipe = rs.pipeline()
prof = pipe.start()
dev = prof.get_device()
ds = dev.query_sensors()[0]
ds.set_option(rs.option.inter_cam_sync_mode, 1)
ds.get_option(rs.option.inter_cam_sync_mode)

MartyG-RealSense, 2023-01-22:
Hi @dashizm  I recommend moving the inter_cam_sync_mode code to the line **before** the pipe start instruction.

Also, the script that you are using was originally designed for setting a RealSense camera as the master that generates the trigger signal.  As you are using an external trigger signal though, inter_cam_sync_mode should be set to 2 (slave) instead of 1 (master).  This is because the function generator is acting as the master and not a RealSense camera.  Setting the mode to 1 would be instructing the camera to also generate its own trigger signal instead of listening for the trigger from the function generator.

This is confirmed in the section of the 400 Series multiple camera hardware sync paper linked to below, which states "If an external sync is being provided, then all (cameras) should be set to slave".

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#a-hw-sync-command

dashizm, 2023-01-22:
Thank you for your reply. The point is that I need to define a device before using the "ds.set_option(rs.option.inter_cam_sync_mode, 2) "command and it needs to define a profile with "pipe. start()".  How I can use the "set_options" command before using the "get_device" command? in other words, I should introduce the sensor to the "set-option" command and it needs to start the pipeline.

MartyG-RealSense, 2023-01-22:
https://github.com/IntelRealSense/librealsense/issues/9483 demonstrates using a **ctx** command to define inter_cam_sync_mode before the pipe start and find the sensor by looking for the first depth sensor available instead of defining a sensor name.

```
ctx.query_devices()[0].first_depth_sensor().set_option(rs.option.inter_cam_sync_mode, 'mode number')
pipe = rs.pipeline(ctx)
```

dashizm, 2023-01-22:
Thanks, it works for me. 
ctx = rs.context()

MartyG-RealSense, 2023-01-22:
You are very welcome, @dashizm - thanks very much for the update about your success!

closed 2023-01-28 by MartyG-RealSense