realsenseai/librealsense #11348: auto exposure limit [ RS2_OPTION_AUTO_EXPOSURE_LIMIT,] clarification request

AndreV84, 2023-01-24:
Could you clarify how to get it work? The example below did not work, as per dev's feedback
```
rs2_set_option((rs2_options*) sensor, RS2_OPTION_AUTO_EXPOSURE_LIMIT, 8000, &e);
```

AndreV84, 2023-01-24:
update : seems it may be firmware version related issue
reference url: https://github.com/IntelRealSense/librealsense/issues/10241#issuecomment-1222140658

MartyG-RealSense, 2023-01-24:
Hi @AndreV84  Please see my responses at https://github.com/IntelRealSense/librealsense/issues/10241#issuecomment-1401621308 in regard to this question.  Thanks!

MartyG-RealSense, 2023-01-30:
Hi @AndreV84  Do you require further assistance with this case, please?  Thanks!

AndreV84, 2023-01-30:
@MartyG-RealSense 
Thank you for following up
the suggestion was to try with downgraded firmware
or with the development branch or with both? or only one of the two?

MartyG-RealSense, 2023-01-30:
Both suggestions are valid, though I would recommend trying the firmware downgrade first as it is the quickest and easiest of the two methods.

AndreV84, 2023-02-06:
@MartyG-RealSense 
We haven’t tested it yet
Once the remote team updates I shall get update you in this regard
Thanks

MartyG-RealSense, 2023-03-17:
Hi @AndreV84  Do you still require assistance with this case, please?  Thanks!

AndreV84, 2023-03-17:
@MartyG-RealSense 
consider the ticket closed, please

closed 2023-03-17 by AndreV84

MartyG-RealSense, 2023-03-17:
Thanks very much @AndreV84  :)

AndreV84, 2023-03-23:
@MartyG-RealSense 
On one of jetsons I've build the sdk with rsusb=on
```cmake -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DBUILD_WITH_CUDA:bool=true -DCMAKE_CUDA_ARCHITECTURES=72 -DCMAKE_CUDA_COMPILER_ENV_VAR=/usr/local/cuda-11.4/bin/nvcc -DCMAKE_CUDA_COMPILER=/usr/local/cuda-11.4/bin/nvcc -D OpenGL_GL_PREFERENCE=GLVND ..```

```
Connected devices:
 23/03 04:58:25,660 WARNING [70368799660288] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
1) Name: Intel RealSense D455
```

comments about the warning?

MartyG-RealSense, 2023-03-23:
In the moment that the warning occurs, there may be a communication problem between the camera and the Jetson.  If the warning only occurs a small number of times during launch and then stops, or generates once or twice per minute after launch, then it can be safely ignored.

AndreV84, 2023-03-23:
@MartyG-RealSense 

https://github.com/NVIDIA-ISAAC-ROS/.github/blob/main/profile/realsense-setup.md
`I'm m trying to run nvidia  container with realsense

it as inside the same sdk version
but it won't list the camera from inside of the container while it will list the camera from outside of the container

MartyG-RealSense, 2023-03-23:
It may be worth first looking at the general non-RealSense video in the link below about using a GMSL camera in Docker with Nvidia.

https://www.youtube.com/watch?v=-swTFYO3Fuc

AndreV84, 2023-03-23:
I do not see where to procure that specific autoware-docker [ as in the video] deployment to test with it
how from nvidia realsense docker we can query camera or camera module from terminal to determine if it is detected by the system or not? thanks

MartyG-RealSense, 2023-03-23:
There are no previously reported cases of RealSense being used with Autoware, so I do not have advice to offer about that.

Autoware.Auto can apparently be used with Docker though.

https://idorobotics.com/2021/08/31/step-by-step-autoware-auto-installation-guide/

AndreV84, 2023-03-23:
I managed to setup autoware
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/
but I do not see any clue how to integrate gmsl into the setup

moreover I have found a separate concerned in gmsl segment of tier4 that supports jetson 
https://github.com/tier4/perception_ecu_container
but haven't managed to deploy it
neither see a clue how to get gsml d457 outputs with it

MartyG-RealSense, 2023-03-23:
I have highlighted your question about using D457 in Docker to my Intel RealSense colleagues.

AndreV84, 2023-03-23:
[image]

MartyG-RealSense, 2023-03-23:
Thanks very much @AndreV84 for the image!  Were you able to access your D457 in an Autoware Docker container using the above information?

My colleagues have advised that D457 has not been tested in Docker and if it cannot be detected in Docker then do not use it in Docker.  They also recommended approaching Nvidia if you require further advice on this subject.

AndreV84, 2023-03-23:
the chat assumption is that there is a camera launch file + camera driver for ros that generates camera topic [ perhaps like v4l topic]
the nvidia limitation for running vslam is to use their container with humble docker
will have to investigate it further

AndreV84, 2023-03-23:
what are the chances that installing udev rules with d455 connected could result in faulty occurance sometimes? does it make sense to reinstall once installed udev rules that were installed while d455 connected m but to reinstall it again with d455 disconnected?

MartyG-RealSense, 2023-03-23:
I would think that having one particular camera model attached during udev rule installation would not negatively affect how the udev rules handle other models, as the same set of udev rules handle all RealSense camera models that are plugged in.

AndreV84, 2023-03-24:
which ros node you could suggest to generate topics from d457? maybe you could share recorded rosbag with such topics for playing them for experimenting with autoware? Thanks

AndreV84, 2023-03-24:
does ros realsense node without docker generate gmsl camera topics?
@MartyG-RealSense

MartyG-RealSense, 2023-03-24:
There is currently no information available about using D457 with RealSense ROS in GMSL mode.  In USB mode it should behave like a D455 and be able to generate topics.  I will highlight your question to the RealSense ROS development team.

MartyG-RealSense, 2023-03-24:
The RealSense ROS team confirmed that D457 support has already been merged into the **ros2-development** wrapper in January 2023 by https://github.com/IntelRealSense/realsense-ros/pull/2557 and confirmed to work with **rs_launch.py**

 You should be able to check which topics are published in the wrapper by a D457 on a GMSL connection with **ros2 topic list**