realsenseai/librealsense #11355: Auto-exposure AE algorithm for RGB sensor on D435

patilnabhi, 2023-01-24:
Hi, is it possible to share the auto-exposure (AE) algorithm for the RGB sensor on D435? I am in the process of understanding how the exposure values change given different scenarios. 

Thank You!

MartyG-RealSense, 2023-01-24:
Hi @patilnabhi  The RealSense RGB auto-exposure algorithm is not available for sharing, unfortunately.  A RealSense team member explains the reason for this in point 2 at https://github.com/IntelRealSense/librealsense/issues/5457#issuecomment-570744452

Also, when RGB auto-exposure is enabled the current exposure values cannot be retrieved.  Whilst the actual exposure values for the depth sensor can be obtained via camera **metadata**, this feature is not supported for the RGB sensor, as confirmed at https://github.com/IntelRealSense/librealsense/issues/7267#issuecomment-688295655

patilnabhi, 2023-01-25:
Thank you for the reply @MartyG-RealSense. Just one follow-up question - when RGB is in **manual** exposure mode, can one set and retrieve the exposure values?

MartyG-RealSense, 2023-01-25:
Yes, you can set and retrieve manual exposure / gain values.  Examples of scripting for exposure get / set are provided below.

**GETTING**

C++
https://github.com/IntelRealSense/librealsense/issues/6019

Python
https://github.com/IntelRealSense/librealsense/issues/10480#issuecomment-1121046166

https://community.intel.com/t5/Items-with-no-label/Enable-Auto-Exposure-Using-Python-Code/m-p/642331/highlight/true#M14554

Note: in the script in the abive link, change this line:

`sensor_dep = profile.get_device().first_depth_sensor()`

to this:

`sensor_dep = profile.get_device().first_color_sensor()`

**SETTING**

C++
https://github.com/IntelRealSense/librealsense/issues/10239#issuecomment-1040104921

https://github.com/IntelRealSense/librealsense/issues/6019

**Python**
https://github.com/IntelRealSense/librealsense/issues/4449

https://github.com/IntelRealSense/librealsense/issues/2683#issuecomment-436395308

https://stackoverflow.com/questions/64700551/rgb-image-captured-by-intel-realsense-camera-is-dark-using-python-code/64718977#64718977

patilnabhi, 2023-01-25:
Thank you for the confirmation

closed 2023-01-25 by patilnabhi