realsenseai/librealsense #11361: Integral Scanning for L515

weisterki, 2023-01-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L500 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   {Win (8.1/10)  |                                    | 
| Platform                                 | PC |
| SDK Version                            |  { 2.53.1 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Smartphone/VR/AR }                          | 

### Issue Description
<Is there any code example for L515 integral scanning in python or C++? I want to record the point cloud of  integrated view, like head scanning, does RealSense have embedded algorithm for point cloud registration?>

MartyG-RealSense, 2023-01-26:
Hi @weisterki  As you have a Windows computer, the L515-compatible commercial 3D scanning software product **DotProduct Dot3D** may meet your requirements for generating a pointcloud scan.

https://www.dotproduct3d.com/blog/l515

https://www.dotproduct3d.com/dot3d.html

*********

If you would prefer an open-source solution then the RealSense SDK's C++ **rs-kinfu** example program may help.  It enables a 360 degree point cloud scan to be captured by a single camera by moving the camera around the scene, building up the detail on the cloud by fusing frames together.  

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

As far as I am aware, rs-kinfu has not been previously tried with the L515 camera model though and so may need some modification to be able to work with it, such as:

1.  Changing the depth resolution from 1280x720 to 1024x768

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L190

2.  Commenting out the visual preset line.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L204

3.  Removing any references to post-processing filters.

weisterki, 2023-01-30:
> Hi @weisterki As you have a Windows computer, the L515-compatible commercial 3D scanning software product **DotProduct Dot3D** may meet your requirements for generating a pointcloud scan. [...]

Very useful suggestion！I tried the kinfu.cpp，but I don't know how to configure the environment. Now the wrong information is 'can not open example.hpp'. Could you give me more instructions?

MartyG-RealSense, 2023-01-30:
I believe that the rs-kinfu instructions are suggesting that once OpenCV has been installed, the librealsense SDK should then be built from source code with CMake, with the following build flags included in the CMake build instruction.

**-DBUILD_CV_EXAMPLES -DBUILD_CV_KINFU_EXAMPLE -DBUILD_GRAPHICAL_EXAMPLES**

For example:

`cmake ../ -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_CV_EXAMPLES=true --DBUILD_CV_KINFU_EXAMPLE=true`

weisterki, 2023-02-01:
> I believe that the rs-kinfu instructions are suggesting that once OpenCV has been installed, the librealsense SDK should then be built from source code with CMake, with the following build flags [...]

> I believe that the rs-kinfu instructions are suggesting that once OpenCV has been installed, the librealsense SDK should then be built from source code with CMake, with the following build flags [...]

I did this:
`cd C:\Users\weist\Desktop\realsensetest\librealsense\wrappers\opencv\kinfu`
`cmake -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_CV_EXAMPLES=true -DBUILD_CV_KINFU_EXAMPLE=true`
but wrong occurred:

     CMake Error at CMakeLists.txt:6 (find_package):
     By not providing "Findglfw3.cmake" in CMAKE_MODULE_PATH this project has
     asked CMake to find a package configuration file provided by "glfw3", but
     CMake did not find one.

    Could not find a package configuration file provided by "glfw3" (requested
    version 3.3) with any of the following names:

    glfw3Config.cmake
    glfw3-config.cmake

     Add the installation prefix of "glfw3" to CMAKE_PREFIX_PATH or set
    "glfw3_DIR" to a directory containing one of the above files.  If "glfw3"
    provides a separate development package or SDK, be sure it has been
    installed.

Is the method I used wrong? or what should I do to solve the glfw3 problem?

MartyG-RealSense, 2023-02-01:
There is a case of this error message at https://github.com/glfw/glfw.github.io/issues/45

GLFW is available for Windows as pre-made binaries or as source code at the link below.

https://www.glfw.org/download.html

weisterki, 2023-02-02:
> There is a case of this error message at [glfw/glfw.github.io#45](https://github.com/glfw/glfw.github.io/issues/45)
> 
> GLFW is available for Windows as pre-made binaries or as source code at the link below.
> 
> https://www.glfw.org/download.html
Thank you for reply. I checked the issue you mentioned, but it seems that the question is still open. I tried some of the methods, like @sparshpriyadarshi, I installed the `libigl[glfw]`, and found the `glfw3Config.cmake`. Then I tried to Cmake the Kinfu profile again, same error occurred. I also tied to edit the `CMakeLists.txt` in Kinfu, add`set(OpenCV_DIR C:/src/vcpkg/packages/glfw3_x86-windows/share/glfw3)` in front of `find_package()`, but nothing changed. What should I do now?
I'm not familiar with Cmake, so if you could sent me a executable Kinfu project? I'm a little urge to have that function.

MartyG-RealSense, 2023-02-02:
I am unable to provide a built executable version of rs-kinfu, unfortunately.

As a possible alternative to rs-kinfu, there is information about Python resources for pointcloud registration at https://github.com/IntelRealSense/librealsense/issues/9832#issuecomment-938701928

There is also a RealSense-compatible Python library for creating and modifying pointclouds called **pyntcloud**.

https://github.com/daavoo/pyntcloud

https://pyntcloud.readthedocs.io/en/latest/

MartyG-RealSense, 2023-02-07:
Hi @weisterki  Were the resources linked to in the comment above helpful to you, please?  Thanks!

weisterki, 2023-02-08:
> Hi @weisterki Were the resources linked to in the comment above helpful to you, please? Thanks!

Yes, thanks a  lot Marty. But I got a question: why is the edge part of  pointcloud shows white color?  Here is one example:
[image]
This bothers me because it shows white texture after I using some pointcloud alignment algorithms.

MartyG-RealSense, 2023-02-08:
If the white area represents an area of the real-world scene where an observed object has a reflective surface then that might manifest as white on the RGB image and so appears on the aligned pointcloud image.

Does setting a **Short Range** camera configuration preset on your L515 improve the image?

```
option = rs.option.visual_preset
preset = rs.l500_visual_preset.short_range
depth_sensor = depth_sensor.set_option(option, preset)
```

weisterki, 2023-02-13:
> If the white area represents an area of the real-world scene where an observed object has a reflective surface then that might manifest as white on the RGB image and so appears on the aligned pointcloud image.
> 
> Does setting a **Short Range** camera configuration preset on your L515 with the Python code below improve the image?
> 
> ```
> option = rs.option.visual_preset
> preset = rs.l500_visual_preset.short_range
> depth_sensor = depth_sensor.set_option(option, preset)
> ```

I added these code to my project and turned on short range mode. I found that the white edge is actually the high reflectance background. I thought that would disappear if I capture in a matte environment. 
Here is another question: I captured pointclouds of object 0.3m from device, but I found black points scattered around the object, seems like strips. As shown in the figure:

[image]
This is side view of the point cloud. Aftering moving the object far from the sensor, this affect will disappear. Hence I believe that was caused by the scattering noise of the object surface. Any suggestion about that?

MartyG-RealSense, 2023-02-13:
Page 19 of the L515's PDF user guide document that is available at the link below advises that when capturing depth from smooth surfaces, you may get better results if the camera is placed above the object and tilted diagonally downwards instead of facing the camera straight-ahead.

https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

You could also try different presets.  If the room has no natural light from windows then **No Ambient Light** is suitable.  If there is a small amount of natural light from windows then try **Low Ambient Light**.  The L515 model is sensitive to infrared light sources such as sunlight.

MartyG-RealSense, 2023-02-19:
Hi @weisterki  Do you require further assistance with this case, please?  Thanks!

weisterki, 2023-02-20:
> Hi @weisterki Do you require further assistance with this case, please? Thanks!

I solved the problem by Open3D, I appreciate your help and you could close this issue.

MartyG-RealSense, 2023-02-20:
I'm pleased to hear that you achieved a solution, @weisterki - thanks very much for the update!

closed 2023-02-20 by MartyG-RealSense