realsenseai/librealsense #11362: pyrealsense2 support for Python 3.11

pasandal, 2023-01-26:
Hi,
Ref the Python 3.11 support request in [this ](https://github.com/IntelRealSense/librealsense/issues/10364)issue.
At what point do you consider Python 3.11 to no longer be in a pre-release state? Can we expect the next pyrealsense2 release to include Python 3.11 support?

MartyG-RealSense, 2023-01-26:
Hi @pasandal  Previously, Python 3.11 was categorized by Python themselves as a **release preview** as of August 8 2022.

https://www.python.org/downloads/release/python-3110rc1/

On October 24 2022, a final **stable** release of 3.11 was made.

https://www.python.org/downloads/release/python-3110/

In light of the availability of a stable release of 3.11, I will discuss the issue with my Intel RealSense colleagues.

pasandal, 2023-01-26:
Ok. Thanks for the update.

MartyG-RealSense, 2023-04-28:
As Python 3.11 support has been merged into the SDK, I will close this case.  Thanks, everyone!

closed 2023-04-28 by MartyG-RealSense

paul-sandal-elas-ai, 2023-05-02:
When will the SDK with Python 3.11 support be published?

MartyG-RealSense, 2023-05-02:
Hi @luigipacheco  I do not have a time estimate for release of the next SDK version.  The source code of the in-progress next version, which has already had Python 3.11 support added, is available on the SDK's **development** branch though.

https://github.com/IntelRealSense/librealsense/tree/development

lgjonathan, 2023-11-01:
Hi,
is there any plan to release for linux_aarch64 platform on python 3.11 version?

MartyG-RealSense, 2023-11-01:
Hi @lgjonathan  You can compile the pyrealsense2 wrapper from source code with modern Python versions. 

In regard to support for **pip install pyrealsense2** installation for aarch64, at the time of writing this the newest Python version supported with aarch64 is Python 3.9.  Support for pip install for Python 3.11 was enquired about recently and my Intel RealSense colleagues provided a response at https://github.com/IntelRealSense/librealsense/issues/12188#issuecomment-1740576882