realsenseai/librealsense #11363: Camera measures are scaled by 0.5 in x,y and z dimensions. How can I change this?

louridoc, 2023-01-26:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

MartyG-RealSense, 2023-01-26:
Hi @louridoc  Please provide more information about your problem.

For example, if you are using the RealSense Viewer and the **Decimation** filter is enabled in the **Post Processing** section of the Viewer's options side-panel then X and Y coordinates will be scaled depending upon the value that the filter is set to.  Left-clicking on the blue icon beside the filter to change it from blue (On) to red (Off) would disable the scaling effect of the filter.

The Z-depth coordinate would not usually be significantly changed in that situation though.

[image]

MartyG-RealSense, 2023-02-02:
Hi @louridoc  Do you require further assistance with this case, please?  Thanks!

louridoc, 2023-02-03:
Hi, MartyG. Thanks so much for your reply, and sorry for getting back so late. I disabled the decimation filter, but it didn't help much. I followed the tare calibration procedure, and that helped. Now I am getting accurate depth measurements.

closed 2023-02-03 by louridoc

MartyG-RealSense, 2023-02-03:
That's excellent to hear. @louridoc - thanks very much for the update!