realsenseai/librealsense #11364: Firmware upgrade on D455 - Failed to locate a device in FW update mode error

kumar-akshay324, 2023-01-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 } | 
| Firmware Version                   | 5.14.0 | 
| Operating System & Version |   Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX  |
| SDK Version                            |  2.53.1                          | 
| Language                            |  -                          | 

### Issue Description
Alright, so I am trying to update the firmware on a realsense. I have the latest firmware binary and using the standard command is returns a `Failed to locate a device in FW update mode` error.

I am able to successfully force the firmware upgrade in the recovery mode though.

Can someone tell me what is happening here?

```
root@camera:~# sudo rs-fw-update -s 203522250838 -f Signed_Image_UVC_5_14_0_0.bin 

Search for device with serial number: 203522250838

Updating device: 
Name: Intel RealSense D455, serial number: 203522250838, update serial number: 210523066031, firmware version: 05.14.00.00, USB type: 3.2
 26/01 18:13:52,589 ERROR [281473524995136] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error
 26/01 18:13:52,590 WARNING [281473524995136] (ds5-device.cpp:179) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error

Failed to locate a device in FW update mode
root@camera:~# rs-enumerate-devices 
Device info: 
    Name                          : 	Intel RealSense D4XX Recovery
    Physical Port                 : 	2-3.2-19
    Product Id                    : 	0ADB
    Product Line                  : 	D400
    Firmware Update Id            : 	210523066031

root@invisai-cam-bdqh:~# rs-fw-update -r -f dev/realsense_calibration/Signed_Image_UVC_5_14_0_0.bin 

Update to FW: dev/realsense_calibration/Signed_Image_UVC_5_14_0_0.bin

Recovering device: 
Name: Intel RealSense D4XX Recovery, serial number: unknown, update serial number: 210523066031, firmware version: unknown, USB type: unknown

Firmware update started

Firmware update progress: 100[%]

Firmware update done
Waiting for new device...
... timed out!
root@invisai-cam-bdqh:~# rs-enumerate-devices 
Device info: 
    Name                          : 	Intel RealSense D455
    Serial Number                 : 	203522250838
    Firmware Version              : 	05.14.00.00
    Recommended Firmware Version  : 	05.14.00.00
    Physical Port                 : 	/sys/devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B5C
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	210523066031
    Firmware Update Id            : 	210523066031
```

kumar-akshay324, 2023-01-26:
Addiitonal Information: This is with the realsense SDK installed inside a Docker image built on top of the `nvcr.io/nvidia/l4t-base:r35.1.0` base image

MartyG-RealSense, 2023-01-27:
Hi @kumar-akshay324  A firmware update typically involves resetting the camera at around the 30% stage in the update process, which disconnects the camera.  The reset then attempts to reconnect to the camera.  Occasionally though the camera may not be re-detected after the disconnection, preventing the firmware update process from being completed.

Can you confirm whether you are performing the firmware update with the official 1 meter USB cable supplied with the camera or with a longer USB cable of your own choice, please?  Thanks!

kumar-akshay324, 2023-01-27:
That is good to know about the reconnection issue. Thanks for sharing @MartyG-RealSense. Is that behavior different in the recovery state firmware update? I am using a 1-foot high-quality USBC cable that has worked on another version of the camera and NOT the official 1 meter cable.

MartyG-RealSense, 2023-01-27:
I do not have information about the mechanics of the firmware recovery.

If the cable is high quality then it should be sufficient.  High quality is important because of the volume of data that RealSense cameras can transmit, as a cable that is low-grade or designed for device charging rather than data transfer could cause degraded performance or disconnections.  It is also important that your choice of cable is a USB 3 cable and not a USB 2 one.

If you have access to the **RealSense Viewer** tool then instead of using rs-fw-update, you can alternatively update the firmware from the **Update Firmware** option in the drop-down menu of the **More** button at the top of the Viewer's options side-panel.

[image]

kumar-akshay324, 2023-01-27:
Sorry, I am running a Jetson in a headless state. Does the GUI follow a different strategy to flash the firmware?

Also, can we get more info on how does data flow speed or volume different between recovery state firmware update and regular serial number specified update?

MartyG-RealSense, 2023-01-28:
My understanding is that the GUI of the RealSense Viewer uses the same RealSense SDK C++ programming instructions to control the firmware update process that rs-fw-update does.

It is possible to write your own firmware updating script as an alternative to using rs-fw-update or the Viewer's mechanism.

At https://github.com/IntelRealSense/librealsense/issues/4416#issuecomment-511053250 a RealSense team member describes the workflow of rs-fw-update, which can be used as a reference to create your own update script and there is another programming reference at https://github.com/IntelRealSense/librealsense/issues/7095

Examining the source code of rs-fw-update, there is not a noticable difference between how a normal and recovery update is carried out that accounts for the update success that you have in recovery mode.    A couple of RealSense users have noted that it may take several attempts with the -r recovery flag for the recovery process to complete successfully.

https://github.com/IntelRealSense/librealsense/issues/9395 further discusses the firmware recovery process.

kumar-akshay324, 2023-01-31:
So the CLI tool kept crashing but the C++ interface was simple enough to break apart the rs-fw-update C++ tool script and build my own which was simple to use and worked just fine.

My hunch says that re-using the same device pointer after moving the device to the update state after calling as `<rs2::updateable>` is somehow making the connection go bad.

closed 2023-01-31 by kumar-akshay324

MartyG-RealSense, 2023-01-31:
It's excellent to hear that you achieved a solution, @kumar-akshay324 - thanks very much for the update!

avizipi, 2023-02-27:
@kumar-akshay324 
can you please share your code?

wernerdresden, 2023-07-19:
Had a similar issue while trying to update.the firmware of a D415 cam (from 5.13.0.50 to 5.15.0.2).
Using the windows executable "Intel.RealSense.FW.Update.exe" (version 2.54.1) was successfully.

MartyG-RealSense, 2023-07-19:
Thanks so much @wernerdresden for sharing what worked for you in your own situation!