realsenseai/librealsense #11365: Enabling IMU Stream stops frames from being received in the RGB Stream

GSNCodes, 2023-01-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   {Linux - Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-58-generic                                   | 
| Platform                                 | PC/  |
| SDK Version                            |  2.53                        | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

Enabling IMU Stream stops frames from being received in the RGB Stream (As well as Depth Steam). I want to record both IMU and RGB Data. Once we stop the motion module/ IMU Stream we get the image frames again in a while. Any help would be greatly appreciated !

MartyG-RealSense, 2023-01-27:
Hi @GSNCodes  Does the issue still occur if you only enable RGB and Motion Module, and do not enable Depth?   The RealSense Viewer tool has a known issue where enabling three streams simultaneously can cause No Frames Received on one of the streams if one of those enabled streams is the Motion Module.  

There is no solution for this in the Viewer except to either not enable the Motion Module or to be careful to only enable two streams simultaneously (RGB and Motion Module, or Depth and Motion Module).

GSNCodes, 2023-01-27:
Hi @MartyG-RealSense  Thank you for the quick response. No Frames are received when I enable only the RGB and Motion Module (As shown in the screenshot attached). I have not enabled the Depth module. Is there a fix or workaround to get data from both the RGB and Motion modules?

MartyG-RealSense, 2023-01-27:
Thank you for the clarification that you have not enabled Depth.  I wanted to make sure that all three streams had not been enabled at some point and then Depth disabled, leaving RGB and Motion Module enabled but the No Frames Received problem still persisting.

I note that the USB connection is shown as **2.1** beside the camera name at the top of the Viewer's options side-panel, indicating that the camera is being detected as being on a slow USB 2 connection and so cannot access its full capabilities.  Can you confirm please that the camera is plugged into a USB 3 port on your computer.  Also, are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice?

GSNCodes, 2023-01-28:
Hi @MartyG-RealSense  The problem persists and I have not enabled all the three streams. The camera is indeed plugged into a USB 3 port and I am using the official cable.

MartyG-RealSense, 2023-01-29:
What method did you use to install the librealsense SDK, please?

1.  From Debian packages.

2.  From source code with CMake, patching the kernel with patch script **patch-realsense-ubuntu-lts-hwe.sh** to provide support for kernel 5.15 (the patch script **patch-realsense-ubuntu-lts.sh** is not suitable for kernel 5.15).

3.  From source code with CMake using the **RSUSB backend** installation method, which bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require kernel patching.

MartyG-RealSense, 2023-02-04:
Hi @GSNCodes  Do you require further assistance with this case, please?  Thanks!

GSNCodes, 2023-02-04:
Hi @MartyG-RealSense 
I used the second method "From source code with CMake, patching the kernel with patch script patch-realsense-ubuntu-lts-hwe.sh to provide support for kernel 5.15 (the patch script patch-realsense-ubuntu-lts.sh is not suitable for kernel 5.15)."

MartyG-RealSense, 2023-02-05:
What computer or computing device are you using Ubuntu 20,04 on, please (for example a Raspberry Pi or a desktop / laptop PC).  If it is a Raspberry Pi board then they also have known problems related to streaming the IMU, as described at https://github.com/IntelRealSense/librealsense/issues/11089

GSNCodes, 2023-02-06:
Hi @MartyG-RealSense 
I'm using a Ubuntu 20.04 laptop.

MartyG-RealSense, 2023-02-06:
Is the laptop that 20.04 is running on a modern machine or an old one?

closed 2023-02-18 by MartyG-RealSense