realsenseai/librealsense #11366: Read access violation in device.cpp

Marenix, 2023-01-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 5.14.0.0 | 
| Operating System & Version |   Win (8.1/10) / Linux (Ubuntu 18/20/22)  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |           2.53.1                 | 
| Language                            |                       | 
| Segment			|                  | 

Hi, 
with the latest realsense library, sometimes we receive an Exception: "read access violation" in device.cpp, line 181, std::lock_guard<std::mutex> lock(_device_changed_mtx).

With a little analysis we think that there is an design issue in constructor of the "device" class (file device.cpp). The issue is that the lambda expression is created  and registered to the librealsense::context in the "device" class constructor. This lambda expression is also accessing member variables of the "device" class.

There is a two issues with this approach: 
1) Immediately after lambda expression is registered to the librealsense::context it can be called, although creation of "device" class object is not completed yet.
2) Second, de-registration of a lambda expression is performed in destructor of the "device" class. This mean that when "device" object is destroyed there is a window in which lambda expression can still be called (for example from some other thread) and then access to a non-existing member variables will occur which will lead  to an Exception: "read access violation".

How to reproduce:

1. Start the rs2::pipeline configured with some serial number of a camera while another camera is connected to the computer.
2. Connect the second camera as the pipeline.start(cfg) runs and looks for the device with the given serial number.
- Explaination: callback from the device watcher is called that the new device connected and notifies the device object that might not exist as it is repeatedly created and destroyed in the process of looking for a device with the given serial number.
3. If timed correctly the exception occurs

[image]

MartyG-RealSense, 2023-01-27:
Hi @Marenix  Thank you for your question.  I will highlight it to my Intel RealSense colleagues.  Thanks very much for your patience.

MartyG-RealSense, 2023-02-10:
Hi again @Marenix  My Intel RealSense collagues have been able to reproduce the 'read access violation in device.cpp' error and have created an official internal report to investigate the issue further.  

I have added an **Enhancement* label to this case to indicate that it should be kept open whilst the internal report is active, but you do not need to take further action.  Thanks again for highlighting this error!

Marenix, 2023-02-13:
Thanks for letting me know.

MartyG-RealSense, 2026-10-29:
Case closed due to the significant amount of time that has passed since the original bug report and no active bug investigation.

closed 2026-10-29 by MartyG-RealSense