realsenseai/librealsense #11376: Jetson Nano use D435i with CUDA

thli96, 2023-01-30:
| Required Info                         |Get the image from the camera with CUDA |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                         | 
| Platform                                 | Jetson Nano  |
| SDK Version                            |   legacy / 2.53.1                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description

Hello, I have compiled the SDK with CUDA, but no matter whether I view the rgb image in "realsense-viewer" or use the pipeline to obtain the image in python, there were a large CPU-Usage. Obviously CUDA is not worked, when I get the image from D435i. So I would like to know, how can I use the CUDA , when I get the Image from the camera. Thanks !

MartyG-RealSense, 2023-01-30:
Hi @thli96  The SDK's CUDA support provides automatic graphics acceleration for three specific types of operation: pointclouds, depth-color alignment and YUY to RGB color conversion.  If a program or script is not using one of these three types of processing then it will not benefit from CUDA acceleration.

thli96, 2023-01-30:
Thanks for your answer. @MartyG-RealSense . That meas, when I read the RGB-Image directly from camera, the CUDA will not work for it. CUDA only works if I use format conversion (YUY-RGB). 

I have tried the depth-color alignment with the example (rs-align) and the pointclouds (rs-pointcloud). But there is still a high cpu usage rate, close to 100%.

MartyG-RealSense, 2023-01-30:
When you compiled the SDK, did you build it from source code with CMake and include the CMake build flag **-DBUILD_WITH_CUDA=true** in the CMake build instruction to enable CUDA support in the SDK?

thli96, 2023-01-31:
Yes, I compiled the SDK with the build flag -DBUILD_WITH_CUDA=true.

MartyG-RealSense, 2023-01-31:
Which JetPack version are you using, please?

thli96, 2023-01-31:
The JetPack version is 4.6.1(L4T 32.7.1)

MartyG-RealSense, 2023-01-31:
JetPack 4.6 is officially supported by librealsense version 2.51.1, so the problem is unlikely to be related to the JetPack version.  Thanks very much for the confirmation of the JetPack version.

Did you build the SDK from source with a patched kernel or did you build using the RSUSB Backend method that does not require kernel patching?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-rsusb-backend

thli96, 2023-02-01:
Thanks for your answer. I can rebuild the right librealsense version firstly, then I can try to run it again. 

I build SDK from source using the RSUSB Backend method. I tried to build SDK from source. When I want to patch the kernel occurred an error: Unsupported Jetpack revision 32.7.1 aborting script.

MartyG-RealSense, 2023-02-01:
An RSUSB build does not need to be kernel patched.

thli96, 2023-02-01:
I want to install the librealsense version 2.51.1 from source. Before the build I run the script : `./scripts/patch-realsense-ubuntu-L4T.sh  `

But the error is same:  Unsupported Jetpack revision 32.7.1 aborting script. So should I change the kernel version？

MartyG-RealSense, 2023-02-01:
The most recent JetPack version that is supported in librealsense is 4.6, which is 32.6.1.  That support was added in librealsense 2.50.0, which you have installed.

Official support for JetPack 5.0.2 will be in the next release of librealsense, likely numbered 2.54.0.

You can bypass the kernel and not have to use a patch script if you build from source code with the RSUSB method by including the build flag **-DFORCE_RSUSB_BACKEND=ON** in the CMake build instruction.

thli96, 2023-02-01:
Thanks for your help. I have another question. I would like to use pyrealsense2 on python2 and python3. So is the right configuration for Cmake？
`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6  - DPYTHON_EXECUTABLE=/usr/bin/python  -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`

MartyG-RealSense, 2023-02-01:
In **-DPYTHON_EXECUTABLE=/usr/bin/python3.6** you should set the Python 3 number to the version that you are using.  So if you are using Python 3.6, use python3.6.  If it is Python 3.9, use python3.9, etc.

It is rare for RealSense users to use both Python 2 and Python 3 on the same computer, with most choosing Python 3 only.  Usually the compilation of the Python wrapper will be done for one version or another (2 or 3) but not both.

[image]

I cannot recall a past case where a RealSense user has attempted installing for both Python 2 and Python 3.  If it were possible though then I would expect that two separate compilations of the wrapper would be performed, one with the Python 2 path to compile for Python 2.7, and another compilation that points to the Python 3 path (not two DPYTHON_EXECUTABLE paths in the same CMake build instruction).

thli96, 2023-02-01:
> 

Thanks for your help! So I should compile two times. When I compile the second time for python 3.6. I just can enter this command? 
`cmake .. / -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 ` 

I don't need to enter `-DFORCE_RSUSB_BACKEND=ON` etc. one more time?

thli96, 2023-02-01:
> The **Note** in the image relates to installing the Python wrapper **only** after librealsense has already been installed, so that you are not installing librealsense twice.
> 
> So I would think that a process for supporting both Python 2.7 and Python 3 might be:
> 
>     1. Install both librealsense and the Python wrapper for Python 3.6 at the same time using:
> 
> 
> `cmake .. / -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6`
> 
>     2. Compile the Python wrapper for 2.7 on its own using the instructions at the link below.
> 
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source
> 
> `cmake .. / -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python2.7`
> 
> My recommendation though would be to only compile for Python 3.6 if possible, or compile for Python 2.7 on a separate computer if you have one. This is because there is not a precedent for compiling the pyrealsense2 wrapper for both Python 2.7 and 3.6 on the same computer and so there is not an established procedure for doing so or what the consequences might be.

Sorry, I didn't all understand. I have just complied and installed the librealsense and at the same time I compiled the Python Wrapper for python3.6. For python 2.7 should I create in a "build" dictionary  `/librealsense/wrappers/python` and in this dictionary compile or enter the `cmake .. / -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python2.7` in `/librealsense/build`. 

Now I used the librealsense version 2.51.1. But When I used the rs-pointcloud and rs-align, there are also high usage of CPU. I don't know whether these Processes run with CUDA :(

thli96, 2023-02-02:
Thanks, I tried to use GL example. But it displays that Process a picture in one second. Maybe the GPU is not so powerful.

MartyG-RealSense, 2023-02-02:
Intel recommend enabling the **barrel jack** connector on the Nano for extra power if your model of Nano has one.  Instructions for doing so are at the link below.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

thli96, 2023-02-03:
My Nano doesn't have this. Yesterday I tried to use C++ read the color stream and display by using OpenCV.  But I got the RGB-Image was too bad.was blurred. 

[image]

MartyG-RealSense, 2023-02-03:
Hi @thli96  The RGB image does not look correct.  It is meant to be a single image and not multiple image frames that are duplicated.

There is a RealSense C++ OpenCV tutorial for reading the color frame at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md

thli96, 2023-02-08:
Thanks a lot. The Problem is solved. I found it was my SDK not right installed. 
I just want to know, whether the y16 format can be used as a reliable graphics format. Because I found that the transmission of rgb format takes a lot of time.

MartyG-RealSense, 2023-02-08:
I'm pleased to hear that you achieved a solution!

Y16 RGB is a monochrome image rather than full RGB color.  Y16 is not typically used by RealSense users, with formats such as RGB8 being preferred.  There is nothing wrong with Y16 RGB though so long as you do not mind it being monochrome.

[image]

Alternatively, https://github.com/IntelRealSense/librealsense/issues/7275 discusses using OpenCV to access the camera's raw original RGB frames and convert them to a format that is usable by the RealSense SDK.

thli96, 2023-02-09:
Thanks for your help. I need a grayscale image, but can the image in Raw16 format be directly used as a grayscale image? I think they are all single channel image.

MartyG-RealSense, 2023-02-09:
You could perhaps specify a monochrome RGB format to convery the raw color image to with **cv2.cvtColor**, such as BGR2GRAY instead of BGR2RGB

A list of OpenCV color-space types is provided at the link below.

https://docs.opencv.org/3.4/d8/d01/group__imgproc__color__conversions.html

MartyG-RealSense, 2023-02-16:
Hi @thli96  Do you require further assistance with this case, please?  Thanks!

thli96, 2023-02-16:
Sorry for lately reply. It works for me. Thanks for your help!

MartyG-RealSense, 2023-02-16:
It's excellent to hear that you are successful, @thli96 - thanks very much for the update!  As you have achieved a solution, I will close this case.  Thanks again!

closed 2023-02-16 by MartyG-RealSense