realsenseai/librealsense #11379: Camera's data doesn't update but still keeps sending

MichaelPan0905, 2023-01-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435f | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.140                                         | 
| Platform                                 |  NVIDIA Jetson AGX Xavier  |
| SDK Version                            |  legacy / 2.53.1                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
When I'm running the camera on my robot, the camera's data suddenly stop updating, both occured on rgb data and pointscloud data(they stopped at the same moment), but the camera kept sending the old data to the robot.After restarted the camera, it recovered to normal state.
here is the pcl data which is not the newest：
[image]
and this is the log printed on the terminal:
[image]

and this is the actual pcl data of the object (showed normally after restarted the camera):
[image]

I want to ask how is this happended and how to avoid or observe it.
Thanks.

MartyG-RealSense, 2023-01-31:
Hi @MichaelPan0905  If you are using librealsense version **2.53.1** then please first try updating your camera firmware driver from 5.13.0.50 to 5.14.0.0, as 2.14.0.0 is the recommended firmware for use with 2.53.1.

MichaelPan0905, 2023-02-01:
> 
Sorry, but I made a mistake that the sdk version is 2.50.0, not 2.53.1.

MartyG-RealSense, 2023-02-01:
Hi @MichaelPan0905  I looked carefully at your problem.  Thanks very much for your patience.

Firmware 5.13.0.50 is the correct version to use with librealsense SDK version 2.50.0.

It does not seem as though new frames have stopped arriving.  If the connection had timed-out (which it does if no new frames arrive after 5 seconds) then the program would likely crash and exit with an error message such as **RuntimeError: Frame didn't arrive within 5000**

It also does not seem likely that the stream stays the same because it is repeating previous frames.  If that were true then the image would likely stay the same for only 2 to 4 frames before recovering and displaying a new frame.

As the error log references the CUDA depth-color alignment process, it may be worth trying to build librealsense from source code with CMake but leave out the **-DBUILD_WITH_CUDA** build flag so that CUDA support for acceleration of depth-color alignment operations is not included in the SDK build.  This would help to eliminate the possibility that the problem is being caused by something going wrong in CUDA.

MichaelPan0905, 2023-02-06:
@MartyG-RealSense Thanks for responding.
The reason I asked for help days before is that I thought maybe there is someone met the same problem with mine or you've done with similar situation. This problem only happened once(which caused our rob crashed with other objects), and I havn't repeated it yet.
And for the CUDA tools, we need it to deal with some picture things so that we can't leave it. Do you mean that this problem was caused by cuda?

MartyG-RealSense, 2023-02-06:
There have been a very small number of cases where a problem has been found to be caused by librealsense's CUDA support and not enabling CUDA in the librealsense build was the solution for that problem.  Omitting that support means that alignment would still take place but a higher processing load may be placed on the CPU, as CUDA enables processing of alignment and pointclouds to be offloaded from the CPU onto the graphics GPU.

MichaelPan0905, 2023-02-06:
@MartyG-RealSense  Thanks for your explanation. 
So according to your words that this problem is caused by librealsense's CUDA support, could there be some official solutions to fix it without just not enabling CUDA crudely?

MartyG-RealSense, 2023-02-06:
The issue being related to the CUDA support is a possibility rather than a certainty, and excluding CUDA support would be the only way to test this possibility.  If the problem has only occurred once then it may never happen again though, making it difficult to conclude whether CUDA really was a cause.

Givwn that it has been a one-time occurrence, the problem could also have been related to a temporary glitch on the computer hardware or the Ubuntu operating system and not the camera or the librealsense SDK.

Until the problem occurs again, the best course of action may be to continue using your CUDA-enabled setup, as your project needs CUDA.  Bear in mind that librealsense's CUDA support only affects librealsense image alignment and pointcloud generation.  If there is a non-RealSense part of your project that makes use of CUDA, excluding CUDA support from librealsense will not prevent those other non-RealSense components from working as you are not removing CUDA itself from your Jetson.

MichaelPan0905, 2023-02-06:
Thank for your patient responding.

closed 2023-02-18 by MartyG-RealSense