realsenseai/librealsense #11380: ARM ubuntu20 distribution

oscarthorn, 2023-01-31:
Hi!

It seems there are no prebuilt binaries for ubuntu20 ARM. Is there a timeline for providing those? With Nvidia moving Jetson devices to ubuntu 20 with jetpack 5 it seems like a lot of people will be needing binaries for ubuntu 20 and arm.

MartyG-RealSense, 2023-01-31:
Hi @oscarthorn  Intel will unfortunately not be providing Arm binaries for Ubuntu 20.04 and the librealsense SDK must instead be compiled from source code when installing for 20.04 on an Arm device.

However, librealsense can currently work unofficially with JetPack 5, and official support for JetPack 5.0.2 is being introduced in the next SDK version (likely numbered 2.54.0).

oscarthorn, 2023-01-31:
May I ask why? How can it be officially supported if you don't intend to provide binaries? If it will be officially supported it stands to reason that you have tried it out so why not just upload the deb files?

oscarthorn, 2023-01-31:
Okey, thanks anyway!

closed 2023-01-31 by oscarthorn