realsenseai/librealsense #11386: rs2::pipeline pipe Infinity loop

Didi3333, 2023-02-01:
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  {2.53.1 }                          | 
| Language                            |  {C++ Into DLL }                          |  

### Issue Description
I link realsens2.ddl to my C++ DLL.

When my program charge all headers of the DLL it freez at line: 

rs2::pipeline pipe;

All others declaration works well if i comment the pipeline declaration: 

rs2::pipeline_profile profile;
rs2::frameset frames;
rs2::frameset aligned;
rs2::spatial_filter spat;
rs2::frameset frameTemp;

It's not a crash it's an infinity loop

I can detect my camera : 

rs2::context ctx;
int size = ctx.query_devices().size(); => Return well 1

MartyG-RealSense, 2023-02-01:
Hi @Didi3333  Of the rs2:: instructions that you have listed, usually the only one that would be placed before a pipeline start instruction is **rs2::pipeline pipe;** and **rs2::context ctx;** and the rest would be placed on or after the pipeline start line.  

The spatial filter and alignment definitions would be placed before a wait_for_frames instruction.

For example:

```
rs2::context ctx;
int size = ctx.query_devices().size(); => Return well 1

rs2::pipeline pipe;
rs2::pipeline_profile profile = pipe.start();

rs2::spatial_filter spat;

rs2::frameset frames = pipe.wait_for_frames();
```

************

A simple introductory C++ test script can be found at the link below.

https://github.com/IntelRealSense/librealsense#ready-to-hack

Didi3333, 2023-02-02:
I know, 

My issue here is if i build my dll with only the line:

 rs2::pipeline pipe

When it is loaded by my program that freez with no error 

I will try with a previous version of the SDK. 

I already succeed to use the SDK few years ago

MartyG-RealSense, 2023-02-02:
Does the problem also occur with the Ready to Hack example that I linked to above, or only in your own script please?

Didi3333, 2023-02-02:
@MartyG-RealSense  yes at the first line : 

// Create a Pipeline - this serves as a top-level API for streaming and processing frames
rs2::pipeline p;

MartyG-RealSense, 2023-02-02:
My understanding is that you are creating your own RealSense project in Visual Studio Enterprise 2022 on Windows and linking it to the realsense2.dll library file.  Is this correct, please?

A typical approach to RealSense C++ development in Visual Studio would be to link your project to three **.props** files provided by the RealSense SDK, as described in the tutorial at the link below.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

Didi3333, 2023-02-02:
Indeed that Works : 

std::shared_ptr<rs2::pipeline> pipe;
pipe = std::make_shared<rs2::pipeline>();

Thank you

MartyG-RealSense, 2023-02-02:
That's great to hear, @Didi3333 - thanks very much for the update!

closed 2023-02-10 by MartyG-RealSense