realsenseai/librealsense #11387: ModuleNotFoundError: No module named 'pyrealsense2 by python script in startup (jetson nano)

PhenonGH, 2023-02-02:
# ModuleNotFoundError: No module named 'pyrealsense2 by python script in startup (jetson nano)
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |  Ubuntu 18.04   |                                 | 
|Kernel Version (Linux Only)	|4.9.253-tegra|
|SDK Version|	2.53.1|
|Firmware Version	                 | 05.13.00.50  |
| Platform                                 | NVIDIA Jetson Nano  |                          | 
| Language                            |  python 3.6.9                           | 
| Segment			|  Robot                           | 

Hello, I'm having a problem with the pyrealsense2 module. When my Python script is called normally, there are no problems. However, I wanted to build a sort of robot using the Jetson Nano and the camera, so I wanted to put my script into the autostart. Unfortunately, I now always receive the error message: ModuleNotFoundError: No module named 'pyrealsense2'. Additionally, I have discovered that if I run the script with sudo rights, I receive the same error message, but that has never been a real problem for me. I would be very grateful for help, as I am still relatively new to the Linux world.

For the camera installation, I used this: https://github.com/35selim/RealSense-Jetson/blob/main/build_pyrealsense2_and_SDK.sh.

autostart:

[Desktop Entry]
Type=Application
Exec=python3 /home/jetson/My/main.py
Hidden=false
NoDisplay=false
Terminal=true
X-GNOME-Autostart-enabled=true
X-GNOME-Autostart-delay=100

Name[en_US]=v2
Name=v2
Type=Application
Comment[en_US]=

MartyG-RealSense, 2023-02-02:
Hi @PhenonGH  There was a past case at https://github.com/IntelRealSense/librealsense/issues/7861#issuecomment-734775839 that used autostart and they defined their Exec path like this:

`[Desktop Entry]
Type=Application
Name=Realsense
Exec=/usr/bin/python3 /home/pi/counter.py`

Does your error still occur if you change your Exec path from **python3** to **/usr/bin/python3**

The Python 3 path on a particular computer can be found by using the command **which python3**

PhenonGH, 2023-02-02:
> 

Thank you for your quick response, unfortunately your solution was not effective and the same error still appears.

MartyG-RealSense, 2023-02-02:
The steps in your **build_pyrealsense2_and_SDK.sh** build script resemble the list of instructions in the guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 for installing librealsense and pyrealsense2 together on Jetson Nano.  I cannot see a problem in your script's sequence of commands.

The error **ModuleNotFoundError: No module named 'pyrealsense2'** may occur if the program cannot find the pyrealsense2 files for some reason.  A workaround for this is to place two pyrealsense2 **.so** files in the same folder location as your project script so that they do not need to be located via the PYTHONPATH.  The two files to place next to your script are **librealsense2.so** and **pyrealsense2.so**.  

When the pyrealsense2 wrapper is built for Python 3, the pyrealsense2.so file will be generated with a complex filename based around your Python 3 version number, such as pyrealsense2.cpython-36m-arm-linux-gnueabihf.so

Aside from https://github.com/IntelRealSense/librealsense/issues/7861#issuecomment-734775839 the pyrealsense2 wrapper has not been used with autostart, so I would recommend using the wrapper in the conventional way if possible.

PhenonGH, 2023-02-02:
Ok it works.

Workaround:
I have copied these files:

pybackend2.cpython-36m-aarch64-linux-gnu.so.2
pybackend2.cpython-36m-aarch64-linux-gnu.so.2.51.1
pyrealsense2.cpython-36m-aarch64-linux-gnu.so
pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.51
pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.51.1

from librealsense>build>wrappers>python

into my program folder. I don't know if all the files or only some of them are required, I will try it out sometime.

Many thanks without your help I probably wouldn't have come up with this solution.

The topic can be closed.

closed 2023-02-02 by PhenonGH

MartyG-RealSense, 2023-02-02:
You are very welcome, @PhenonGH - it's excellent news that copying the .so files worked for you.  Thanks very much for the update!

TheMuhammadRafay, 2024-08-28:
@MartyG-RealSense I am using the Librealsense SDK 2.55.1

MartyG-RealSense, 2024-08-28:
Please try moving the **pyrealsense2.cpython-36m-arm-linux-gnueabihf.so** file to the same folder that your pyrealsense2 project script is in if you have not done so already, and also move the file **librealsense2.so** to that folder.

If you cannot find both of these .so files then please check for them in the **build > wrappers > Python** directory of your librealsense2 source code folder.