realsenseai/librealsense #11391: Error : Unsupported Jetpack revision 32.1 aborting script

MikioEguchi, 2023-02-03:
* I tryed to recognizing realsense on Jestson TX2 with J121, but it didn't work. 
* The devices used were as follows.
    * Jetson TX2 (installed Jetpack4.2)
    * J121 38350(carrier board)
* Also, Here is what I did and the error statement.
    * Upgrade from Ubuntu 18.04 to 20.04, but there are some error. (Sorry, I didn't leave the error message.)
    * The following command was entered.

```
cd ~/Downloads
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
./scripts/libuvc_installation.sh
./scripts/patch-realsense-ubuntu-L4T.sh
```

The following error was output.

```
Jetson Board (proc/device-tree/model): quill
Jetson L4T version: 32.1
Unsupported Jetpack revision 32.1 aborting script
```

I ignored this error and typed the following command and got the following error.

```
sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev -y
./scripts/setup_udev_rules.sh
mkdir build && cd build
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install
```

The following error was output.

[image]
[image]
[image]

Please let me know how to resolve this.

MartyG-RealSense, 2023-02-03:
Hi @MikioEguchi  When installing for Jetson from source code and applying the patch script **patch-realsense-ubuntu-L4t.sh**, at the time of writing this the patch script supports L4T versions ranging from a minimum of **32.2.1** (4.2.2) to a maximum of 32.6.1 (4.6).  You are using 32.1 (4.2) and so may need to upgrade your JetPack version slightly to at least 32.2.1 / 4.2.2 in order to apply the patch script successfully.

In the next SDK version (which should be numbered 2.54.0), JetPack 5.0.2 will be supported.

RealSense users are typically using JetPack 5 with Ubuntu 20.04, as it is the first JetPack version designed by Nvidia for use with 20.04 whilst previous JetPack versions were designed for use with Ubuntu 18.04.  

Although JetPack 5 is not officially supported until the next SDK version, it can be used unofficially with 20.04 if the kernel patch script is not applied or if the SDK is built with the **RSUSB backend** installation method (which is not dependent on Linux versions or kernel versions and does not require kernel patching).

MartyG-RealSense, 2023-02-10:
Hi @MikioEguchi  Do you require further assistance with this case, please?  Thanks!

MikioEguchi, 2023-02-12:
Hi @MartyG-RealSense Thanks for reply! Sorry for the late reply, I had done another work. I can understand what you say, so I try to update Jetpack. Thanks!

MartyG-RealSense, 2023-02-12:
Thanks very much @MikioEguchi for the update.  I look forward to your next report.  Good luck!

MartyG-RealSense, 2023-02-23:
Hi @MikioEguchi  Do you have an update about this case that you can provide, please?  Thanks!

MikioEguchi, 2023-03-01:
Hi @MartyG-RealSense Sorry for the late reply. I try to recognizing Realsense on Jestson TX2 with J120 installed Jetpack 4.3. I succeeded in that. Thanks to your advice the problem is solved!

MartyG-RealSense, 2023-03-01:
That's excellent news, @MikioEguchi - thanks very much for the update about your success.  As you have achieved a solution, I will close this issue.  Thanks again!

closed 2023-03-01 by MartyG-RealSense