realsenseai/librealsense #11393: Permission issues when Updating udev rules Raspberry Pi 3B

ChanceHamilton59, 2023-02-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |N/A | 
| Operating System & Version |   Raspbian  | 
| Kernel Version (Linux Only)    |  5.15.84-v7+                                         | 
| Platform                                 | PC/Raspberry Pi 3B | 
| SDK Version                            |  Downloaded today  | 
| Language                            |  python3                          | 

### Issue Description
Hello,

I am following the instructions outlined in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md 
According to the read me, all my system requirements are met. My issue happens when I execute the 

`$ sudo udevadm control --reload-rules && udevadm trigger `

This fails to complete due to Failed to write due to permissions denied. Exact errors provided below (sorry for the long post)

Any suggestions?

`Failed to write 'change' to '/sys/devices/armv7_cortex_a7/uevent': Permission denied
Failed to write 'change' to '/sys/devices/breakpoint/uevent': Permission denied
Failed to write 'change' to '/sys/devices/gpiochip1/uevent': Permission denied
Failed to write 'change' to '/sys/devices/gpiochip1/gpio/gpiochip100/uevent': Permission denied
Failed to write 'change' to '/sys/devices/kprobe/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/Fixed MDIO bus.0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/Fixed MDIO bus.0/mdio_bus/fixed-0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/arm-pmu/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/cam1_regulator/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/cam1_regulator/regulator/regulator.1/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/cam_dummy_reg/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/cam_dummy_reg/regulator/regulator.2/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/cpufreq-dt/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/fixedregulator_3v3/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/fixedregulator_3v3/regulator/regulator.3/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/fixedregulator_5v0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/fixedregulator_5v0/regulator/regulator.4/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/kgdboc/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/leds/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/leds/leds/led0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/leds/leds/led1/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/phy/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/reg-dummy/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/reg-dummy/regulator/regulator.0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/regulatory.0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/serial8250/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan1/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan2/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan3/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan4/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan5/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan6/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan7/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan8/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f007000.dma/dma/dma0chan9/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/bcm2835-camera/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/media2/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video10/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video11/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video12/uevent': Permission denied
Failed to write 'change' to '/sys/devices/platform/soc/3f00b840.mailbox/bcm2835-codec/video4linux/video18/uevent': Permission denied
[... 300 lines cut ...]
Failed to write 'change' to '/sys/devices/virtual/vc/vcs5/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcs6/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcs7/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa1/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa2/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa3/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa4/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa5/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa6/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsa7/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu1/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu2/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu3/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu4/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu5/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu6/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vc/vcsu7/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vtconsole/vtcon0/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/vtconsole/vtcon1/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/workqueue/blkcg_punt_bio/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/workqueue/iscsi_conn_cleanup/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/workqueue/iscsi_eh/uevent': Permission denied
Failed to write 'change' to '/sys/devices/virtual/workqueue/writeback/uevent': Permission denied`

MartyG-RealSense, 2023-02-04:
Hi @ChanceHamilton59  Are you using Raspbian **Bullseye** please?  Whilst this error has not previously been reported in relation to librealense, a similar error was reported by a couple of Bullseye users at the link below who had not had the same problem with the previous OS version Raspbian Buster.

https://stackoverflow.com/questions/72198150/docker-bullseye-image-and-udevadm-command-failing

Intel's Raspbian installation guide for RealSense has not been updated since 2020 and so Buster would have been the current Raspbian version at that time.

There have been previously reported problems with using librealsense with Raspbian Bullseye, as described at https://github.com/IntelRealSense/librealsense/issues/10033#issuecomment-1011206534

ChanceHamilton59, 2023-02-04:
> Hi @ChanceHamilton59  Are you using Raspbian **Bullseye** please?  Whilst this error has not previously been reported in relation to librealense, a similar error was reported by a couple of Bullseye [...]

Thank you for your quick response. While attempt this using the previous OS version.

ChanceHamilton59, 2023-02-07:
Your suggestion worked. I will close this issue. Thanks for your help.

closed 2023-02-07 by ChanceHamilton59

MartyG-RealSense, 2023-02-07:
It's great to hear that Buster worked for you, @ChanceHamilton59 - thanks very much for the update!