realsenseai/librealsense #11401: Realsense on wsl2 (Ubuntu 20.04)

juanmcc, 2023-02-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {WSL2 / Windows 11/ Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.15.74.2-microsoft-standard-WSL2                                    | 
| Platform                                 | PC  |
| SDK Version                            |  { SDK 2.0 }                          | 
| Language                            |  ROS2
| Segment			|  VR/AR  

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, I'm trying to use Realsense with WSL2 on Windows 11, but I'm not able to do this without sudo. 

The installation process that I followed is this:
https://robots.uc3m.es/installation-guides/install-realsense2.html

Because this process (https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) doesn't work for me.  I receive this after the process:

> Loading the modified modules into kernel...Current status:
> librealsense2-dkms, 1.3.19, 5.4.0-1094-aws, x86_64: installed
> librealsense2-dkms, 1.3.19, 5.4.0-137-generic, x86_64: installed
> modinfo: ERROR: Module alias videodev not found.
> mod: videodev
> modinfo: ERROR: Module alias uvcvideo not found.
> mod: uvcvideo
> modinfo: ERROR: Module alias hid_sensor_gyro_3d not found.
> mod: hid_sensor_gyro_3d
> modinfo: ERROR: Module alias hid_sensor_accel_3d not found.
> mod: hid_sensor_accel_3d

And the building process shows this:

> Presets deployment:
> no src dir
> tgt dir exists
> Presets deployment is skipped
> Setting up librealsense2-dkms (1.3.19-0ubuntu1) ...
> Loading new librealsense2-dkms-1.3.19 DKMS files...
> It is likely that 5.15.74.2-microsoft-standard-WSL2 belongs to a chroot's host
> Building for 5.4.0-1094-aws and 5.4.0-137-generic
> Building initial module for 5.4.0-1094-aws
> Done.

Anyway, sudo realsense-viewer works fine for me, but without sudo I receive this:

> 06/02 10:22:54,638 ERROR [139661380933376] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
>  06/02 10:22:54,704 ERROR [139661540656896] (sensor.cpp:664) acquire_power failed: failed to set power state
>  06/02 10:22:54,705 WARNING [139661540656896] (rs.cpp:310) null pointer passed for argument "device"
>  06/02 10:22:54,705 WARNING [139661540656896] (rs.cpp:2704) Couldn't refresh devices - failed to set power state

I need to run this without sudo because ros2 can't access to the camera neither, but I don't know if this is possible on WSL2. Thank you so much.

MartyG-RealSense, 2023-02-06:
Hi @juanmcc  RealSense users of WSL2 at https://github.com/IntelRealSense/librealsense/issues/10506 reported the same issue of being able to use realsense-viewer with sudo but not without it.  In that case, one of the users succeeded in Windows 11 by unplugging and re-inserting the camera and restarting WSL2.

juanmcc, 2023-02-06:
> Hi @juanmcc RealSense users of WSL2 at #10506 reported the same issue of being able to use realsense-viewer with sudo but not without it. In that case, one of the users succeeded in Windows 11 by unplugging and re-inserting the camera and restarting WSL2.

Thank you so much for your reply MartyG. I was trying you say, but sometimes works, sometimes doesn't.