realsenseai/librealsense #11403: Underwater Calibration using On-chip Calibration tool

TheNemo05, 2023-02-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  win11, ubuntu 20 | 
| Kernel Version (Linux Only)    |  NA  | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1  | 
| Language                            |  ROS/Python   | 
| Segment			|  NA     | 

### Issue Description
Is it possible to calibrate D455 using On-chip calibration tool. If not what methods are to be followed for underwater calibration. If possible kindly provide Python code.

MartyG-RealSense, 2023-02-06:
Hi @TheNemo05  Yes, the D455 camera model can be calibrated with On-Chip calibration.  It can calibrate both intrinsics and extrinsics, though as separate calibration operations rather than both at the same time.

The RealSense SDK has an official example Python program for On-Chip calibration at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/depth_auto_calibration_example.py

There is also an extensively detailed technical discussion about underwater calibration at https://github.com/IntelRealSense/realsense-ros/issues/1723 though the discussion is mainly focused on calibration with the **Dynamic Calibration** tool rather than On-Chip.

The discussion is so long that some of it is hidden under a cut, so I recommend first visiting https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-802741234 and left-clicking on the **Load More** link to display the entire discussion, then return to the start of it.

The Dynamic Calibration tool can be installed on Ubuntu by following the instructions on page 14 onwards of the PDF user guide document for the tool.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

There is also a pre-built executable Windows version of the tool that can be downloaded here:

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

Dynamic Calibration can calibrate both depth and RGB, though the normal version of the tool only calibrates extrinsics.  An **OEM** version of the tool that calibrates both intrinsics and extrinsics is provided with Intel's $1500 USD **OEM Target Calibration** product, though the free version of the tool that only calibrates extrinsics is sufficient for most RealSense users.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

TheNemo05, 2023-02-06:
Hello, can you please provide customized technique for customized calibration. A python code is also appreciated.

MartyG-RealSense, 2023-02-06:
If your preference is to use On-Chip calibration, a guide to the basics of accessing and controlling the calibration through Python is provided in the tool's user guide at the link below.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#appendix-c-on-chip-calibration-python-api

I believe that custom calibration would be required if calibrating whilst the camera is out of the water in the air, but if calibration is carried out whilst the camera is already underwater then the standard calibration process should be sufficient.

TheNemo05, 2023-02-07:
Also how to get Intrinsic parameters of depth image ?

MartyG-RealSense, 2023-02-07:
By standard calibration I mean that if the calibration is initiated when the camera is already underwater then you should not need to make any special changes to settings to compensate for the camera being underwater.

If you are using a custom camera housing with an acrylic transparent window then it is recommended that both sides of the window have an anti-reflective coating applied to them, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2566#issuecomment-431852571

Intrinsics can be retrieved in Python with the RealSense SDK instruction **get_intrinsics()** as demonstrated at https://github.com/IntelRealSense/librealsense/issues/3986

TheNemo05, 2023-02-09:
Hi according to the https://github.com/IntelRealSense/librealsense/issues/2566#issuecomment-431852571 they have not considered any refractive parameter & distance between camera & housing boundary the medium will also be different as we are using acrylic sheet & then sea water.

MartyG-RealSense, 2023-02-09:
So your custom housing is like the underwater one at https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-805760519 ?

[image]

In the underwater environment, the creators of that housing were experiencing **fisheye distortion** of the image.

[image]

A RealSense team member provided detailed feedback about that housing design at https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-808406783 in which they advised to perform a full calibration underwater if possible to correct for the fisheye issue, or if in-water calibration was not possible then calibrate in the air and correct the depth map in post-processing by adding an x-axis remapping of the depth.

TheNemo05, 2023-02-13:
As per what i ve understood, they used the customised chessboard tool underwater. https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-834374057 from this comment can you let me know if they are using the dynamic calibration tool or any specified algorithm.

MartyG-RealSense, 2023-02-13:
In https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-834374057 the Intel RealSense team members used the Dynamic Calibration tool.  

They created a smaller custom version of the OEM Calibration Target Board product and tested it in a fish tank.  If calibrating in a larger underwater environment such as a pool then they recommended using the actual full-size OEM Calibration Target Board.

TheNemo05, 2023-02-14:
i do have one more question while referring to above thread https://github.com/IntelRealSense/realsense-ros/issues/1723#issuecomment-805760519 what transparent material they are using in the front of housing. Is it glass / acrylic sheet ???

MartyG-RealSense, 2023-02-14:
The creator of the custom housing's transparent window do not state which material they are using.  The most important factors though are likely to be that the chosen material is non-reflective or has an anti-reflective coating applied on both sides.  Intel's cover material guidelines suggest acrylic (plexiglass) as an example of a cover material.

TheNemo05, 2023-02-17:
Hello 
https://studios.disneyresearch.com/wp-content/uploads/2019/03/Underwater-3D-Capture-using-a-Low-Cost-Commercial-Depth-Camera.pdf according to this research carried out by Disney, they have used Realsense camera but they no have not mentioned which model. They are indirectly changing some parameters for underwater operation in the camera default parameters. Can u please recommend some methods by which we can change the camera parameters as shown in this paper.

MartyG-RealSense, 2023-02-17:
Hi @TheNemo05  This paper was originally published in 2016, which pre-dates the launch of the RealSense 400 Series camera range at the end of 2017.  The paper makes multiple references to testing with a **structured light** RealSense camera.  This would make the tested model likely to be the **SR300** model as the other RealSense cameras at that time (R200 and ZR300) were stereo depth cameras, the predecessors of the 400 Series.

As the librealsense SDK is not mentioned in the paper and the researchers were also testing a Microsoft Kinect camera, it is likely that they were using the original **RealSense SDK for Windows** from 2014-2016, which is incompatible with librealsense.  So parameter changes that they made may not be directly applicable to the modern-day librealsense SDK, especially as they were configuring the parameters of a structured light camera (which is a different camera technology to the stereo depth of the 400 Series).

A modern equivalent to the linked-to project is **FishSense**, which uses the D455 model.

https://kastner.ucsd.edu/wp-content/uploads/2021/08/admin/oceans21fishsense.pdf

https://www.youtube.com/watch?v=ey7oqCiqm6o

MartyG-RealSense, 2023-03-01:
Case closed due to no further comments received.

closed 2023-03-01 by MartyG-RealSense

TheNemo05, 2023-03-06:
how can Fishsense be implemented with Realsense ?

MartyG-RealSense, 2023-03-06:
The **FishSense** project made use of the RealSense D455 camera.  

https://e4e.ucsd.edu/fishsense

https://e4e.ucsd.edu/fishsense/fishsense-prototype-testing

https://kastner.ucsd.edu/wp-content/uploads/2021/08/admin/oceans21fishsense.pdf