realsenseai/librealsense #11406: Incorrect distance display

Gaurav37, 2023-02-07:
Intel Realsense D435
Ubuntu 18.04
Python code 
I am using the tutorial on https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb and changed the code only to make it for realtime.

However, it displays wrong distance most of the times.
It displays 1.x(1.1,1.4,1.5 etc.) meter distance only. It rarely shows 2 meters or 3 meters even if I move further away.

MartyG-RealSense, 2023-02-07:
Hi @Gaurav37  Did you make the code realtime by removing the line `cfg.enable_device_from_file("../object_detection.bag")` please?

Gaurav37, 2023-02-07:
I'm sharing my full code below.
Its divided in two files. The first file is realsense_depth.py which is imported into the main function file.
##########################main function code#####################
import cv2                                # state of the art computer vision algorithms library
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
import pyrealsense2 as rs                 # Intel RealSense cross-platform open-source API
from realsense_depth import *
print("Environment Ready")

net = cv2.dnn.readNetFromCaffe("MobileNetSSD_deploy.prototxt", "MobileNetSSD_deploy.caffemodel")
inScaleFactor = 0.007843
meanVal       = 127.53
classNames = ("background", "aeroplane", "bicycle", "bird", "boat",
              "bottle", "bus", "car", "cat", "chair",
              "cow", "diningtable", "dog", "horse",
              "motorbike", "person", "pottedplant",
              "sheep", "sofa", "train", "tvmonitor")

dc = DepthCamera()
while True:
    ret, depth_frame, color_frame = dc.get_frame()
    height, width = color_frame.shape[:2]
    expected = 300
    aspect = width / height
    resized_image = cv2.resize(color_frame, (round(expected * aspect), expected))
    crop_start = round(expected * (aspect - 1) / 2)
    crop_img = resized_image[0:expected, crop_start:crop_start+expected]

    blob = cv2.dnn.blobFromImage(crop_img, inScaleFactor, (expected, expected), meanVal, False)
    net.setInput(blob, "data")
    detections = net.forward("detection_out")

    label = detections[0,0,0,1]
    conf  = detections[0,0,0,2]
    xmin  = detections[0,0,0,3]
    ymin  = detections[0,0,0,4]
    xmax  = detections[0,0,0,5]
    ymax  = detections[0,0,0,6]

    className = classNames[int(label)]
    
    scale = height / expected
    xmin_depth = int((xmin * expected + crop_start) * scale)
    ymin_depth = int((ymin * expected) * scale)
    xmax_depth = int((xmax * expected + crop_start) * scale)
    ymax_depth = int((ymax * expected) * scale)
    #xmin_depth,ymin_depth,xmax_depth,ymax_depth
    depth = depth_frame[xmin_depth:xmax_depth,ymin_depth:ymax_depth].astype(float)
    depth= depth*(dc.scale)
    dist,_,_,_ = cv2.mean(depth)
    cv2.rectangle(crop_img, (int(xmin * expected), int(ymin * expected)),(int(xmax * expected), int(ymax * expected)), (255, 255, 255), 2)
    cv2.putText(crop_img, className+str(dist)+"m", (int(xmin * expected), int(ymin * expected) - 5),cv2.FONT_HERSHEY_COMPLEX, 0.5, (255,255,255))
    cv2.imshow("crop_image",crop_img)
    print("Detected a {0} {1:.3} meters away.".format(className, dist))
    key = cv2.waitKey(1)
    if key == 27:
        break
############################realsense_depth.py
realsense_depth.py:
import pyrealsense2 as rs
import numpy as np

class DepthCamera():
    scale=0.0
    def __init__(self):
        # Configure depth and color streams
        self.pipeline = rs.pipeline()
        config = rs.config()

        # Get device product line for setting a supporting resolution
        pipeline_wrapper = rs.pipeline_wrapper(self.pipeline)
        pipeline_profile = config.resolve(pipeline_wrapper)
        device = pipeline_profile.get_device()
        device_product_line = str(device.get_info(rs.camera_info.product_line))

        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
        
        # Start streaming
        profile= self.pipeline.start(config)
        self.scale= profile.get_device().first_depth_sensor().get_depth_scale()
        print(self.scale)
    
    #def get_scale(self)
    #    return self.profile.get_device().first_depth_sensor().get_depth_scale()

    def get_frame(self):
        frames = self.pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()

        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        if not depth_frame or not color_frame:
            return False, None, None
        return True, depth_image, color_image

    def release(self):
        self.pipeline.stop()

MartyG-RealSense, 2023-02-08:
First, please test whether the problem is in the `depth= depth*(dc.scale)` line of the main code by using the default depth unit scale of the 400 Series cameras instead.

`depth= depth*0.001`

In the original distance_to_object script, the following code is used:

```
depth = depth[xmin_depth:xmax_depth,ymin_depth:ymax_depth].astype(float)
# Get data scale from the device and convert to meters
depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
```

In your own project though, it appears that the depth scale is being retrieved from the camera in the separate **realsense_depth.py script,** whilst the above code was all in the same script.  If a hard-coded fixed value of 0.001 in the main script works then it likely means that the retrieved **self.scale** depth scale is not being imported from realsense_depth.py into the main script.

Gaurav37, 2023-02-08:
Hi Marty,

The ```depth = depth*0.001``` gives better performance in that it increases distance as the person moves away from realsense but it is still showing 1 meter extra distance so instead of 1 meter it shows 2 meters, instead of 2 meters it shows 3 meters.

MartyG-RealSense, 2023-02-09:
If the error factor is consistently 1 meter then a workaround might be **depth = (depth*0.001)-1** so that it calculates the distance in meters and then subtracts 1 from the result.

I agree that it looks as though `depth = depth[xmin_depth:xmax_depth,ymin_depth:ymax_depth].astype(float)` causes only depth information within the bounding box area to be considered.

closed 2023-02-23 by MartyG-RealSense