realsenseai/librealsense #11410: Librealsense Packages prevent usage of integrated/external WebCams

lupeterm, 2023-02-08:
This is more of a general issue, likely with the installation process. Not sure how relevant this still is, given that the software used is now legacy / deprecated. 

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455, T265 | 
| Firmware Version                   |  not sure, writing this issue in retrospect|  
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15                                         | 
| Platform                                 | PC  |
| SDK Version                            |  not sure, writing this issue in retrospect                         | 
| Segment			|  General                          | 

### Issue Description
After installing librealsense and realsense-ros (the by now deprecated ROS1 version), I noticed that my laptop was not able to use any webcams (integrated or external). 

Side effects include:
- no entry for `/dev/video*`
- no unusual behavior in the `dmesg` output
- the `uvcvideo` driver is missing
- Integrated Webcam still shows in `lsusb` output

### My "Fix"
Since I am now done with the project, I removed the librealsense packages with `apt remove librealsense` which prompted the software updater to notify me about a new kernel update. Now the integrated camera works again.

MartyG-RealSense, 2023-02-08:
Thanks very much @lupeterm for the report.  This issue has been reported previously in past cases such as https://github.com/IntelRealSense/librealsense/issues/5868 and https://github.com/IntelRealSense/librealsense/issues/7287

It was thought to be related to a kernel conflict.

lupeterm, 2023-02-08:
Ah, thank you. :+1: 
Seems like I missed those issues before.  
This issue can be closed for the sake of redundancy, as far as I am concerned. @MartyG-RealSense

MartyG-RealSense, 2023-02-08:
You are very welcome!  As you suggest, I will close the issue.  Thanks again!

closed 2023-02-08 by MartyG-RealSense