realsenseai/librealsense #11413: Stream live video feed to web server...

ParthPatel2910, 2023-02-08:
Hello, 

I am using Intel's RealSense D435 camera. I just want to send a live video stream captured by realsense camera to the web server over a  WiFi network. Is that possible? If yes, can anyone please suggest the way to do it!! Thanks in advance.

[image]

MartyG-RealSense, 2023-02-08:
Hi @ParthPatel2910  If your project is able to use Python then a **Flask server** may suit your needs, like the Flask webapp project at the link below that has been tested with the D435 camera model.

https://github.com/soarwing52/Remote-Realsense

An alternative would be to stream the camera directly to a web-page, like with the **Depth2Web** project that is compatible with the RealSense SDK 2.0 that RealSense 400 Series cameras use.

https://github.com/js6450/depth2web

https://www.youtube.com/watch?v=eV5NIPKC_pc

MartyG-RealSense, 2023-02-14:
Hi @ParthPatel2910  Do you require further assistance with this case, please? Thanks!

ParthPatel2910, 2023-02-15:
Can you please explain the flow of the project? Like how the flask script works. How it will connect with camera? How it will start streaming the live feed to the dedicated server? Thanks!

MartyG-RealSense, 2023-02-15:
I recommend contacting the author of the script **soarwing52** using the email address on their GitHub profile at the link below.  They have contributed to this support forum in the past and have been friendly.

https://github.com/soarwing52

closed 2023-03-01 by MartyG-RealSense