realsenseai/librealsense #11420: 3D Calibration Tool

TheNemo05, 2023-02-09:
| Camera Model                       | D455 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  NA    | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            | 2.53.1         | 
| Language                            |  Ros/Python
| Required Info                         |                                                                |

### Issue Description
Can you please provide calibration tool for 3D edge detection

MartyG-RealSense, 2023-02-09:
Hi @TheNemo05   In Intel's white-paper guide **Depth Map Improvements for Stereo-based Depth Cameras on Drones** it is suggested that an OpenCV **Scharr operator** be used for edge detection.

https://dev.intelrealsense.com/docs/depth-map-improvements-for-stereo-based-depth-cameras-on-drones#c-post-processing-of-depth-images

If you wanted to generate an edge-map then you could do so with a downloadable visual preset provided by Intel called **EdgeMapD435.json**.  

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

The json can be found in the **Additional Presets not in the Viewer** section of the above link.

[image]

MartyG-RealSense, 2023-02-09:
https://github.com/IntelRealSense/librealsense/issues/4061 is the best available reference regarding setting custom intrinsics with Python.  I do not have an equivalent reference for custom extrinsics.

As an alternative to using Python, I have previously suggested at https://github.com/IntelRealSense/librealsense/issues/10822#issuecomment-1227460195 to export the calibration information as an .xml file from the camera, edit the file's values and then load the file back in and save the customized values to the camera hardware's **calibration table**.

closed 2023-02-23 by MartyG-RealSense