realsenseai/librealsense #11425: Realsense Viewer cannot find the camera after system update

GeanRain, 2023-02-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version                       |  5.15.50 | 
| Platform                                 | Intel NUC  |
| SDK Version                            |  2.53.1                           | 

### Issue Description
Before the system update, Realsense Viewer can find the camera and display the image.
After executing `apt update && apt upgrade` to update the system, Realsense Viewer can't find the camera and can't find the video device in `/dev`.

When I rollback librealsense2-dkms from version `1.3.19` to `1.3.18`, everything works fine.

Before the system update, the kernel version was `5.13` and the SDK version was `2.51.1`. After the update, the kernel version is `5.15` and the SDK version is `2.53.1`.

I hope the SDK can be easily updated to the latest version (like `apt update && apt upgrade`), how to solve this problem. Thank.

MartyG-RealSense, 2023-02-10:
Hi @GeanRain  The recommended camera firmware driver version for SDK 2.53.1 is **5.14.0.0**.  Please first try updating your firmware.  This can be performed in the RealSense Viewer tool.

If the camera is still not detected then the next step I recommend is to remove all librealsense related packages by inputting the command below into the Ubuntu terminal and then repeat the installation process.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

GeanRain, 2023-02-10:
Hi @MartyG-RealSense I upgraded the firmware and repeat the installation process as you said. The video device is still not found in `/dev`. I try to connect a mono camera separately. No camera device was found in `/dev` too.

MartyG-RealSense, 2023-02-10:
2.53.1 has official support for kernel 5.15 included in its packages, so there should not be a kernel related issue with the upgrade.

Are you still able to roll back to librealsense2-dkms 1.3.18 after having performed the package remova instruction?

GeanRain, 2023-02-13:
Secure Boot is enable on the NUC. 
When I installed librealsense2-dkms, I entered the MOK password. But after reboot the password is not confirmed. After that, there is no page to confirm the password when the system starts up.
So when I reinstall librealsense2-dkms and confirm the password. everything works fine.
I think my problem has been solved. Questions can be closed. Thanks again @MartyG-RealSense

closed 2023-02-13 by GeanRain

MartyG-RealSense, 2023-02-13:
It's excellent to hear that you achieved a solution, @GeanRain - thanks very much for the update and for sharing the details of your solution with the RealSense community!

cheslee-z, 2023-05-31:
> Secure Boot is enable on the NUC. When I installed librealsense2-dkms, I entered the MOK password. But after reboot the password is not confirmed. After that, there is no page to confirm the [...]

hi, I meet this problem too, but i can't understand your solution, how did you confirm the password?
maybe i can disable the secure boot on the NUC?