realsenseai/librealsense #11426: Shadow in Depth Images

TheNemo05, 2023-02-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Win (8.1/10) / Linux (Ubuntu 14/16/17  | 
| Kernel Version (Linux Only)    |  NA  | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1  | 
| Language                            | Ros/Python     | 
| Segment			| NA         | 

### Issue Description
While capturing some Depth stream, I ve noticed a huge amount of shadow around my Objects. This has been observed when stream is viewing in 3D.
[image]

[image]

MartyG-RealSense, 2023-02-10:
Hi @TheNemo05  The shadow outline is a common phenomenon in RealSense depth images.  It may be reduced if you change your **depth scale** from 0.001 to 0.001 in the Stereo Module > Controls section.

TheNemo05, 2023-02-10:
No option such as Depth scale was available, the only option under control section was Depth Gain. & No changes were observed after changing Depth Gain from 0.001 to 0.0001.

MartyG-RealSense, 2023-02-10:
My apologies, I mean **Depth Units**.

[image]

TheNemo05, 2023-02-10:
No use in changing Depth Units from 0.001 to 0.0001.

MartyG-RealSense, 2023-02-10:
I replicated the shadow and tested the Viewer settings thoroughly.  The only method that resolved it was to enable the post-processing **Hole Filling Filter**.

**Hole Filling filter disabled**

[image]

**Hole Filling filter enabled**

[image]

TheNemo05, 2023-02-10:
The output is better but there are still some shadow & noise in depth stream. Its not as perfect as yours

MartyG-RealSense, 2023-02-10:
In the post-processing Temporal Filter, changing the 'Smooth Alpha' setting from its default of '0.4' to '0.1' can reduce depth noise and fluctuation.

TheNemo05, 2023-02-10:
The minimum Smooth Alpha we can set for D455 is 0.25. Even though there are many fluctuations in the stream. https://drive.google.com/file/d/1VkgT-Hjg7dzPlF8pMU0o2PyDLuB_1J0V/view?usp=share_link

Please check the above google drive link

MartyG-RealSense, 2023-02-10:
After applying the Hole Filling Filter, it appears that the remaining noise in the bottom foreground of the image may be due to the D455's minimum depth sensing distance of 0.4 meters, compared to 0.1 m on the D435 / D435i camera models.  This means that if an area of the observed scene is closer to the camera than 0.4 m / 40 cm then that area's depth information starts to break up into loose pixels.

Setting the **Disparity Shift** option to a value of '100' instead of its default '0' may help to render missing close-range depth coordinates on your image.  This setting is under **Advanced Controls > Depth Table**

[image]

Changing the 'DS Second Peak Threshold' setting under **Advanced Controls > Depth Control** from '325' to '0' can also reduce depth fluctuation.

[image]

TheNemo05, 2023-02-13:
These settings are creating more & various forms of distortions. Can a firmware upgrade be helpful ?

MartyG-RealSense, 2023-02-13:
A newer firmware version is not currently available as 5.14.0.0 is the newest version at the time of writing this.  A newer firmware may also not have an improvement effect on depth, depending on which features or bug-fixes a particular firmware version adds.

TheNemo05, 2023-02-13:
i think i ve solve some of the issue, by increasing the focal length between camera lens & the object, i was able to eliminate some of the shadow.

MartyG-RealSense, 2023-02-13:
That's great news that you made significant progress!

closed 2023-02-14 by TheNemo05