realsenseai/librealsense #11430: Warning when building on a Jetson Xavier NX with `-DBUILD_WITH_CUDA=true`

nakai-omer, 2023-02-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.14.0.0 | 
| Operating System & Version |  Linux, Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                     | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  2.53.1   | 
| Language                            |  C++    | 
| Segment			| Robot        | 

### Issue Description
When trying to build using the `-DBUILD_WITH_CUDA=true` flag we get the following warning:
```
CMake Warning (dev) in CMakeLists.txt:
  Policy CMP0104 is not set: CMAKE_CUDA_ARCHITECTURES now detected for NVCC,
  empty CUDA_ARCHITECTURES not allowed.  Run "cmake --help-policy CMP0104"
  for policy details.  Use the cmake_policy command to set the policy and
  suppress this warning.

  CUDA_ARCHITECTURES is empty for target "realsense2".
This warning is for project developers.  Use -Wno-dev to suppress it.
```

(Other related issues here are related to windows, and there an answer states it should work over Jetson)

MartyG-RealSense, 2023-02-12:
Hi @nakai-omer  I have not previously seen the **CUDA_ARCHITECTURES is empty for target** error in relation to librealsense.  Can you confirm please that you have used the CMake build flags on the Jetson installation instructions page?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

`cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install`

**-DCMAKE_BUILD_TYPE=release** is particularly important.  By default CMake builds librealsense from source code in developer **Debug** mode, whilst including this flag builds librealsense in **Release** mode.

nakai-omer, 2023-02-12:
@MartyG-RealSense Thanks for the quick response, since this is Jetpack 5.0.1 I ran it like this comment: https://github.com/IntelRealSense/librealsense/issues/10416#issuecomment-1314342663, just added the `-DBUILD_WITH_CUDA=true`, I will try again using the link you sent...

MartyG-RealSense, 2023-02-12:
At the time of writing this, librealsense does not officially support JetPack 5, though it is possible to build for JetPack 5 if kernel patching is not performed, or if **-DFORCE_RSUSB_BACKEND** is set to True in order to build librealsense in **RSUSB backend** mode.  An RSUSB build bypasses the kernel and so does not require kernel patching to be performed.

Support for JetPack 5.0.2 will be introduced in the next librealsense version, which is likely to be numbered **2.54.0**.

nakai-omer, 2023-02-12:
thanks for the update, is there an expected date of release for 2.54.0?

MartyG-RealSense, 2023-02-12:
I do not have a release date estimate for the next version, though the JetPack 5.0.2 support has been merged into the **development** branch of librealsense.  The development branch becomes the latest 'master' version of librealsense when released.

https://github.com/IntelRealSense/librealsense/tree/development

If you are not using multiple cameras in your project then it should be fine to build the current 2.53.1 version with DFORCE_RSUSB_BACKEND = True to use 2.53.1 with JetPack 5.

nakai-omer, 2023-02-12:
Got it thanks a lot!

closed 2023-02-12 by nakai-omer