realsenseai/librealsense #11437: How to get true depth from depth data

Xue21, 2023-02-14:
I get the depth data with D435i, it's encoding is16UCI, How can I get the true depth value?

MartyG-RealSense, 2023-02-14:
rHi @Xue21  The real-world distance in meters can be found by multiplying the 16-bit depth value by the Depth Unit Scale of the camera.  The depth unit scale value of the D435i is 0.001.  So if the 16-bit depth value was 6500 and it was multiplied by 0.001, the real-world depth would be 6 meters.

Xue21, 2023-02-19:
I am using d435 by Python,   color_image is not a array, how can I get one image from color frame? Below are some codes.
    if self.pushButton_play.text() == "放映":
            self.timer = QTimer()
            self.timer.timeout.connect(self.show_image)
            self.timer.start(10)
            self.pushButton_play.setText("就绪")
        else:
            self.pushButton_play.setText("放映")
            self.timer.stop()
            self.Image_show.clear()
            self.Image_show.setText("Image Video")

    def show_image(self):
        # Get frames from camera
        frames = self.pipeline.wait_for_frames()
        aligned_frames = self.align.process(frames)
        self.color_frame = aligned_frames.get_color_frame()
        self.depth_frame = aligned_frames.get_depth_frame()
        # Convert frames to numpy arrays
        color_image = np.asanyarray(self.color_frame.get_data())
        depth_image = np.asanyarray(self.depth_frame.get_data())

MartyG-RealSense, 2023-02-19:
If you wish to export color as an image, a Python script at https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225 exports color as a PNG image file and depth as an array of scaled matrices saved as an .npy file, whilst a Python script at https://github.com/IntelRealSense/librealsense/issues/2877#issuecomment-446102880 exports both depth and color as PNG images.

closed 2023-03-05 by MartyG-RealSense