realsenseai/librealsense #11443: Unable to run SDK on Apple Silicon

HasanTheSyrian, 2023-02-15:
I follow the guide [here](https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk).

But when I run `viewer`, it opens then crashes, and I get this screen on Xcode.

[image]

`libusb` is installed, of course.

I did the build via opening 
```
/Users/hasan/Downloads/librealsense-2.53.1/build/librealsense2.xcodeproj
```

and clicking `Product > Run` and `Product > Build` then I clicked the start button at the top left.

[image]

Excuse me as I have never used Xcode before.
___
I also tried doing 
```bash
brew install librealsense
```
then running
```bash
brew --prefix librealsense
/opt/homebrew/Cellar/librealsense/2.53.1/bin/realsense-viewer; exit
```

which worked fine until I plug in my D455 (running the latest firmware) which made the application crash.

HasanTheSyrian, 2023-02-15:
When I unplug the camera the program runs fine.

When I plug back the camera it gives me an error relating to not having enough power (I don't remember exactly it flashes before the program crashes) 

I think my USB-A to USB-C dongle might be the bottleneck.

MartyG-RealSense, 2023-02-15:
Hi @HasanTheSyrian  Are you using your USB A to USB C dongle to connect the camera to a USB-C port on your Mac, please?  If you are then RealSense cameras typically have a greater chance of problems occurring when a USB-C connection on a Mac is used.

HasanTheSyrian, 2023-02-15:
> Hi @HasanTheSyrian Are you using your USB A to USB C dongle to connect the camera to a USB-C port on your Mac, please? If you are then RealSense cameras typically have a greater chance of problems occurring when a USB-C port on a Mac is used.

Yes, I mentioned I'm using a dongle in my comment above, however, I tried 2 USB-C to USB-C cables but none worked. 
The one included in the box is quite big (I think it might be USB-C PD). I'm thinking that the cables I have are not meeting some sort of wattage threshold. 

I'm unsure how much power the camera needs. I looked in the datasheet, but couldn't find anything.

MartyG-RealSense, 2023-02-15:
I recommend budgeting for a maximum power draw of 2 watts per camera.  However, if you are simply inserting the camera without enabling the streams then the power draw should be less than that.

As the USB port of the camera is connected to a circuit board inside the camera called the **Vision Processor D4 Board**, it makes sense to quote the power requirements for the D4 board.

[image]

HasanTheSyrian, 2023-02-15:
I can't enable anything. The program crashes almost the moment I plug it in. 2 Watts seems like a pretty low amount for any USB-C cable though, no? Seems like it should be supplied with enough power.

W = V(~3) x I(~1.5)

[image]

This is the error BTW

MartyG-RealSense, 2023-02-15:
If you have tried installing with **brew install librealsense** then that method should be compatible with Apple Silicon M1 and also with MacOS Big Sur and Monterey.

In regard to the original guide for Mac installation that you linked to at the top of the issue, it is not recommendable to use that guide for Apple Silicon M1 as it was written for Intel-based Macs.  There is a highly detailed guide for Apple Silicon installation at the link below that a RealSense community member contributed.

https://lightbuzz.com/realsense-macos/

HasanTheSyrian, 2023-02-15:
> If you have tried installing with **brew install librealsense** then that method should be compatible with Apple Silicon M1 and also with MacOS Big Sur and Monterey.
>
> In regard to the original guide [...]

Thanks, I'll check it out and report back later.

MartyG-RealSense, 2023-02-15:
At https://github.com/IntelRealSense/librealsense/issues/2811#issuecomment-443909901 another user also suggests a method for launching the Viewer after having installed librealsense with brew.

HasanTheSyrian, 2023-02-15:
I tried the things you gave me, same issue sadly.

Here's the crash report macOS gives.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               realsense-viewer [1877]
Path:                  /opt/homebrew/*/realsense-viewer
Identifier:            realsense-viewer
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [1544]
Responsible:           iTerm2 [1539]
User ID:               501

Date/Time:             2023-02-15 22:10:49.2764 +0300
OS Version:            macOS 13.1 (22C65)
Report Version:        12
Anonymous UUID:        08DEBF21-C414-F108-A750-CE1F7DEDBC79

Time Awake Since Boot: 640 seconds

System Integrity Protection: enabled

Crashed Thread:        1

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Thread 0::  Dispatch queue: com.apple.main-thread
0   AGXMetalG13X                  	       0x1d30150d4 0x1d2cbb000 + 3514580
1   AGXMetalG13X                  	       0x1d3225b40 0x1d2cbb000 + 5679936
2   AGXMetalG13X                  	       0x1d3225b40 0x1d2cbb000 + 5679936
3   AGXMetalG13X                  	       0x1d3224d90 0x1d2cbb000 + 5676432
4   AGXMetalG13X                  	       0x1d32373c4 0x1d2cbb000 + 5751748
5   AGXMetalG13X                  	       0x1d32375f0 0x1d2cbb000 + 5752304
6   AGXMetalG13X                  	       0x1d3237294 0x1d2cbb000 + 5751444
7   AppleMetalOpenGLRenderer      	       0x107039570 makeTextureViewRscIfNeeded(GLRTextureResource*, MTLPixelFormat, MTLTextureType, _NSRange, _NSRange, unsigned int) + 348
8   AppleMetalOpenGLRenderer      	       0x1070393f0 GLRTextureResource::getGLViewResource(bool, bool, bool, eGLRSamplerType) + 356
9   AppleMetalOpenGLRenderer      	       0x10704ad98 GLDContextRec::addRenderPassResources() + 92
10  AppleMetalOpenGLRenderer      	       0x107053674 gldClearFramebufferData + 248
11  GLEngine                      	       0x1d796ee6c glClear_Exec + 536
12  realsense-viewer              	       0x100230b94 rs2::viewer_model::handle_ready_frames(rs2::rect const&, rs2::ux_window&, int, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) + 3020
13  realsense-viewer              	       0x1002b7a60 main + 8916
14  dyld                          	       0x1880cfe50 start + 2544

Thread 1 Crashed:
0   libsystem_kernel.dylib        	       0x1883c21b0 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x1883f8cec pthread_kill + 288
2   libsystem_c.dylib             	       0x1883322c8 abort + 180
3   libsystem_c.dylib             	       0x188331620 __assert_rtn + 272
4   libusb-1.0.0.dylib            	       0x100c33390 darwin_init.cold.1 + 40
5   libusb-1.0.0.dylib            	       0x100c2d2e0 darwin_init + 80
6   libusb-1.0.0.dylib            	       0x100c27ca8 libusb_init + 600
7   librealsense2.2.53.1.dylib    	       0x102adc9c8 librealsense::platform::usb_context::usb_context() + 96
8   librealsense2.2.53.1.dylib    	       0x102ae27bc librealsense::platform::usb_enumerator::query_devices_info() + 100
9   librealsense2.2.53.1.dylib    	       0x102db75e8 librealsense::platform::query_uvc_devices_info() + 52
10  librealsense2.2.53.1.dylib    	       0x102aeb4ec librealsense::polling_device_watcher::polling(dispatcher::cancellable_timer) + 84
11  librealsense2.2.53.1.dylib    	       0x1029e19f8 void std::__1::__invoke_void_return_wrapper<void, true>::__call<active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::do_loop()::'lambda'(dispatcher::cancellable_timer)&, dispatcher::cancellable_timer>(active_object<std::__1::function<void (dispatcher::cancellable_timer)> >::do_loop()::'lambda'(dispatcher::cancellable_timer)&, dispatcher::cancellable_timer&&) + 68
12  librealsense2.2.53.1.dylib    	       0x102899658 void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, dispatcher::dispatcher(unsigned int, std::__1::function<void (std::__1::function<void (dispatcher::cancellable_timer const&)>)>)::$_0> >(void*) + 240
13  libsystem_pthread.dylib       	       0x1883f906c _pthread_start + 148
14  libsystem_pthread.dylib       	       0x1883f3e2c thread_start + 8

Thread 2:
0   libsystem_pthread.dylib       	       0x1883f3e18 start_wqthread + 0

Thread 3:
[... 280 lines cut ...]
      },
      "deploymentId" : 240000022
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,1, BootROM 8419.60.44, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  3 2022 02:35:52 version 20.10.965.9.8.7.129 FWID 01-b0e84a9b
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: Intel(R) RealSense(TM) Depth Camera 455
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

MartyG-RealSense, 2023-02-16:
The best place to ask about this problem is likely to be https://github.com/IntelRealSense/librealsense/issues/9916 as this is a central location for MacOS RealSense users with Apple Silicon M1 / Monterey installation problems.

MartyG-RealSense, 2023-02-23:
Hi @HasanTheSyrian  Bearing in mind the advice in the comment above, do you require further assistance with this case please?  Thanks!

HasanTheSyrian, 2023-02-23:
Im storing the camera at university, currently. Ill try again when I go
there. I will not forget about closing the issue.

MartyG-RealSense, 2023-02-23:
Okay, that's no problem at all.  Thanks very much for the update!

HasanTheSyrian, 2023-03-03:
I'll work on the camera from my desktop PC. If I need to use it on my Mac, I'll check the links and maybe re-open an issue.

closed 2023-03-03 by HasanTheSyrian