realsenseai/librealsense #11445: How to install RealSense SDK in itx-rk3588j

Aya-nuomi, 2023-02-16:
My hardware: itx-rk3588j
ubuntu kernel: 5.10.110

issues: I want to install realsense SDK in itx-rk3588j, but the kernel version is not supported, I tried to change the kernel verision of itx-rk3588j, but didn't succeed.

First, i install the realsense SDK follow the librealsense/doc/installation.md, and stuck on step4,  I run `./scripts/patch-realsense-ubuntu-lts.sh`here.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
[image]

Then, i tried to run `./scripts/patch-realsense-ubuntu-odroid.sh`,it didn't work also.
[image]

Then, I saw issue#11050, but my situation is a little different, i didn't install realsense SDK successfully. I follow the instruction in this issue, try to build librealsense from source code without having patched the kernel，
`cmake ../ -DFORCE_RSUSB_BACKEND=flase -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`，but I can't make successfully.
https://github.com/IntelRealSense/librealsense/issues/11050#issuecomment-1310082438

do you have any idea how can i use realsense D435 in itx-rk3588j?
thanks a lot!

MartyG-RealSense, 2023-02-16:
Hi @Aya-nuomi  For an ITX RK chipset I would recommend building the SDK from source code with **-DFORCE_RSUSB_BACKEND=true**

Setting RSUSB to True bypasses the kernel and so the SDK build is not dependent on Linux versions or kernel versions and does not require kernel patching.

MartyG-RealSense, 2023-02-23:
Hi @Aya-nuomi  Do you require further assistance with this case, please?  Thanks!

Aya-nuomi, 2023-02-28:
Forgive me for reply now, thanks for your advice, but I need to use multi-cameras, so I use camera of other brands now, I don't require futher assistance.
Thanks again, and looking forward to your support to kernel type 5.10.110.

MartyG-RealSense, 2023-02-28:
Thanks very much @Aya-nuomi for the update.  Although RSUSB is not as ideally suited to multiple camera applications as a kernel-patched installation of the SDK, more than one camera can be used with it.

As you do not require further assistance, I will close the issue.  Thanks again!

closed 2023-02-28 by MartyG-RealSense