realsenseai/librealsense #11449: Cannot set image size?

dennisushi, 2023-02-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |  Ubuntu 20  | 
| Platform                                 | PC  |
| SDK Version                            | Version: 2.51.1.4348                         | 
| Language                            | python                          | 

### Issue Description
```python
class Camera():
    def __init__(self):
        self.rs_p = rs.pipeline()
        config = rs.config()
        config = rs.config()
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

    def capture(self):
        frames = self.rs_p.wait_for_frames()
        
        align_to = rs.stream.color
        align = rs.align(align_to)
        # Align the depth frame to color frame
        if not aligning_depth_to_color:
            aligned_frames = frames
        else:
            aligned_frames = align.process(frames)

        aligned_depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()

        decimate = rs.decimation_filter()
        decimate.set_option(rs.option.filter_magnitude, 1)
        aligned_depth_frame = decimate.process(aligned_depth_frame)

        #
        if use_filtering:
            filters = [rs.disparity_transform(),
                    rs.spatial_filter(),
                    rs.temporal_filter(),
                    rs.disparity_transform(False)]
            for f in filters:
                aligned_depth_frame = f.process(aligned_depth_frame)

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        depth_sensor = self.rs_cfg.get_device().first_depth_sensor()
        depth_scale = depth_sensor.get_depth_scale()
        depth_image = depth_image.astype(np.float32)*(depth_scale) # in m
        return color_image, depth_image
```

I believe similar code worked for me before with a different camera, but now I am seeing that the depth image has shape  (480, 848) and rgb image has shape (720, 1280, 3), instead of (480,640) as requested by the config. Am I doing something wrong or is this a bug?

MartyG-RealSense, 2023-02-17:
Hi @dennisushi  I would first recommend trying to remove the Decimation filter if you want the image to have the defined 640x480 resolution, as the purpose of the Decimation filter is to reduce the resolution of the image.  

The supported range of values for this filter is **2 to 8,** with 2 being the default value.  Your magnitude is set to '1'.

The original resolution will be divided by the magnitude value.  So if the magnitude is set to '2', a 1280x720 resolution would be divided by 2 to be downscaled to 640x360.

https://github.com/IntelRealSense/librealsense/issues/2356#issuecomment-465766323 provides a Python example of using the decomation filter in combination with alignment.

dennisushi, 2023-02-20:
Without the decimiation filter, the sizes are still the same. It was my understanding that if set to 1, it doesn't do anything.

I figured it out though - `self.rs_cfg = self.rs_p.start(config)` I had forgotten to write `config` here.

closed 2023-02-20 by dennisushi

MartyG-RealSense, 2023-02-20:
That makes sense, as the absence of config in the start instruction's brackets will cause the program to ignore a custom stream configuration and apply the default stream configuration of the particular camera model that is being used.  Thanks very much for the update about your success!