realsenseai/librealsense #11451: L515 output Point Cloud with Raw RGB data

weisterki, 2023-02-17:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L500 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Platform                                 | PC |
| SDK Version                            |  {2.53.1}                          | 
| Language                            |  {python }                          | 
| Segment			|  {Smartphone/VR/AR }                          | 

### Issue Description
How to get L515 point cloud stream with Raw RGB data? 
The available example codes like opencv_viewer_example.py are all combined the depth points with `frames.get_color_frame()`, which colours are processed after ISP(Image Signal Processor) pipeline, such as AWB. Is there any code which could output raw RGB point cloud stream?

MartyG-RealSense, 2023-02-17:
Hi @weisterki  It may be possible to capture raw **RAW16** frames from the camera hardware and use OpenCV's **cv2.cvtColor** instruction to convert them to **RGB8** format that is usable by a RealSense program script.  https://github.com/IntelRealSense/librealsense/issues/7275 provides Python code for doing so, though this method has not been previously tested on the L515 model.

weisterki, 2023-02-18:
> Hi @weisterki It may be possible to capture raw **RAW16** frames from the camera hardware and use OpenCV's **cv2.cvtColor** instruction to convert them to **RGB8** format that is usable by a [...]

Thanks Marty，I checked that case but I can not find the relative code of acquiring  **RAW16** ，cause that is about how to transform RAW to RGB. Could you give me some advice of acquiring  **RAW16**? That is the key problem.

MartyG-RealSense, 2023-02-18:
It is possible that the cvtColor conversion method that works for 400 Series cameras cannot work with the L515 model.  As mentioned above, the conversion process has never been tested with L515 before now.

I do not know of any other way in which raw camera frames could be accessed and used on L515, unfortunately. I do apologize.

weisterki, 2023-02-18:
> It is possible that the cvtColor conversion method that works for 400 Series cameras cannot work with the L515 model. As mentioned above, the conversion process has never been tested with L515 [...]

The code 
     `rawframes = pipeline.wait_for_frames() `
     `raw_frame = rawframes.get_color_frame() `
      `imgRAW = np.asanyarray(raw_frame.get_data())  `
means getting frames from hardware. I don't know whether the data here is RAW or after ISP? Could you point out which one controls aquiring RAW data? because in another examples, they use the same code to get RGB after pipeline, as shown in the example **opencv_pointcloud_viewer.py** , it uses
`color_frame = frames.get_color_frame()`
`color_image = np.asanyarray(color_frame.get_data())`
to get processed RGB data.
And the code
`imgRGB = cv2.cvtColor(imgRAW, cv2.COLOR_BGR2RGB)`
is just transform data from BGR to RGB, this step assumes that the RAW data has been already loaded into imgRAW.
I just want to use the same method of D400 series onto my L515. I thought there suppouse to be another code to control the aquiring of RAW data, like 
`config.enable_raw_stream(rs.stream.color, rs.format.bgr8, 30)`.

MartyG-RealSense, 2023-02-18:
https://github.com/IntelRealSense/librealsense/issues/7275 is the only reference that uses **rawframes.get_color_frame()** and so whilst this method of accessing raw RGB frames clearly works for 400 Series cameras, it is not possible to determine whether that method is also appropriate for the L515 and its unique hardware design.

Raw frames need to be converted because they cannot be directly used by the SDK as a color stream format, such as RGB8.  For example, trying to access them in the RealSense Viewer with a stream format such as RAW10 provides an error message.

[image]

This is because whilst the camera firmware driver supports raw frames, access to them is not exposed to the end-user through the RealSense SDK.

weisterki, 2023-02-18:
> This is because whilst the camera firmware driver supports raw frames, direct access to them by the end-user without a conversion step is not provided by the RealSense SDK.

Got it. So the `rawframes.get_color_frame()` is the method to read RAW data. Do I need to change `config.enable_raw_stream(rs.stream.color, rs.format.bgr8, 30)`bgr8 to RAW16 ? Or as you said the SDK doesn't support RAW frame, so I just use (bgr8) here, and add a OpenCV transform from RAW to RGB in the frame running part.

MartyG-RealSense, 2023-02-18:
An OpenCV cvtColor transformation from raw to RGB is likely to be the best way to achieve the goal if it is possible to do so on L515.  I apologize sincerely again for the uncertainty regarding this particular question.

weisterki, 2023-02-20:
> An OpenCV cvtColor transformation from raw to RGB is likely to be the best way to achieve the goal if it is possible to do so on L515. I apologize sincerely again for the uncertainty regarding this particular question.

I tried the code in https://github.com/IntelRealSense/librealsense/issues/7275, and I got a point cloud without colour.
[image]
Then I just tied to map the cloud colour without transformation, I got the same colour with ISP processed, which means that the `rawframes.get_color_frame()` can not acquire the RAW16 data on my L515 device. I also tried in RealSense Viewer, and I found there was no RAW16 option in RGB.
[image]
We are working on the colour calibration of point clouds, so getting data as RAW  as possible is our vital target. :(

MartyG-RealSense, 2023-02-20:
If you have the option to use Y16 then it will be the closest to raw because Y16 is **unrectified** and unlike the other streams, it does not have a distortion model applied to it as Y16 is used for camera calibration.  It is a monochrome RGB format though.

weisterki, 2023-02-21:
> RGB

In the L515 datasheet, Y series data is the image presenting the reflected IR intensity. I don't know exactly about Y16 data, could you PLZ show me more info about it?
 Is Y16 similar to the mosaic image, could I extract the R, G, and B channels separately?

MartyG-RealSense, 2023-02-21:
There is no information available about the Y16 format's structure unfortunately other than that it is a "16-bit per-pixel grayscale image" and it is unrectified.

To extract R, G, B channels separately from an RGB image, **OpenCV** code would need to be used, like in the tutorial at the link below.

https://lindevs.com/extract-individual-channels-from-rgb-image-using-opencv

weisterki, 2023-02-21:
> There is no information available about the Y16 format's structure unfortunately other than that it is a "16-bit per-pixel grayscale image" and it is unrectified.
>
> To extract R, G, B channels [...]

Ok, so you mean the Y16 data is just greyscale data captured from a monochrome camera, then how exactly the Y16 data be captured? Does this data captured by one channel of the RGB sensor of L515? or the IR sensor?
 As I know there are two sensors in L515: RGB and IR sensor. The RGB sensor could capture the light within visible wavelength, and I don't know how IR sensor captures IR images.

MartyG-RealSense, 2023-02-21:
A component in the L515 camera hardware called **MEMS** (Micro-Electrical-Mechanical-System) scans an IR laser beam over the entire field of view (FOV) of the camera.  An IR image is then constructed based upon the intensity of the IR light that is reflected off objects in the scene.

There is no further information available about Y16 that I can provide.  I do apologize.

weisterki, 2023-02-21:
> 

OK， thank you so much, Marty. The IR image is not only related to the reflectance of the surface but also the distance, right? 
By the way, if I want to get RGB  as RAW as possible, then I need to turn off most of the processes of ISP pipeline, so how could I control the pipeline, for example turn off the gain, auto white balance and so on?

weisterki, 2023-02-21:
> A component in the L515 camera hardware called **MEMS** (Micro-Electrical-Mechanical-System) scans an IR laser beam over the entire field of view (FOV) of the camera. An IR image is then constructed [...]

Ok, thank you so much, Marty. The IR image is related to both distance and surface reflectance, right? 
Put aside the Y16 data, if I want to get the RGB data as RAW as possible,  I need to know what the ISP pipeline does and turn off most of its procedures. For example,  turn off the gain,  auto white balance and so on.  Could you show me the process of L515 ISP and how to turn off them?

MartyG-RealSense, 2023-02-21:
I would speculate that distance may have an effect on the infrared image in terms of reflections that are further away being less visible to the camera.

If a particular setting is listed in the RGB section of the RealSense Viewer's options side-panel when using an L515 camera then it should be controllable with Python code.  An example of Python code for controlling auto white-balance one-off is at  https://github.com/IntelRealSense/librealsense/issues/10143

MartyG-RealSense, 2023-02-22:
If you can stream Y16 RGB in the Viewer then you should not need to perform an OpenCV color conversion like you have to with raw frames.  The stream config instruction should look like this:

`config.enable_stream(rs.stream.color, 640, 480, rs.format.y16)`

weisterki, 2023-02-22:
> If you can stream Y16 RGB in the Viewer then you should not need to perform an OpenCV color conversion like you have to with raw frames. The stream config instruction should look like this:
> 
> `config.enable_stream(rs.stream.color, 640, 480, rs.format.y16)`

Yes, I set my stream config as you said. **But** if I don't transform the stream, the ERROR will throw
[image],
in the function of point cloud UV mapping. I thought that is because the Y16 stream is [1080,1920] as I set, but the UV mapping needs [1080,1920,3]. I tried to copy and expand the Y16 array to three dimensions, like
    `temp = np.full((color.shape[0], color.shape[1], 3), -1)`
   ` temp[:,:,0] = color`
    `temp[:,:,1] = color`
    `temp[:,:,2] = color`
   `color = temp`
and that works, seems the result is same with using OpenCV `COLOR_BAYER_GR2BGR` command to transform to 3 dimensions.
So I am wondering how the Y16 data stored, maybe using Bayer pattern doesn't make sense.

weisterki, 2023-02-22:
> If you can stream Y16 RGB in the Viewer then you should not need to perform an OpenCV color conversion like you have to with raw frames. The stream config instruction should look like this:
> 
> `config.enable_stream(rs.stream.color, 640, 480, rs.format.y16)`

By the way, I captured a Y16 point cloud through RealSense Viewer. I checked the point cloud by MeshLab and I found the point cloud is green, so I tried to read the PLY data. The data shows that there are only values in the G channel of each of the vertexes. So that means the Y16 data is finally stored as GREEN colour. Why is that?
[image]

MartyG-RealSense, 2023-02-22:
As mentioned earlier, there is no information available about the structure of Y16 format and so I am not able to answer questions about its channels, unfortunately.  Again, I apologize.

weisterki, 2023-03-03:
> oned earlier, there is no information available about the structure of Y16 format and s

Hi Marty, I bought another RealSense device, D435i, and I successfully captured RAW16 data, but I don't know the data structure of RAW16, is it RGGB or BGGR or else? I could transfer it if I know its arrangement.

MartyG-RealSense, 2023-03-03:
There is not an official RealSense reference source for this, but I am aware that the camera's raw image is **bayered**.  My research indicates that both RGGB and BGGR can apply to a raw image.  however, the term most commonly associated with a bayered image is **RGGB**.

weisterki, 2023-03-03:
> There is not an official RealSense reference source for this, but I am aware that the camera's raw image is **bayered**. My research indicates that both RGGB and BGGR can apply to a raw image. However, the term most commonly associated with a bayered image is **RGGB**.

Thanks, I will try.

weisterki, 2023-03-03:
> There is not an official RealSense reference source for this, but I am aware that the camera's raw image is **bayered**. My research indicates that both RGGB and BGGR can apply to a raw image. However, the term most commonly associated with a bayered image is **RGGB**.

I tried cv2.COLOR_BAYER_RG2BGR, and I got pink tone image, like this.
[image]
I thought that is not what it looks like as RAW image, RAW images usually present like deep green tone. I also capture a point cloud  mapped from the RAW data, and it shows mosaic style.
[image]
I've already transformed the bayer data to 3-channel RGB from the OpenCV code I mentioned earlier.

MartyG-RealSense, 2023-03-03:
In https://github.com/IntelRealSense/librealsense/issues/7275 the RealSense user performing the conversion in that case used the color-space **COLOR_BAYER_GR2BGR** to obtain a greened image.

`imgRGB = cv2.cvtColor(imgRAW, cv2.COLOR_BAYER_GR2BGR)`

[image]

So they were using GR2 instead of RG2.

weisterki, 2023-03-06:
> In #7275 the RealSense user performing the conversion in that case used the color-space **COLOR_BAYER_GR2BGR** to obtain a greened image.
> 
> `imgRGB = cv2.cvtColor(imgRAW, cv2.COLOR_BAYER_GR2BGR)`
> 
> [image]
> 
> So they were using GR2 instead of RG2.

I dug into the detail of the D435i Raw image. Firstly I captured a raw image with a red sample in the scene.
[image]
Then, I observed the rule of data arrangement and found that the mosaic data should be G **G. Then I split the data and then merged them, finally proving the D435i RGB camera should be GRBG bayer pattern. 
[image]

MartyG-RealSense, 2023-03-06:
Great, thanks so much for the detailed feedback that will benefit RealSense users in future!

closed 2023-03-17 by MartyG-RealSense