realsenseai/librealsense #11453: Access IMU and calculate Yaw Pitch and Roll while streaming RGB and Depth using Python on D435i

pfcouto, 2023-02-17:
Hello. I was trying to access the IMU and get the Yaw Pitch and Roll while also streaming the RGB and Depth. I came across this issue #5628 saying that it isn't possible, is it still the case? I tried to follow this issue #3409 (https://github.com/IntelRealSense/librealsense/issues/3409#issue-418046178) and I get the same error. My code and the error are shown in the pictures. Can you help me get through this error and calculate Yaw Pitch and Roll? Thanks!

[image]

[image]

[image]

pfcouto, 2023-02-17:
Managed to get the x, y and z following #6031. How can I get the Yaw Pitch and Roll using this values on D435i? I saw an issue for the T265, but didn't find any for the D435i. Also, now I have 2 pipelines in different threads, there is no way to synchronize both right?

MartyG-RealSense, 2023-02-18:
Hi @pfcouto  There is a discussion at https://github.com/IntelRealSense/librealsense/issues/4391 regarding obtaining roll, pitch and yaw from a D435i IMU with Python code.

MartyG-RealSense, 2023-02-24:
Hi @pfcouto  Do you require further assistance with this case, please?  Thanks!

pfcouto, 2023-02-24:
Hi. I managed to get the IMU based on other issues. I am closing this issue. Thanks!

closed 2023-02-24 by pfcouto

MartyG-RealSense, 2023-02-24:
That's great news, @pfcouto - thanks very much for the update!  :)

yash7321, 2024-07-01:
@pfcouto i am trying to do the same for a project, how reliable are the imu values, i am trying to do slam from only Realsense-D435i

pfcouto, 2024-07-01:
Hello @yash7321, to be honest I don't know since I didn't work on it a lot after opening and closing this issue, however just by moving around and seeing the values I think it wasn't that bad. A slam using only the d435i might be difficult because it doesn't have some specific hardware (I don't remember what it was but I think it was something the T265 had and this one didn't). But if you manage to make a slam using python and using only this camera please update this issue, I would like to see how you accomplished that.