realsenseai/librealsense #11455: Proper use of create_pointcloud function in C api?

giannishatzo, 2023-02-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1                          | 
| Language                            |  C                         | 
| Segment			|                           | 

### Issue Description
I have so far used successfully the C api to do basic functionality (streaming depth & IR frames, get/set options, etc.) but I struggle to implement more complex tasks like aligning depth to IR frame and extracting the pointcloud.

Is there an example of proper use of rs2_create_pointcloud? Do we also need a frame_queue, and do we need to push new frames on every iteration using rs2_allocate_points or other function? Thanks in advance!

MartyG-RealSense, 2023-02-18:
Hi @giannishatzo  The RealSense SDK's **Projection** documentation provides example C code for pointcloud generation with **rs2_create_pointcloud** and alignment with **rs2_create_align** in the section of the documentation linked to below.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#processing-blocks-helpers

It is not usually necessary to align depth to infrared, as they originate from the same sensor and so are already perfectly aligned, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2782#issuecomment-442379838

giannishatzo, 2023-02-18:
Hi @MartyG-RealSense , thanks a lot for the quick reply and directing me to the right documentation. 
Following those instructions and the relevant issue #1626 I was able to get my pointcloud data!

For other people that might struggle with it, extra attention is needed at the name of the frame you release at the end. This changes after you process the frame to the new processed frame name.

Also, the rs2_wait_for_frame(frame_queue, time_out, error) can be a bit confusing as I was already doing rs2_pipeline_wait_for_frames(pipeline, time_out, error) but the first one is waiting for the processing to finish while the second one is the one getting the depth/color frame(s) if I understood correctly.

Thanks a lot and keep up the great work!

closed 2023-02-18 by giannishatzo

MartyG-RealSense, 2023-02-18:
You are very welcome, @giannishatzo - thanks so much for the update and for sharing your experience for the benefit of other RealSense users who are working with C  :)