realsenseai/librealsense #11466: How to Get second infrared frame in D405

Daa98, 2023-02-20:
Hello, I'm sorry I didn't find answers to these two problems.

1. how to get the second infrared frame in the python SDK.
2. the band of D405 infrared camera

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | (latest) | 
| Operating System & Version |   {Win10)   | 
| SDK Version                            | latest                         | 
| Language                            | python                       |

Daa98, 2023-02-21:
> Hi @Daa98  The right-side 'Infrared 2' stream is only available if the camera is on a USB 3 connection.  
> 
> To access both the left and right IR streams with scripting, you should give each stream its own index number in order to differentiate between the two. https://github.com/IntelRealSense/librealsense/issues/3878#issuecomment-569550710 has an example Python script for doing so.
> 
> Can you describe what you mean by the 'band' of the infrared camera, please?  If you mean the projector and its infrared dot pattern, the D405 camera model is not equipped with this.

Thanks for your help.

I want to use the ir cameras in dark. So I bought an 850nm fill light, but it doesn't work. I don't know the problem casued by the band of my fill light or the power. 
In addition, can d405 get depth from two ir without rgb?

MartyG-RealSense, 2023-02-21:
Infrared lamps will not work with D405 as that particular camera model is equipped with **IR Cut** filters on its sensors that prevent them from seeing infrared frequencies such as 850 nm.  They can only see normal visible light from a conventional lamp or torch.  Other 400 Series camera models can use the light of external IR illuminator lamps as they do not have an IR Cut filter.

The generation of depth frames on RealSense 400 Series cameras is carried out inside the camera hardware by using **raw** left and right infrared images and then sending the completed depth frame along the USB cable to the computer.  It is not related to RGB or to the left and right infrared streams.  So depth can be streamed even when the Infrared and Infrared 2 streams are disabled.

Daa98, 2023-02-21:
> Infrared lamps will not work with D405 as that particular camera model is equipped with **IR Cut** filters on its sensors that prevent them from seeing infrared frequencies such as 850 nm. They can [...]

Oh how sad.  My plan was to get infrared images in the dark.
Now I'm confused about the IR camera in D405. If all IR light is cut off, which band will the IR camera capture from. Or are infrared cameras actually just monochrome cameras in the visible band?
Can I get more information about IR cameras and IR cutting parameters? Then I can decide to change the camera or use a suitable  illuminator lamps.

Thanks again for your answer!

MartyG-RealSense, 2023-02-21:
The IR Cut filters on the D405's infrared sensors do not prevent the infrared streams from functioning.  Instead, they reduce the range of light frequencies that are able to enter the sensor by blocking the near-infrared range (800 nm +) so that only the lower visible-light frequencies are allowed to enter the sensor.  

A consequence of this is that if the visible light in a scene is reduced (because of dimmer sunlight or a room light being turned off, for example) then the infrared image becomes darker because it cannot use any infrared light sources that are present in the scene.

The **Part 2a. Long or short pass** section of Intel's white-paper guide at the link below discusses IR Cut filters, which are a 'shortpass' type of filter.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#3-types-of-optical-filters

A little further down that section of the paper, a wavelength frequency chart of a shortpass filter is provided.

closed 2023-03-02 by Daa98