realsenseai/librealsense #11467: librealsense2-udev-rules postinst is broken on systems with NFS homedirs

steven676, 2023-02-20:
While attempting to run an apt update on some Ubuntu 20.04 machines in a university computer lab, we encountered errors during the configuration stage when upgrading librealsense2-udev-rules to version 2.53.1-0~realsense0.8250:
```
Setting up librealsense2-udev-rules:amd64 (2.53.1-0~realsense0.8250) ...

Postinst script activated...
mkdir: cannot create directory ‘/home/tor’: Permission denied
dpkg: error processing package librealsense2-udev-rules:amd64 (--configure):
 installed librealsense2-udev-rules:amd64 package post-installation script subprocess returned error exit status 1
```
Attempting to downgrade to the previous version (2.50.0-0~realsense0.6128) raised the same error.

The problem is in the postinst script for the librealsense2-udev-rules package:
```
#!/bin/sh -e
 
echo "\nPostinst script activated..."
# Copy the preset files to the actual $HOME location
target_dir="/home/${SUDO_USER}/Documents/librealsense2"
mkdir -p "${target_dir}"
```

The mkdir will fail on most systems with /home on NFS -- including our lab machines -- because the NFS server treats attempts to access the filesystem as root from a client as unprivileged (root_squash).  Because `-e` was passed to the shell, this causes the postinst to exit immediately with an error.

A minimal fix would be to ignore any errors from the mkdir (`mkdir -p "${target_dir}" || true`), though there are a number of other problems with this postinst (`$SUDO_USER` is not guaranteed to be set in the environment, and the script will leave root-owned files in the user's home directory even if it succeeds).

We're working around this problem by allowing the configure step to fail, editing `/var/lib/dpkg/info/librealsense2-udev-rules:amd64.postinst` to comment out the offending `mkdir` line, and then running `dpkg --configure -a`; however, this requires manual intervention every time we attempt to update the systems (including potentially while doing unrelated security updates), so we'd prefer this be fixed upstream.

MartyG-RealSense, 2023-02-21:
Thanks very much @steven676 for your report.  I will highlight it to my Intel RealSense colleagues and update you as soon as I have information to provide.

MartyG-RealSense, 2023-03-07:
Hi @steven676  Thanks very much for your patience.  My colleagues thought that this issue may be primarily related to NFS system administration and noted that you were using a manual WA.

I found an article about errors when installing the SDK on an NRS system that may have helpful advice that leads to further solutions.

https://copyprogramming.com/howto/error-updating-from-such-a-repository-can-t-be-done-securely-and-is-therefore-disabled-by-default

steven676, 2023-03-09:
Thanks for the reply, but I don't think that addresses the permission
denied errors we're seeing (nor does it address things like the
incorrect use of $SUDO_USER).  root_squash is a standard security
feature of NFS servers and we're not going to disable it (see the "User
ID Mapping" heading of
https://man7.org/linux/man-pages/man5/exports.5.html or the NetApp manual:
https://docs.netapp.com/us-en/ontap/nfs-admin/handle-superuser-access-requests-concept.html),
so trying to write to /home from the postinst is just not going to work.

From our perspective, the librealsense2-udev-rules postinst is broken
and needs to be fixed.

-Steven Luo

MartyG-RealSense, 2023-03-09:
Thanks very much for your feedback.  I will discuss it with my colleagues.

MartyG-RealSense, 2023-03-10:
Hi @steven676  My colleagues will look at it.  In the meantime I recommend continuing with your manual workaround.

MartyG-RealSense, 2023-03-27:
Hi @steven676  I have an upate about your case.  My Intel RealSense colleagues have created an internal bug report to further explore the problem of postinst being broken on systems with NFS homedirs.  I have therefore added an **Enhancement** label to this case to signify that it should be kept open whilst the internal bug report is active.  Thanks very much for your continued patience!

dmipx, 2023-05-29:
Hi.
As i see, the content of udev-rules is very simple, it has only one file that appends rules for permission for usb camera device.
I see in other packages that postinst script trying to write presets to homedir.

`dpkg -c librealsense2-udev-rules_2.55.0-0~realsense.9543_amd64.deb 
drwxr-xr-x root/root         0 2023-05-29 02:25 ./
drwxr-xr-x root/root         0 2023-05-29 02:25 ./lib/
drwxr-xr-x root/root         0 2023-05-29 02:25 ./lib/udev/
drwxr-xr-x root/root         0 2023-05-29 02:25 ./lib/udev/rules.d/
-rw-r--r-- root/root      9311 2023-05-29 02:25 ./lib/udev/rules.d/60-librealsense2-udev-rules.rules
drwxr-xr-x root/root         0 2023-05-29 02:25 ./usr/
drwxr-xr-x root/root         0 2023-05-29 02:25 ./usr/share/
drwxr-xr-x root/root         0 2023-05-29 02:25 ./usr/share/doc/
drwxr-xr-x root/root         0 2023-05-29 02:25 ./usr/share/doc/librealsense2-udev-rules/
-rw-r--r-- root/root       267 2023-05-29 02:25 ./usr/share/doc/librealsense2-udev-rules/README.Debian
-rw-r--r-- root/root      1277 2023-05-29 02:25 ./usr/share/doc/librealsense2-udev-rules/changelog.Debian.gz
-rw-r--r-- root/root      6743 2023-05-29 02:25 ./usr/share/doc/librealsense2-udev-rules/copyright
`

MartyG-RealSense, 2023-06-03:
Hi @steven676  Was the advice of @dmipx in the comment above helpful, please?  Thanks!

MartyG-RealSense, 2023-06-10:
Hi @steven676  Could you assist @steven676 further with his question in the comment above, please?  Thanks!

MartyG-RealSense, 2023-06-21:
Hi @dmipx Could you assist @steven676 further with his question above, please? Thanks!

dmipx, 2023-06-21:
> @dmipx If committed to the package postinst, I believe that fix should work for us, though it won't do exactly what the original did. (apt is almost always run as root, so this will write to `/root/Documents/librealsense2`, rather than the home directory of a user running apt using sudo. Because there's no guarantee apt is being run using sudo, I admit I can't think of an easy way to do better.)

we fixed it to have user ownership in user homedir.

MartyG-RealSense, 2023-06-21:
Thanks @dmipx   :)

MartyG-RealSense, 2023-07-05:
Hi @steven676  As a fix was implemented to have user ownership in user homedir, do you require further assistance with this case please?  Thanks!

steven676, 2023-07-05:
I'm hoping to be able to verify that the fix works on our systems
sometime later this month, but there are other higher-priority items on
my to-do list at this moment.

MartyG-RealSense, 2023-07-05:
That's no problem at all.  This case can be kept open until you are ready to test.  Thanks very much for the update!

steven676, 2023-07-27:
I still get the same error updating to 2.54.1-0~realsense.9590, which
seems to be the currently-available version in the apt repository.  The
postinst looks identical to the ones in previous releases, so it doesn't
look like the fix has reached a public release of the RealSense packages
yet.

MartyG-RealSense, 2023-07-28:
Hi @dmipx  Can you comment on the continued problem of @steven676 described in the comment above, please?  Thanks!

MartyG-RealSense, 2023-08-03:
Hi @steven676  Do you still have your problem, please?

steven676, 2023-08-03:
I have not had time to test again, but I still don't see a version in
the repository newer than 2.54.1-0~realsense.9590, which is buggy -- my
assumption is that the problem is still present.  For reference, the
sources.list.d configuration we're using is

deb [arch=amd64] https://librealsense.intel.com/Debian/apt-repo focal main

MartyG-RealSense, 2023-08-04:
Please let me know when you have had an opportunity to test in order to confirm whether your problem is still present.  Thanks!

MartyG-RealSense, 2023-08-20:
Hi @steven676  Do you have an update about this case that you can provide, please?  Thanks!

steven676, 2023-08-21:
I ran a general apt-get update on the affected machines last week and
did not receive a librealsense update -- I believe it's up to Intel to
release an updated version of the librealsense Ubuntu packages with the
fix included.

MartyG-RealSense, 2023-08-22:
I will discuss this case further with my Intel RealSense colleagues.

MartyG-RealSense, 2023-09-04:
Hi @steven676  Thanks very much for your patience.  After discussion with my Intel RealSense colleagues about your question, it was confirmed that the fix will be in the next SDK release.

MartyG-RealSense, 2023-09-15:
Hi @steven676  Bearing in mind the information in the comment above, do you require further assistance with this case please?  Thanks!

closed 2023-09-21 by MartyG-RealSense