realsenseai/librealsense #11476: D435 Z accuracy is poor

k211, 2023-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | PC |
| SDK Version                            |  2.53.1                       | 
| Language                            |  opencv / C++                    | 
| Segment			|  Robot                          | 

### Issue Description
Hello, I am currently trying to test the Z accuracy of D435 camera using Depth Quality Tool. From the website "Best Known Methods for Optimal Camera Performance over Lifetime" (https://www.intelrealsense.com/best-known-methods-for-optimal-camera-performance-over-lifetime/), It says that Z-accuracy specificaton  D435 is 2% accuracy at 2 meters. However, my test result is quite poor. Even after I did the Tare Calibraton, the Z accuracy is only under 2% ± 10cm around the calibration point. Just as shown in the graph, If I do Tare calibraion at 944 mm, the accuracy is under 2% only from 850mm to 1500 mm; and if I do Tare calibration at 696 mm, the accuracy is under 2% only from 600mm to 800 mm. I think the main problem is that the slope of Realsense camera is larger than true distance.

The measurement was made indoor with good lighting. The result was collected by translating the D435 camera away from a wall, while carefully measuring the distace to the wall with the depth camera. The Groud Truth was measured using a laser range finder.  

Do you know how to fix it? Thanks a lot!

[image]

MartyG-RealSense, 2023-02-23:
Hi @k211  On RealSense 400 Series cameras, ground truth can be found by adding a particular minus-value to the depth measurements.  The value depends on the RealSense camera model being used.  For the D435i, **-4.2** (mm) should be added to depth measurements to determine ground truth.

With tare calibration, it is also recommended to use the **Get** button to calculate a custom GT value for the scene that your camera is being used in instead of using the default vlaue of '1200'.  

One of my Intel RealSense colleagues who is a calibration expert also provides advice at https://github.com/IntelRealSense/librealsense/issues/10213#issuecomment-1030589898 and also at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/6285037530259/comments/6381786011667

k211, 2023-02-23:
Hi @MartyG-RealSense, thanks so much for your quick reply.

I forgot to mention that I did use the **GET** button to calculate the custome GT value. 

I went through the advice you mentioned and found out that when doing Tare calibration, I forgot to turn the laser off. I might need to do the calibration again. 

One more question, if I have done the on-chip calibration, and the **Health-Check** values is 0.04 (in the range of **Good**), do I need to do **Dynamic Calibration** again in oreder to increase the Z accuracy?

Thanks again for your help!

MartyG-RealSense, 2023-02-23:
I would recommend choosing one calibration method or the other (Dynamic Calibration or On-Chip) rather than using both methods.  Either method can provide good accuracy.

A key advantage of Dynamic Calibration is that it calibrates the RGB sensor as well as the depth sensor, whilst On-Chip and Tare only calibrate depth. 

On-Chip and Tare have the advantages of the health check score system and the ability to be controlled with librealsense script code.  They also do not require the installation of a separate software package, as they are built into the camera firmware driver.  

Furthermore, On-Chip can calibrate both intrinsics and extrinsics (though as separate calibration operations rather than both at the same time), whilst the free version of the Dynamic Calibration software only calibrates extrinsics.  An OEM version of the software is available packaged with Intel's $1500 USD **OEM Calibration Target** product that calibrates both intrinsics and extrinsics, though ths product is targeted at engineering departments and manufacturing facilities and the free version of Dynamic Calibration is fine for the majority of RealSense users.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

On-Chip improves depth image **quality**, whilst Tare improves depth measurement **accuracy**.

k211, 2023-02-24:
Hi @MartyG-RealSense, thanks again for your advice!

closed 2023-02-24 by k211