realsenseai/librealsense #11480: Memory Leak in example.h with colorizer filter when rendering multiple depth streams

HerrmannM, 2023-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 |
| Firmware Version                   | 5.14.0.0 | 
| Operating System & Version |   Arch Linux  | 
| Kernel Version (Linux Only)    | 6.1.12  | 
| Platform                                 | PC with RSUSB backend |
| SDK Version                            |  2.53     | 
| Language                            |  C++     | 
| Segment			|  others                         | 

### Issue Description
Hi!

Working with two cameras D415 here. Using the provided example.h to display multiple depth streams causes a memory leak.
The bug appears here, when the same colorizer is used to color multiple distinct depth streams
https://github.com/IntelRealSense/librealsense/blob/master/examples/example.hpp#L677

If I instead color my depth streams (with their own colorizer) before calling
https://github.com/IntelRealSense/librealsense/blob/master/examples/example.hpp#L686
then all works well.

In the documentation, it is specified that filter can be used across multiple stream (although it is not recommended, e.g. for the temporal filter) https://dev.intelrealsense.com/docs/post-processing-filters, so I assume it is a bug in the colorizer rather than in the example?

I also have some other questions regarding the behaviour of the cameras/driver:

- Timing the filtering part only (frame->decimation->disparity->spatial->temporal->to depth->cloud point), I obtain slower time in 6FPS than in 30 FPS, and the 6FPS only uses half the CPU power compare to the 30 FPS version. Of course, lot of things can cause this, but is there any busy loops or things like that in the driver waiting for the frame or in the filter processing?

- With two cameras used in asynchronous mode, depth stream only, with RS2_OPTION_FRAMES_QUEUE_SIZE enforced to 1, I notice a bit of latency. My experiment is as follow:
  - both streams at 30FPS, but one of the cameras waits 1 seconds in the callback and observe both renders.
  - Quickly (less than a seconds) move my hand in front of the cameras
  - expected behaviour: the waiting camera, without any buffer (queue size of 1), should not display any intermediate frame (or maybe 1)
  - observed behaviour: several intermediate frames displayed by the waiting cameras

- In the documentation https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html the behaviour regarding frameset may be different between poll_for_frames and wait_for_frames:
  - poll_for_frames: wait for a coherent framesets - does not mention duplicating frames (see next point), so does not duplicate frames from the slowest streams
  - wait_for_frames: may duplicate a frame from the slowest streams
 
 Is it indeed the case that the framesets are not build the same way (without vs with "old" frames)?

Thank you!

MartyG-RealSense, 2023-02-24:
Hi @HerrmannM  In the RealSense SDK's **rs-multicam** multiple camera example, the program automatically creates a list of all attached cameras and their serial numbers. It then maps each camera's serial number to its own individual colorizer.

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L34

*************

You are correct that at 6 FPS there may be periods when the program is waiting for new frames to arrive.  This can lead to an increased risk of the program 'timing-out' if it does not receive a new frame after a certain period.  Using 15 FPS instead of 6 can help to avoid this whilst still saving on processing compared to 30 FPS.

The **spatial** post-processing filter can have a long processing time, whilst not providing a significant benefit in return.  So you may achieve better performance if you do not use this filter.

*************

RS2_OPTION_FRAMES_QUEUE_SIZE is already '1' by default.  This is the setting where performance is emphasized over latency, though with an increased risk of frame drops.  In Intel's frame buffering management documentation at the link below, they recommend setting it to '2' to add latency if two streams (such as depth and color) are enabled.

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0#latency-vs-performance

************

On RealSense cameras models that have a fast 'global' shutter on the depth sensor and a slower 'rolling' sensor on the RGB sensor (such as D415, D435 and D435i), fast motion such as a quick hand motion can cause lag and blur in the RGB image.  

If auto-exposure is enabled then setting depth to 30 FPS and RGB to 60 FPS can compensate for this.  

If you need to use a low FPS then disabling auto-exposure and setting a manual RGB exposure value of 78 and an FPS of 6 can also reduce the lag and blur.

The **D455** camera model has a fast global shutter on both its RGB and depth sensors.

************

For multiple camera applications, it is recommended to use **poll_for_frames()** instead of wait_for_frames() if you are not doing so already.  poll_for_frames delivers new frames immediately, whilst wait_for_frames() waits until a complete frame is available before it delivers it.  It is also recommended though to control when the CPU is put to sleep and for how long when using poll_for_frames(), otherwise the CPU's usage may max-out at 100%.  

More information about this subject can be found at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

***********

At https://github.com/IntelRealSense/librealsense/issues/5107#issuecomment-546870736 a RealSense team member advises that poll_for_frames() "is guaranteed to provide each new frame only once to avoid duplicates.  This however transfers the responsibility to synchronize the frames (required for cases such as align/pointcloud) to the user".

There is a discussion at https://github.com/IntelRealSense/librealsense/issues/5885 about 'stale frames' (old frames) when using wait_for_frames().

HerrmannM, 2023-02-27:
Hi @MartyG-RealSense 
Thanks a lot for you detailed answer! Please feel free to close this issue.

Regarding the colorizer, I still believe there is an issue as the doc does not reflect the fact that using the same colorizer for several depth streams actually does not work (i.e. leaks memory).

All the best!

MartyG-RealSense, 2023-02-27:
You are very welcome, @HerrmannM  - thanks very much for the update! As you suggested, I will close the issue.  Thanks again!

closed 2023-02-27 by MartyG-RealSense